pbatard/rufus #986: Is Windows To Go still a viable option on Rufus?

SaschaNaz, 2017-07-20:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.

Issue description
-----------------
I found some previous issue posts including #631 and #823 said that we can just use official ISOs from Microsoft instead of the ISO created by Media Creation Tool, but it seems now the two mentioned official sources [1](https://www.microsoft.com/en-us/software-download/techbench) [2](https://www.microsoft.com/en-us/software-download/windows10ISO/) are both unavailable. Still there are nonofficial sources e.g. [windowsiso.net](http://windowsiso.net/windows-10-iso/windows-10-th2-u1-download-build-10586-104/windows-10-th2-u1-iso-download-standard/), but probably this is not a recommendable way.

So it seems there is no official way to create Windows To Go media from Rufus, is it still a supported feature?

pbatard, 2017-07-20:
[2](https://www.microsoft.com/en-us/software-download/windows10ISO/) does work... if you are not using Windows 10 to download, or modify the `User-Agent` string of your browser.

Or you can also use [this tool](https://www.heidoc.net/joomla/technology-science/microsoft/67-microsoft-windows-and-office-iso-download-tool).

Despite Microsoft's efforts to annoy its user base, there are still plenty of ways you can download Microsoft official ISOs, so, of course, the Windows To Go option from Rufus is still very much supported.

closed 2017-07-20 by pbatard

SaschaNaz, 2017-07-20:
I'm currently on Windows 10 and it just redirects to https://www.microsoft.com/en-us/software-download/windows10/ which shows Media Creation Tool (or subscribers-only MSDN, etc).

The downloader tool works BTW, thanks :smile:

pbatard, 2017-07-20:
Yes, I know. Microsoft checks the `User-Agent` string of your browser to see if you're running Windows 10, and if you are, then https://www.microsoft.com/en-us/software-download/windows10ISO/ redirects to https://www.microsoft.com/en-us/software-download/windows10/ (which prevents you from downloading ISOs).

Part of me things this is a somewhat devious ploy from Microsoft to force Windows 10 people to use their inferior Media Creation Tool, instead of something better like Rufus 😄