pbatard/rufus #992: GPT/FAT32 flash drive not visible in UEFI boot options

zachkont, 2017-08-02:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
On my Dell XPS 9550 in UEFI mode (legacy mode is disabled) the bootable flash drives (for both windows and linux images) that I make in GPT/FAT32 with Rufus are not visible in the UEFI boot options.  To be exact, the first time it was visible and would boot into grub but wouldn't boot into the OS (linux mint 17.3) so I thought maybe the ISO was corrupted and redid the operation, but after that nothing is visible.

I know it's probably not a rufus problem but what am I doing wrong?

Log
---
```
Rufus version: 2.16.1170 (Portable)
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found non-USB removable device 'HGST HDN724030ALE640' => Eliminated
Found USB 2.0 device 'USB Flash Memory USB Device' (0930:6545)
Found non-USB removable device 'Samsung SSD 840 PRO Series' => Eliminated
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 489, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {3EF937C3-1D5C-414D-BFC1-63C47233CDC8}
Max parts: 128, Start Offset: 17408, Usable = 4029643264 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {CC8B3D32-A247-4C61-8F90-45AB1A9D1901}
  Size: 3.8 GB (4028612096 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.05/20140113 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Linux Mint 17.3 Cinnamon 64-bit'
  Size: 1.5 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v4.05
  Uses: EFI
Using image: linuxmint-17.3-cinnamon-64bit.iso
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.05/20140113 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Linux Mint 17.3 Cinnamon 64-bit'
[... 393 lines cut ...]
Extracting: D:\preseed\linuxmint.seed (633 bytes)
Extracting: D:\README.diskdefines (221 bytes)
Created: D:\syslinux.cfg
Finalizing, please wait...
Created: D:autorun.inf
Created: D:autorun.ico

Found non-USB removable device 'HGST HDN724030ALE640' => Eliminated
Found USB 2.0 device 'USB Flash Memory USB Device' (0930:6545)
Using autorun.inf label for drive D: 'Linux Mint 17.3 Cinnamon 64-bit'
Found non-USB removable device 'Samsung SSD 840 PRO Series' => Eliminated
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 489, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {71EBA7A7-62D8-470A-8CD5-6C18EB19810B}
Max parts: 128, Start Offset: 17408, Usable = 4029643264 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {320EBE1D-A940-47B7-A342-2685F7B380A3}
  Size: 3.8 GB (4028612096 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

```

pbatard, 2017-08-02:
I went through all the possible contention points, but I don't see anything that should prevent an UEFI compliant firmware from booting that drive:
```
Disk type: Removable, Sector Size: 512 bytes
```
Your USB uses a standard sector size, so it shouldn't be a problem (some drives use 4096-byte sectors, and are not recognized for that reason, but that is not the case here)
```
Partitioning (GPT)...
(...)
Formatting (FAT32)...
```
Your drive was properly partitioned as GPT and formatted as FAT32, so any UEFI firmware should be able to look at its content.
```
Extracting: D:\EFI\BOOT\BOOTx64.EFI (1.2 MB)
```
An `EFI\BOOT\BOOTx64.EFI` file was extracted, so an UEFI firmware should detect that and present you with the option to boot from it. The fact that it's called `EFI\BOOT\BOOTx64.EFI` and not `efiI\boot\bootx64.efi` (or any other combination of case) shouldn't matter, as compliant UEFI firmwares should not bother about case, especially on a FAT file system.
Provided that you are using an x64 UEFI firmware on your Dell XPS 9550 (which I'm pretty sure is the case, especially as you mention the same ISO worked before, and it only provides an x64 boot loader), there's nothing obvious as to why your device should fail to be detected as UEFI bootable.

The log also shows that your USB creation completed successfully, so everything should be properly set for UEFI boot.

The one thing I would suspect, since you mention that your USB booted once, but not afterwards, is that the flash memory on your USB could be defective, but if you ran a bad blocks check, and it reported no errors, and tried a different USB flash drive, as you mention, then this hypothesis should be eliminated as well. Intermittent boot issues (worked some time, doesn't work other time) do suggest a problem with the hardware, as under the same conditions, software will behave in the same manner. But then again, that's why I ask to try plugging the USB on a different port, to eliminate the possibility of a connectivity issue, and you indicate that you try that.

So, if you have really tried everything from the check list, then I don't see anything I can advise you with. As far as I'm concerned, your USBs should boot, and if they don't, you will have to take that either with your flash drive manufacturers or Dell.

zachkont, 2017-08-02:
Ok, thanks for your time

pbatard, 2017-08-02:
OK. I don't think there's anything more I can do to help, so I will close the issue.

closed 2017-08-02 by pbatard