pbatard/rufus #994: Ricoh internal SD reader not supported

ehrmann, 2017-08-05:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus is eliminating the internal SD reader in my Thinkpad X1 Carbon since it's not a USB device or on a whitelist.

Windows reports this in the hardware ids:

```
SCSI\DiskRicoh_Storage_SD&REV_3.0
SCSI\DiskRicoh_Storage
DiskRicoh_Storage
SCSI\GenDisk
GenDisk
```

Let me know if you need any more info for identifying the device.

Log
---
```
Rufus version: 2.16.1170 (Portable)
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found non-USB removable device 'Ricoh SD Disk Device' => Eliminated
0 devices found
```

pbatard, 2017-08-05:
Can you please post a log, but with <kbd>Alt</kbd>-<kbd>.</kbd> (USB Debug) enabled. It should have a lot more details about device enumeration and will help with adding support for your card reader.

ehrmann, 2017-08-06:
Sorry--I was looking for how to do that.

```
Found non-USB removable device 'Ricoh SD Disk Device' => Eliminated
0 devices found
Found non-USB removable device 'Ricoh SD Disk Device' => Eliminated
0 devices found
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub30#4&1c46e756&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[096]: USB\VID_0BDB&PID_1926\9CDDA66F0FD915C0
Processing Hub 2:
Processing Hub 3:
  Hub[2] = '\\?\usb#vid_8087&pid_0024#5&18c73dba&2&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[139]: USB\VID_147E&PID_2020\6&6CDBC03&0&3
  Found ID[188]: USB\VID_0A5C&PID_21E6\083E8EE60CEE
  Found ID[184]: USB\VID_04F2&PID_B315\6&6CDBC03&0&6
Processing Hub 4:
  Hub[3] = '\\?\usb#root_hub20#4&2f36d08&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[226]: USB\VID_8087&PID_0024\5&18C73DBA&2&1
Processing Hub 5:
  Hub[4] = '\\?\usb#root_hub20#4&3981b9c9&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[117]: USB\VID_8087&PID_0024\5&3923A75A&2&1
Processing 'IDE' device:
  Disabled by policy
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskRicoh_Storage_SD&REV_3.0'
Found non-USB removable device 'Ricoh SD Disk Device' => Eliminated
If you *REALLY* need, you can enable listing of this device with <Ctrl><Alt><F>
0 devices found
```

pbatard, 2017-08-06:
Thanks. Just for the record, and as indicated in the log, you should be able to enable access to your card reader with <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd> for the time being, though you might want to be careful about enabling this cheat mode, as it make make other disks visible in Rufus...

It'll probably be a few days before I can work on adding that Ricoh device (and probbaly a few __months__ before I release a version of Rufus that integrates it).

commit 3a150dde (2017-08-11): [core] add support for more non-USB card readers

* Closes #994
closed 2017-08-11 by pbatard