pbatard/rufus #996: "LOCALE" directory (DOS) is visible

marcosfrm, 2017-08-10:
A directory containing locale files ("LOCALE" on filesystem root) is created in DOS bootable devices. Users hardly should touch what is inside it, so maybe it can have FILE_ATTRIBUTE_HIDDEN|FILE_ATTRIBUTE_SYSTEM like COMMAND.COM and KERNEL.SYS?

I tested marking it as SH (FreeDOS bootable UFD, created with Rufus 2.16 on Windows 10) and FreeDOS locale stuff still works.

pbatard, 2017-08-10:
I'm sorry but I don't see a good reason to do so. On the contrary, I'd rather have people realize that they do have localization settings they may be able to use if the keyboard locale that Rufus tried to select is not the one they wanted (Rufus tries to select a locale automatically, but any process that's automatic may not produce the results that the user wants). Being able to look into the `LOCALE` folder may help.

I'm afraid I will have to decline that request.

closed 2017-08-10 by pbatard

marcosfrm, 2017-08-10:
No problem. :-)