pbatard/rufus #1000: Undetermined error

Ottobalance, 2017-08-13:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I now got an undetermined error. I use a 16gb ONN 2.0 USB.

Log
---
```

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'I:' as volume mountpoint
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'I:' after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1912, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x006BE283
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 14.6 GB (15727591424 bytes)
  Start Sector: 2048, Boot: Yes
Closed Windows format prompt

```

Ottobalance, 2017-08-13:
On my last post I did the checklist but I didn't know that I had to keep it up on my post.

pbatard, 2017-08-13:
Can I see the output with the bad blocks check?
Also, can you try with a different USB drive or from a different computer?

Finally, what happens if you try to format that drive in Windows (i.e. not using Rufus)?

pbatard, 2017-08-14:
Any update on this? If no, I will have to close this issue...

pbatard, 2017-08-21:
No update &rarr; Closed.

closed 2017-08-21 by pbatard