pbatard/rufus #1001: UEFI is not bootable

ebonweaver, 2017-08-14:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
MBR for BIOS and UEFI boots in both legacy and EFI mode as the name implies.
MBR for UEFI results in a non-bootable drive.  
GPT for UEFI results in a non-bootable drive.
Interestingly, the drive IS bootable on a Mac in the second two schemes, but not on a Dell (in either legacy or EFI modes just as a sanity check).

Log
---
```
Rufus version: 2.16.1170
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
0 devices found
```

pbatard, 2017-08-14:
Your log is incomplete. Without the complete log, as requested, I am __unable__ to understand how you create your USB, and therefore I cannot tell you why it isn't bootable.

As far as your log indicates, you have not used Rufus to do anything. Therefore, you can hardly expect your USB drive to be bootable.

ebonweaver, 2017-08-14:
I was trying to spare the spam since it happens all the time, but ok, here first is a log for making MBR for UEFI.  You'll also see in here an incident that Rufus likes to do now and then where it says the drive is in use when it's not, and works fine the second time you click start.

Rufus version: 2.16.1170
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'WD Elements 25A2 USB Device' (1058:25A2)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found USB 2.0 device 'WD Elements 25A2 USB Device' (1058:25A2)
1 device found
Disk type: FIXED, Sector Size: 512 bytes
Cylinders: 121597, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {FCC866A2-CBB2-4D3F-BF0B-3516F6A7E247}
Max parts: 128, Start Offset: 17408, Usable = 1000170551808 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {6E996705-CE6D-4E64-A8AC-47B8878F35AD}
  Size: 931.5 GB (1000168436736 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {0BFFDB1D-C3D0-4423-A4CD-5E467C59F004}
  Size: 256 KB (262144 bytes)
  Start Sector: 1953456026, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CEDA_X64FREV_EN-US_DV5'
  Size: 3.9 GB (Projected)
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Win10 EDU-64-1703.iso
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CEDA_X64FREV_EN-US_DV5'
  Size: 3.9 GB (Projected)
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Win10 EDU-64-1703.iso
Warning: Conflicting process search failed to complete due to timeout

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE3 for exclusive write access
Requesting lock...
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
[... 1810 lines cut ...]
   4224916 KB in 1717 files.
       584 KB in 110 indexes.
     97804 KB in use by the system.
     65536 KB occupied by the log file.
 972403716 KB available on disk.
      4096 bytes in each allocation unit.
 244181755 total allocation units on disk.
 243100929 allocation units available on disk.
NTFS Fixup completed.

Found USB 2.0 device 'WD Elements 25A2 USB Device' (1058:25A2)
1 device found
Disk type: FIXED, Sector Size: 512 bytes
Cylinders: 121597, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x197A871D
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 931.5 GB (1000168468992 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 1953456089, Boot: No

ebonweaver, 2017-08-14:
Here is a log for making a GPT drive:

Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CEDA_X64FREV_EN-US_DV5'
  Size: 3.9 GB (Projected)
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Win10 EDU-64-1703.iso
Warning: Conflicting process search failed to complete due to timeout

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE3 for exclusive write access
Requesting lock...
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Found volume GUID \\?\Volume{87b340aa-ec03-492c-929d-81b1523b0899}\
\\?\Volume{87b340aa-ec03-492c-929d-81b1523b0899}\ was already mounted as G:\
Successfully remounted \\?\Volume{87b340aa-ec03-492c-929d-81b1523b0899}\ on G:
Copying ISO files...
Extracting files...
Image is an UDF image
Extracting: G:\autorun.inf (128 bytes)
Extracting: G:\boot\bcd (16 KB)
Extracting: G:\boot\boot.sdi (3 MB)
Extracting: G:\boot\bootfix.bin (1 KB)
Extracting: G:\boot\bootsect.exe (103.4 KB)
Extracting: G:\boot\en-us\bootsect.exe.mui (16 KB)
Extracting: G:\boot\etfsboot.com (4 KB)
Extracting: G:\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: G:\boot\fonts\cht_boot.ttf (3.7 MB)
Extracting: G:\boot\fonts\jpn_boot.ttf (1.9 MB)
Extracting: G:\boot\fonts\kor_boot.ttf (2.3 MB)
Extracting: G:\boot\fonts\malgun_boot.ttf (173.3 KB)
Extracting: G:\boot\fonts\malgun_console.ttf (80.5 KB)
Extracting: G:\boot\fonts\malgunn_boot.ttf (170.9 KB)
Extracting: G:\boot\fonts\meiryo_boot.ttf (142 KB)
Extracting: G:\boot\fonts\meiryo_console.ttf (89.5 KB)
Extracting: G:\boot\fonts\meiryon_boot.ttf (140.4 KB)
Extracting: G:\boot\fonts\msjh_boot.ttf (160.5 KB)
Extracting: G:\boot\fonts\msjh_console.ttf (199.1 KB)
Extracting: G:\boot\fonts\msjhn_boot.ttf (158.5 KB)
Extracting: G:\boot\fonts\msyh_boot.ttf (152.6 KB)
Extracting: G:\boot\fonts\msyh_console.ttf (95.2 KB)
Extracting: G:\boot\fonts\msyhn_boot.ttf (150.8 KB)
Extracting: G:\boot\fonts\segmono_boot.ttf (43.8 KB)
Extracting: G:\boot\fonts\segoe_slboot.ttf (84.2 KB)
[... 1713 lines cut ...]
 972403684 KB available on disk.
      4096 bytes in each allocation unit.
 244181747 total allocation units on disk.
 243100921 allocation units available on disk.
NTFS Fixup completed.

Found USB 2.0 device 'WD Elements 25A2 USB Device' (1058:25A2)
1 device found
Disk type: FIXED, Sector Size: 512 bytes
Cylinders: 121597, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {1B76C223-461E-4BEE-B2D1-77E8C38D657F}
Max parts: 128, Start Offset: 17408, Usable = 1000170551808 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {87B340AA-EC03-492C-929D-81B1523B0899}
  Size: 931.5 GB (1000168436736 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {DB1FDA76-B7CB-48ED-8B8B-4CA39D505F4F}
  Size: 256 KB (262144 bytes)
  Start Sector: 1953456026, Attributes: 0x0000000000000000

pbatard, 2017-08-14:
Now we're talking.

1. You're using UEFI:NTFS. That'll not boot __unless__ you temporarily disable Secure Boot, which is probably enabled on the Dell. I don't think there's a Secure Boot option for Mac (since Microsoft controls the signing of the bootloaders).
2. Alternatively, since your `sources/install.wim` is less than 4 GB, you can use FAT32 instead of NTFS, in which case trying to boot in UEFI mode with Secure Boot "should" work.
3. In case you are under the impression that __temporarily__ disabling Secure Boot is a security liability, I encourage you to read the second part of [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Blah_UEFI_Blah_FAT32_therefore_Rufus_should_Blah) (though, ironically, the part about thinking that NTFS cannot be booted from UEFI does not apply to you, which is refreshing).
4. I think I should also point out that when Rufus says "UEFI-CSM" in the first option, it means "BIOS compatibility mode" (which is what CSM stands for) and not UEFI. So, if you saw "MBR partition scheme for BIOS or UEFI-CSM" when you created your drive for the test where you say that _"MBR for BIOS and UEFI boots in both legacy and EFI mode as the name implies"_, then, unless your UEFI firmware has an NTFS driver (I don't believe Dell machines have such a thing yet, but if your machine is recent who knows...), you shouldn't have been able boot in EFI mode. That is, unless you use the [<kbd>Alt</kbd>-<kbd>E</kbd> cheat mode of Rufus](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes). In other words, either you have been seeing "UEFI-CSM" and not "UEFI" as you reported, or you have been using kbd>Alt</kbd>-<kbd>E</kbd>. In either case, there was some more vital information that was inaccurate or missing from your report, which makes it tricky to provide help. Note that if you want to create dual BIOS+UEFI bootable drives for Windows, you may also want to read on [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_dual_BIOSUEFI_mode).

pbatard, 2017-08-14:
PS: If Rufus says the drive is in use, you can be pretty sure it is, even if the application or service that is using the drive isn't obvious. Especially, Windows has a habit of creating files and folders (e.g. `System Volume Information`) on newly inserted drives, which of course, when not completed, would produce drive access errors. This is why Rufus will tell you if there are other processes (including system ones) that are accessing your drive.

ebonweaver, 2017-08-15:
Thanks for the reply!
1. Odd, any time I've run afoul of secure boot thus far, it says something about secure boot, and this did not.  We'll poke around tomorrow for that option and see if that fixes the issue.  If so, I'll pester our Dell rep about their lack of clear errors (I'm sure they'll get right on that...)
2. Yeah, after making the drive we replace the install.wim with one that is upwards of 100gb, hence using NTFS (hence using Rufus in fact, and thanks for the great tool btw!)
3. No problems with temp disabling for installs, and glad to be refreshing!
4. I haven't seen CSM anywhere, not sure what you're referring to there.  It does not appear in any options in Rufus.  I didn't know about Alt-E, but using that indicates that by default dual mode is enabled, because pressing that combination pops the message that it's disabled, and pressing again enables it once more.  Either way, CSM does not appear anywhere.  So to my knowledge, nothing was omitted from the report.  If this behavior does not line up with your expectations, something else would seem to be at play, or I'm not understanding what you're pointing at.

PS: Well isn't that rude of Windows to get in the way of essentially it's own operation.  MS is grand like that.  To compensate, perhaps if your alert message added "or possibly wait a moment and try again as Windows may be playing with the drive" or something to that effect to help people understand how it's in use when it's "not in use"?

Full story as you may find it interesting, what I'm ultimately working towards is trying to make pocket drives that can be used to image both PC systems, as well as single and dual boot Mac systems.  The routine thus far is use Rufus, shrink partition by 50% and make second partition, put in custom install.wim, image the other partition with Mac OS that has DeployStudio.  Thus far this only works when using the MBR for UEFI partition option (#2) in Rufus.  If you use MBR for BIOS or UEFI (#1), the Mac can't boot it.  If you use GPT (#3), the Mac can't reformat the second partition for some reason, even though they also use GPT (it claims there is no space).  Not sure if it's related to the tiny partition Rufus puts at the end of the drive with that option or something else.  The bummer so far then has been not being able to boot anything but the first option on the Dells, so hopefully it is just a stupid secure boot issue as you suggest.

pbatard, 2017-08-15:
> I haven't seen CSM anywhere, not sure what you're referring to there.

I'm referring to the following. Without <kbd>Alt</kbd>-<kbd>E</kbd>, you will see this in Rufus when you try to create a bootable media from a Windows ISO and use the "MBR for BIOS..." option:

[image]

This means that the USB should only ever boot in BIOS mode, even on an UEFI system, since the drive is will be formatted as NTFS, and very few UEFI firmwares are able to natively perform a lookup for EFI boot files on an NTFS file system (the vast majority are limited to FAT32 __only__). In other words, pure UEFI boot should always fail with such a drive, and this is why Rufus indicates that the boot mode that will be used for an UEFI system with be CSM.

On the other hand, if you apply the <kbd>Alt</kbd>-<kbd>E</kbd> cheat mode, the dropdown selection changes to:

[image]

This is because, if you use <kbd>Alt</kbd>-<kbd>E</kbd>, Rufus will also install UEFI:NTFS on the drive, which will load an NTFS driver and enable chain booting of the EFI loader from the NTFS partition in pure EFI mode. Thus, in this case, and as opposed to the default setting, either of BIOS or pure UEFI are supported.

Thus, when you said _"MBR for BIOS and UEFI boots in both legacy and EFI mode as the name implies."_, I'm a bit puzzled, because, unless you used <kbd>Alt</kbd>-<kbd>E</kbd> what you should have seen is _MBR for BIOS and UEFI-CSM_ not  _MBR for BIOS and UEFI_, and since you didn't mention using <kbd>Alt</kbd>-<kbd>E</kbd>, I have to assume that this is really what you got. But in that case, unless the Dell machine you are using have an EFI NTFS in their firmware (which is very possible, as manufacturers are slowly coming to the conclusion that limiting themselves to the mandated FAT32 really won't cut it, and are starting to add NTFS support), EFI boot should not have worked. Yet you mentioned that it did. So I'm just trying to get some clarification on what you actually saw, as well as get a hint as to whether Dell has finally joined Intel, Gigabyte and others, in starting to provide NTFS support in their UEFI firmwares.

> Well isn't that rude of Windows to get in the way of essentially it's own operation.

Sure wish that was the only place it did that. It could also be another application, as some security solutions (Antivirus software and the like) are keen on keeping a lock on disk devices for longer than they should.

> perhaps if your alert message added "or possibly wait a moment and try again as Windows may be playing with the drive" or something to that effect to help people understand how it's in use when it's "not in use"?

Yeah, I thought about that, but I believe that's rather obvious. If something doesn't work, I doubt many people will go "oh well" and give up, before trying a second time. And, if you want to consider the problem from another perspective, my guess is that people __will__ get pretty annoyed if, by the 3rd or 4th time they try it, they still see a "this may be temporary - please try again" message, and they'll probably start yelling stuff like _"And what do you think I'm doing?!?"_ at the application. Overall, I think it may be better not venture an hypothesis that may be wrong, in order to avoid potentially antagonizing users, and just stick to presenting the facts.

> If you use GPT (#3), the Mac can't reformat the second partition for some reason

Well, I know that Apple has placed some weird limitations on their disk partitioning and formatting utility, such as refusing to operate on a drive where the ESP is less than 200 MB. #979 may be of interest to you in that respect (though I'm not planning to do anything in Rufus, as this is really something that Apple should fix).

ebonweaver, 2017-08-15:
Well as I said, never seen that CSM.  Ever.  In any instance.  No matter what I do, can't make it appear, on any hardware of any model.  Not sure what else to say on that, but the idea that Dell "finally" added something seems unlikely when we're talking 5 year old hardware.

Checked the BIOS settings of the machines we're trying to boot, no option for secure boot, seems they are too old to have that (Optiplex 7010).  As such, no idea why we can't boot the machines from the Rufus drive.  Logs were from an Optiplex 790, which DOES have secure boot in the BIOS, but that should be irrelevant vs what we're trying to boot.
The exact boot error is "Selected boot device failed" for what that may or may not be worth.  

Sounds like our experience isn't lining up with your expectation for some reason.

I get ya on the alert message, can't disagree given all I did was "huh?  no it's not try again" myself  :)

pbatard, 2017-08-15:
> can't make it appear, on any hardware of any model.

Just to be clear, that's in the Rufus UI, and that will only appear __after__ you have selected an ISO.
Unless you enabled <kbd>Alt</kbd>-<kbd>E</kbd> previously, as soon as you select that ISO and select the first option under "Partition scheme and target system type" in Rufus, you should see "UEFI-CSM" and not "UEFI". And if you press <kbd>Alt</kbd>-<kbd>E</kbd> you should definitely see it toggle between UEFI and UEFI-CSM.

Of course, as long as you haven't selected an ISO, and since the UEFI-CSM restriction is only applied for Windows, Rufus can't tell if what you're going to create with the first option will be compatible with dual BIOS+UEFI, so it tentatively assumes that it will... until you actually select an ISO.

Please try again after selecting your Windows ISO and after pressing <kbd>Alt</kbd>-<kbd>E</kbd> a couple of times. I can __guarantee__ that you will see the option change between UEFI and UEFI-CSM.

And, if you also pay attention to what after you selected your ISO but before you pressed <kbd>Alt</kbd>-<kbd>E</kbd>, you should be able to tell me whether the USB you created with the first option were dual BIOS+UEFI bootable, or BIOS-only.

> seems they are too old to have that (Optiplex 7010)

Or they might have an UEFI firmware that prevents Secure Boot from being disabled. At one stage, Microsoft was pushing quite hard to force manufacturers to go with something like that (though mostly for ARM devices). Maybe Microsoft strongarmed Dell into hiding the Secure Boot option as part of their licensing deal. Either that or, maybe you somehow thought you were booting in UEFI mode, whereas you were booting in UEFI-CSM mode (i.e. "BIOS mode"). Especially, there should be __zero difference__ in UEFI boot capabilities between a drive  actually created with "MBR for BIOS or UEFI" or "MBR for UEFI". There too, I can guarantee that you'll be hard pressed to find any way for Rufus to screw this up, as, in both case,  Rufus will create the UEFI:NTFS partition in the same manner (it's actually a flat 256KB FAT image that we apply wholesale at the end of the drive), and the only other element that is required for boot is the presence of a `efi\boot\bootx64.efi` bootloader (since, from what I can see online, the Optiplex 7010 uses an x86_64 CPU), which would have been extracted from the ISO, and would produce an error if it wasn't.

Moreover, provided that the 7010 doesn't have an EFI NTFS driver (which I'm 99.9% sure it hasn't if it's 5 years old), then, if the drive you created using "MBR for BIOS or UEFI" actually booted in UEFI mode, you would see about 10 lines of logging from UEFI:NTFS before it hands over the boot to the NTFS partition. While these messages occur pretty quickly, they're not so fast to go unnoticed (since there's incurred delay while the NTFS driver is being loaded), and I'm pretty sure you would be able to spot the first line that explicitly mentions that UEFI:NTFS is launching, which I would expect you to recall.

Considering this, I have to assume that:
1. You didn't use <kbd>Alt</kbd>-<kbd>E</kbd> but kept the default, which means that the drive you created using the "MBR for BIOS or UEFI" option was actually with "MBR for BIOS or UEFI-CSM"
2. When booting that drive in what you thought was pure UEFI mode, you were actually booting it in BIOS mode, which is why you didn't see any mention of UEFI:NTFS being used, and don't remember doing anything special for the first option to work.
3. The reason why "MBR for UEFI" or "GPT for UEFI" options don't work is simply because you haven't properly set your Optiplex 7010 to boot in pure UEFI mode. Especially, there's no way "MBR for UEFI" would not work if you did indeed use "MBR for BIOS or UEFI" for your first test (i.e. with <kbd>Alt</kbd>-<kbd>E</kbd> having been applied) and if you were actually booting in pure UEFI mode on the target.

I would strongly encourage you to double check your UEFI options on the 7010, because everything you have reported so far seems to provide a lot of circumstantial evidence that you've never actually managed to boot that machine in pure UEFI mode, and that this has nothing to do with Rufus.

For instance, to completely eliminate Rufus out of the equation, in case you are still concerned that Rufus might be the issue, you may want to check what happens if you:
1. Format your drive to FAT32 using Windows alone (you may also use diskpart to set the drive to GPT or MBR, but it really shouldn't matter if the Dell firmware is UEFI compliant, as it will be able to boot either)
2. Extract the data from your Windows 10 ISO (using 7-zip or any other ISO extraction utility) onto the flash drive
3. Plug that drive into the Optiplex 7010.

If the 7010 is set to boot from UEFI, then it should boot a drive created using the above steps without any trouble. But if you find it doesn't, then it means that you still haven't enabled the right firmware options to enable pure UEFI boot on that machine, and that has been your one issue all along.

ebonweaver, 2017-08-15:
Ah!  Important footnote on order of operations and what is needed to see that option.  Yes, now I have been able to make CSM appear.  Since it ONLY shows up when you use alt+e to disable dual mode, and I never saw it before now, that seems to say the defaults in Rufus are to have dual mode enabled, so yes all the drives we've made are dual mode.  I'm unclear however if dual mode has any impact on options 2 and 3 or just option 1 (hence only 1 changes its description)?

I'm guessing only option 1, and what you're saying is we've never been able to put a machine into UEFI mode which is why they don't boot using options 2 and 3.  That seems legit, and based on this conversation we're finding BIOS anomalies.  One big issue is BIOS versions WAY out of date, which should not be the case, which is a conversation I need to have with my techs....  the older BIOS versions have no secure boot option, new BIOS for the same machine does.  Unclear so far if that means enabled by default on old version, or with option missing boot fails anyway with modified media in UEFI.  New version also has an additional legacy option in addition to the legacy vs UEFI main option, so plot thickens, consistent with your assertions.  Trying to tweak these settings however just killed the test machine's video output which is a new set of WTF mystery.... 

So, once we untangle this BIOS version and setting mess and I can get some more reliable tests, I will update with results.  I suspect you got us pointed in the right direction now.  At least I hope so, I hate things that require a young priest and an old priest to sort out....  Assuming that's the crux, appreciate the help running this to ground, sorry it turned out to be internal version control  -_-

pbatard, 2017-08-15:
>  Important footnote on order of operations and what is needed to see that option.

Yeah, I should have been clearer on that.

> that seems to say the defaults in Rufus are to have dual mode enabled

No, the default is precisely to prevent dual BIOS+UEFI. I am therefore positive that you used the <kbd>Alt</kbd>-<kbd>E</kbd> cheat mode previously if you saw "MBR for BIOS or UEFI" instead of "MBR for BIOS or UEFI-CSM". Be mindful that this option is preserved between sessions, so it might have been months ago. However, the first time you run Rufus, dual BIOS+UEFI is disabled, and that is something I'm 100% positive about (even more so as this assertion is very easy to test).

> so yes all the drives we've made are dual mode.

Okay. Then if they didn't boot in "MBR for UEFI" but did in "MBR for BIOS or UEFI", the most logical explanation is that you didn't actually boot in pure UEFI with the first option.

> New version also has an additional legacy option in addition to the legacy vs UEFI main option

That's weird. All I can say is that it took a few years for manufacturers to get up to speed with UEFI, and I'm not entirely sure we still quite there (I have an intel NUC firmware that somehow managed to completely disable UEFI boot from USB, unless you manually do it from the UEFI shell, and a modern Gigabyte motherboard that also is unable to recognize that the GPU firmware is UEFI compatible, and therefore forces legacy to be enabled, which is the same problem you're seeing on your Dell), so I would be surprised if some of your firmware options are hit & miss.

>  Trying to tweak these settings however just killed the test machine's video output which is a new set of WTF mystery....

Nah, that's just UEFI not liking legacy stuff: unless your GPU says it's UEFI compatible (which is not something I'd expect from a 5 year old machine unless you have a recent video card in there), a pure UEFI boot process will refuse to use it, so you'll get a blank screen (usually until the OS takes over, but maybe not always). Welcome to the BIOS &rarr; UEFI transition hell. Oh, and for the record, if you want to have Secure Boot, you can't have anything legacy running...

> sorry it turned out to be internal version control

Not at all. I prefer seeing issues reported, where it turns out that Rufus isn't really involved in the problem, than people withholding on reporting issues, where it turns out there's an actual Rufus bug to fix.

ebonweaver, 2017-08-15:
Video cards have to be UEFI compatible??  THAT I was unaware of... it also means another email to the Dell reps because they shipped systems that aren't compatible with their own options, and that's just sloppy in my book.  Also explains a lot of forum threads I've found with people having this problem, not a single one of which, even with Dell tech reps, said oh, it's because your card is too old for the option in the BIOS of the system it shipped in....   ugh... 

As for the Alt+e thing, I didn't know about that hack until you mentioned it.  That's all the more I can say on that as to what happened on my end.  *shrug*  Can't rule out a random keyboard smash I guess?  Sunspots?  UFOs?

At any rate, the graphics thing clearly shows we never got in pure EFI mode at any point, hence the boot failures.  It also shows we are FAR from ready for EFI deployment in our fleet, glad we found that sooner than later.  The real pain of course being some new hardware that has no legacy support.... 

Anyway, thanks again, it's been an invaluable experience!

pbatard, 2017-08-21:
I'm gonna close this issue now. Feel free to reopen it if you think that there's something off with UEFI bootable drives created by Rufus.

closed 2017-08-21 by pbatard

makedir, 2018-08-22:
I have the same issue since forever with Rufus, it never worked on any PC. The problem seems to be, that Rufus oesnt properly write a boot sector or a flag to a USB stick. I dont know why but this works:

Let Windows Media Creation tool run and create a USB stick
NOW use Rufus.

Only that will work. Otherwise, the stick wont show up as bootable UEFI stick.

The problem now is, that the Let Windows Media Creation tool is bugged and I cant use it it will always return an error.

Anyone know how to do it with just Rufus?

pbatard, 2018-08-22:
Well, if an UEFI firmware requires a boot sector, then that firmware is __not compliant__ with the UEFI specs.

All UEFI mandates for booting a disk drive (such as a USB Flash Drive) is that there exists a partition with a file system it recognizes (usually FAT32, but that is not mandatory, as some systems, such as intel NUCs will have no trouble if NTFS is used), and that the relevant EFI bootloading application (`bootx64.efi`, `bootia32.efi`, `bootarm.efi`, etc) exists in a `/efi/boot/` directory on that file system.

Especially, and this needs to be stressed out, __UEFI completely does away with boot sectors__ (which are a __pure__ BIOS concept). So, unless you are booting in CSM mode (in which case you are really booting in BIOS emulation), there is simply no way the presence of a boot sector can have any incidence on boot. UEFI firmwares are simply not designed to use them (again outside of CSM). There are also no "boot flags" in the (non CSM) UEFI world, at least for the initial boot.

Now, possibly, some systems might be anal and require that the partition where the `.efi` boot files reside is of ESP type (in other word, that it is created as a GPT partition with a specific GUID), but I have yet to come across such a system, especially as manufacturers will naturally tend to avoid doing that to be able to also boot MBR partitions in UEFI mode.

Moreover, the first thing Rufus does when creating a drive is erase all the partitions and boot records, so it seems very surprising that anything that MCT applied to your drive would persist.

Just as with the original issue, I suspect that there is some confusion here with regards to what the actual underlying cause of your problem is, and that you might be a bit too quick to jump to conclusions with regards to the lack of a boot sector being the issue, because if you know how UEFI works, it simply makes no sense. You may therefore want to investigate this some more, and especially make sure as to whether you are really booting in pure UEFI mode or UEFI-CSM...

makedir, 2018-08-22:
No. It is, like it is. I had 4 PCs over the past 10 years, and just using a brand new USB stick with a windows.iso + Rufus NEVER worked, never. I am just writing here, because I had to do a fresh Windows 10 install on a dew Dell XPS 9570 laptop. Secure boot, true UEFI ect ect ect is enabled (default, nothing changed). No CSM mode or whatever...

Using a new factory formatted 16GB UB stick + Rufus + Windows.iso (downloaded via Win media tool) doesnt work. The stick wont show up in the UEFI boot list of the laptop.

Doing the above steps, clean sticks bootsector, write iso to stick WITH MS too, then just do it again for fun with Rufus => now stick will show up as UEFI device, and I just did a Win 10 install via it...

I am no expert on this and whats causing it, I just know that it is like that. Rufus never worked for me and 4 PCs I had with clean stick + iso.

ibuprophen1, 2018-08-23:
This looks familiar to another situation I helped someone with on an XPS 9560 that wouldn't recognize a Windows 10 UEFI USB Stick as well. 

What had resolved the problem was going into the BIOS Boot Mode and enabling both the UEFI and Secure Boot functions. 

Then, after you save and reboot the PC, then just select (i believe it's the same) the F12 for your Boot Menu and the USB UEFI option should then show up. 

I hope that I had explained this okay via text.... 👍

~Ibuprophen

pbatard, 2018-08-23:
> No. It is, like it is. I had 4 PCs over the past 10 years, and just using a brand new USB stick with a windows.iso + Rufus NEVER worked, never.

Then it certainly looks to me like you were unaware as to whether these PCs were using pure UEFI or CSM mode to boot (or BIOS), and used the wrong settings in Rufus.

Rufus is designed to prevent booting of Windows images if you don't use the mode that is appropriate for your target, as per [this entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_dual_BIOSUEFI_mode), so if you aren't certain of which mode your target is using and aren't using the right settings, it would certainly explain a lot of things especially as Rufus will try to re-select some of the settings of the USB you plug in. For instance, if you had a GPT partitioned drive and plugged in into Rufus, Rufus will try to re-select GPT mode when re-creating the USB for Windows, which obviously mean it will only boot in UEFI mode, and not in BIOS or UEF-CSM. Then I can very much see the MCT re-partitioning the drive to MBR mode when you use it (since it tries to make it UEFI+BIOS bootable), meaning that the next time you plug that drive in Rufus, Rufus will see that it is MBR partitioned and __now__ it will use MBR (and possibly CSM/BIOS-compatibility mode) to create the drive, in which case, that drive will now boot on a BIOS or UEFI-CSM system.

All this to say that, you are very much supposed to know (and understand some properties) of the type of system you are trying to boot to use Rufus, as you must change settings accordingly to be able to install Windows.  And considering that you are the __only__ person, in tens of millions of people who used Rufus to install Windows, to report that no matter what they tried, they couldn't make it work unless they first went through a drive creation with the MCT, which makes no logical sense unless the person using Rufus is also "forgetting" to change the Partition scheme and Target system options according to the system they are actually trying to boot.

In other words, if the issue you describe was real (Rufus needs to write bootloaders for some UEFI systems to boot), there would be a lot more people reporting it. On the other hand, what you describe certainly looks consistent with someone who is simply letting Rufus use the default options instead of changing them in the appropriate manner for the system they are trying to boot, which one should __always__ do for Windows, as it is not possible for Rufus to "guess" which of UEFI/UEFI-CSM/BIOS and Secure Boot/non-Secure Boot is enabled on your target system, which has a __direct__ impact on your ability to be able to boot the drive, and, if you don't properly understand why that is happening, you will of course be tempted to blame Rufus, whereas Rufus is simply respecting the options you selected for booting, which, if you selected pure UEFI boot for a BIOS system, will never work...

Hope that makes sense.

makedir, 2018-08-23:
@pbatard Read my posts again, maybe you will notice or understand the circumstances. I wont say it again. The information you misread or didnt understand correctly are in the post.

pbatard, 2018-08-23:
And please read what I am indicating about your circumstances being unlikely to be a Rufus issue if you are the only person in millions to report that issue. If, as you assert, the only consistent way to make an USB created by Rufus boot was to first use the MCT, I'd get hundreds of reports about this. Therefore, __logically__, there has to be another component to it.

I certainly have never seen an issue like the one you describe in all my years of obviously testing Windows bootable images on as many UEFI PCs as I can lay my hands on, so, even if I can understand that, because you have reached a specific conclusion, it may sound frustrating to see it being dismissed, what you describe as a consistent issue is simply not logically possible, and there has to be something else at play, even more so as you described yourself as "not an expert". For instance, have you ever tried the <kbd>Alt</kbd>-<kbd>E</kbd> mode as described in the FAQ?

ebonweaver, 2018-08-23:
@makedir it's pretty arrogant to say you're not an expert and then argue with the person who is.  It's clear that you're simply overlooking something in the configuration that you don't understand, and yet instead of trying to figure that out you're simply arguing that the product does not work, which it does.  I use it all the time, and outside of this original report that was due to very strange hybrid systems from Dell during a few month period, or cheap substandard flash drives, it has never failed.  Which is to say, Rufus itself has never been the point of failure.  It's also not the basic "you don't need to understand anything because it will do everything for you in a single set of circumstances tool" that you seem to be looking for, like the MS one.  It is an advanced tool that can be used in many situations, but that means you have to understand how to use it and what you're using it with.

Sorry, but your disrespect for the creator, who is extremely knowledgeable and helpful, is just uncalled for, especially since you're providing zero information to support your case, just anecdote.  To answer your original question, yes, we all know how to do it with just Rufus, you just don't seem interested in doing it.