pbatard/rufus #1005: In Dutch systems, if you choose `English`, then the `Yes` and `No` buttons in dialogs are still in Dutch

jpluimers, 2017-08-21:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [X] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
In dialogs, the buttons stay Dutch on a Dutch system, even if you choose English inside Rufus.

In addition, you get an error message when downloading the menu (I'm not sure if this occurs on an English system as well): `Unknown internet error 0x00002F0D`.

Note that on localised Windows 7 versions, you [can only install the English language pack on Windows 7 Ultimate and Enterprise](https://support.microsoft.com/en-us/help/972813/windows-7-language-packs-are-available-for-computers-that-are-running), so for instance on Professional or Home Premium editions, you cannot install the English language pack to force all native controls to be English. This renders this log entry kind of "moot": `No Windows Language Pack is installed for English (English) on this system.
This means that some controls may still be displayed using the system locale.
Localization set to 'en-US'`

Let me know if you want the internet error in a different issue (if it's not related to localisation:  It *might* be a certificate error as for me https://www.google.com/search?q=0x00002F0D%20site%3Amicrosoft.com returns https://support.microsoft.com/en-us/help/182888/how-to-handle-invalid-certificate-authority-error-with-wininet).

Steps to reproduce:

1. Switch your computer system to the `Dutch` language
2. Download https://rufus.akeo.ie/downloads/rufus-2.16p.exe
3. Run `rufus-2.16p.exe`
4. Choose the `English` language inside Rufus
5. Select image type `ISO image`
6. Select an image (in my case `VMware-VMvisor-Installer-201701001-4887370.x86_64.iso` from https://my.vmware.com/group/vmware/evalcenter?p=free-esxi6)
7. Press the `Start` button
8. You get a warning with English text, but Dutch buttons:
    [image] 

    ```
    ---------------------------
    Replace menu.c32?
    ---------------------------
    This ISO image seems to use an obsolete version of 'menu.c32'.
    Boot menus may not display properly because of this.
    
    A newer version can be downloaded by Rufus to fix this issue:
    - Choose 'Yes' to connect to the internet and download the file
    - Choose 'No' to leave the existing ISO file unmodified
    If you don't know what to do, you should select 'Yes'.
    
    Note: The new file will be downloaded in the current directory and once a 'menu.c32' exists there, it will be reused automatically.
    ---------------------------
    Ja Nee
    ---------------------------
    ```
9. If you press `Ja` (which corresponds to `Yes`), you get this error:
    [image]

    ```
    ---------------------------
    File download
    ---------------------------
    Unknown internet error 0x00002F0D
    ---------------------------
    OK
    ---------------------------
    ```

10. Press `OK` on the error, then `Start` in the menu and `Nee` at the menu file download.
11. Now you get another dialog with one button in English (`OK`)and one in Dutch (`Annuleren` which corresponds to English `Cancel`:

    ```
    ---------------------------
    Rufus
    ---------------------------
    WARNING: ALL DATA ON DEVICE 'NO_LABEL (F:) [4.1GB]' WILL BE DESTROYED.
    To continue with this operation, click OK. To quit click CANCEL.
[... 262 lines cut ...]
Created: F:autorun.ico

Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Using autorun.inf label for drive F: 'ESXI-6.5.0-20170104001-STANDARD'
Found USB 2.0 device 'Generic- Compact Flash USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- MS/MS-Pro/HG USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SD/MMC/MS/MSPRO USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SD/MMC USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Generic- SM/xD-Picture USB Device' (0BDA:0181)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 497, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0063AD02
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.8 GB (4088397824 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2017-08-21:
The dialog buttons are intentional (or at least as intentional as it gets with Windows programming), as localizing everything would mean replacing the `MessageBox` API calls of Windows everywhere, which always display their buttons in the default locale.

Considering that doing this would be a lot of work for very little benefit, since the vast majority of Rufus is expected to use Rufus in the current Windows locale, and if people use Rufus on a system that uses locale XYZ, then they are expected to understand button labels displayed in XYZ, I am not planning to do anything about this.

Now, the error when downloading `menu.c32`, I need to test, which I will do as soon as I get a chance.

For the record, because I suspect this might have to do with the Windows 7  default libraries having some issue with modern versions of SSL, can you tell me if you can access https://rufus.akeo.ie/files when using __Internet Explorer__ (It has to be the native Internet Explorer, other browsers will not be using the native SSL DLLs)? If not, then you may have to wait for Rufus 2.17, which I already set to work around of the issue of some older Windows systems having issues with modern https protocols, by using http for downloads. Anyway, I should have more insight about this issue once I have tried to replicate your issue.

jpluimers, 2017-08-21:
You get the below Dutch error, which in Englis is this one: https://support.microsoft.com/en-us/help/931850/-there-is-a-problem-with-this-website-s-security-certificate-when-you

```
Er is een probleem met het beveiligingscertificaat van deze website. 
 
   
 Het beveiligingscertificaat dat door deze website wordt gebruikt, is niet verleend door een vertrouwde certificeringsinstantie. 
```

The problem is that Windows 7 thinks the certificate is self signed: 
[image]

What API function do you use for download? Maybe I can reproduce with a small console app and help with a better workaround than plain http download (with all the man-in-the-middle hassle of contemporary providers).

pbatard, 2017-08-21:
> The problem is that Windows 7 thinks the certificate is self signed

Or, technically, the problem is that your copy of Windows 7 (for whatever reason) does not appear to have the "Globalsign Root CA" in your Trusted Certification Authorities. Could be because that certificate comes with a Windows Update, which you haven't applied (Trusted CA do evolve on a regular basis and must be kept up to date through OS updates, just like they are being kept up to date in Chrome or Firefox through browser updates), or maybe that's because your Sysadmin or yourself deleted that certificate from the Trusted CA store.

An up to date vanilla version of Windows 7 should trust "Globalsign Root CA", so it would look like your Windows 7 installation was made nonstandrad in one way or another.

> What API function do you use for download?

The standard Win32 Microsoft APIs to access internet content: https://github.com/pbatard/rufus/blob/master/src/net.c#L261-L322

> and help with a better workaround than plain http download (with all the man-in-the-middle hassle of contemporary providers).

Then you probably want to read [this](https://github.com/pbatard/rufus/wiki/Security#Download_of_SyslinuxGRUB_resources) and the next paragraph to understand why it is possible to have complete confidence that the `menu.c32` being downloaded by Rufus has not been tampered with in any way, even if this occurred through http rather than https, because the SHA-256 hash of the downloaded will be calculated and validated against the application internal database (and because the application is signed, you can also trust that the DB has not been altered).

The thing is, it is actually a lot more likely that hackers will be able to compromise my download website and insert a malicious `menu.c32`, which using https vs http won't give you any protection against, than having a mitm attack being conducted to insert a malicious download. So be very weary to equate https with trust. All HTTPS ensures is that the transmission will be done in a secure manner. However it does not guarantee that the download source has not been compromised, and, without additional measures, such as the ones Rufus uses, there's actually no way to ensure that a download isn't malicious, regardless of whether it came from a reputable server (no matter your security measures, you should consider that a server can always be breached, be it through zero day, spear fishing, or even through the age old method of physically threatening on of its administrator), and that https alone does nothing to protect you from malicious content...

Besides the http workaround is mostly aimed at being temporary, while I release the last ever version of Rufus that'll be compatible with XP and Vista (since XP can't access the downloads through https at all, no matter what). After 2.17, I will switch back to using https.

pbatard, 2017-08-22:
Confirmed I had no issue in a vanilla (and up to date) install of Windows 7 SP1:
```
Rufus version: 2.16.1170
Windows version: Windows 7 SP1 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found non-USB removable device 'VMware, VMware Virtual S SCSI Disk Device' => Eliminated
Found non-USB removable device 'VMware, VMware Virtual S SCSI Disk Device' => Eliminated
Found USB 2.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'd-live 9.1.0 xfce amd64'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3818, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x009D2E48
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.2 GB (31405899776 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux.cfg' for Syslinux
  Detected Syslinux version: 3.86 (from '/isolinux.bin')
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESXI-6.5.0-20170104001-STANDARD'
  Size: 328.2 MB (Projected)
  Uses: Syslinux/Isolinux v3.86
    With an old menu.c32
  Uses: EFI
Using image: VMware-VMvisor-Installer-201701001-4887370.x86_64.iso
Downloading 'menu.c32' from https://rufus.akeo.ie/files/syslinux-4.07/menu.c32
File length: 55140 bytes
Successfully downloaded 'menu.c32'

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE2 for exclusive write access
Requesting lock...
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{9c84f822-872c-11e7-b10c-000c29b29968}\
Installing Syslinux 3.86...
Opened \\?\Volume{9c84f822-872c-11e7-b10c-000c29b29968} for shared write access
[... 166 lines cut ...]
Extracting: F:\weaselin.t00 (1.9 MB)
Extracting: F:\xhci_xhc.v00 (73.6 KB)
Extracting: F:\xorg.v00 (1.3 MB)
Created: F:\syslinux.cfg
Finalizing, please wait...
Created: F:autorun.inf
Created: F:autorun.ico

Found non-USB removable device 'VMware, VMware Virtual S SCSI Disk Device' => Eliminated
Found non-USB removable device 'VMware, VMware Virtual S SCSI Disk Device' => Eliminated
Found USB 2.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'ESXI-6.5.0-20170104001-STANDARD'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3818, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0004E417
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.2 GB (31405899776 bytes)
  Start Sector: 2048, Boot: Yes
```

Can you confirm that your Windows 7 system is up to date?

marcosfrm, 2017-08-22:
Can this be related?

https://security.stackexchange.com/questions/148402/should-antivirus-https-scanning-be-left-on-is-it-secure

pbatard, 2017-08-22:
> Can this be related?

Possibly. Security applications do run elevated and have therefore the possibility to block connections that they don't like. However, I doubt that this is the case here, as the security solution would tell you more explicitly that access to the server was deliberately blocked, as opposed to a message indicating that Windows isn't trusting the GlobalSign Root CA. At any rate, if you have a security solution active, you may try to temporarily disable it to see what happens.

jpluimers, 2017-08-23:
@pbatard: Thanks for the very elaborate explanation. The certificates are indeed wrong so I will re-install that machine as soon as I've a day or two to spare. Pending time to re-install Windows 7 Dutch, I'm closing this issue.

If a re-install causes the same failure, I will re-open the issue, but I'm pretty convinced you're right.

On a site note: why use the A variants of the API calls? I was expecting the W functions.

@marcosfrm: it's not related as the machine does only has Windows Defender installed, but thanks for the pointer as it's a very interesting question.

pbatard, 2017-08-23:
> If a re-install causes the same failure, I will re-open the issue, but I'm pretty convinced you're right.

Sounds good to me.

> On a site note: why use the A variants of the API calls? I was expecting the W functions.

I'm using 'A' because Unicode is not used for any of the parameters that are being passed to these functions. These calls are only going to fill a buffer in memory from an URL as well as options that I can guarantee to always use plain ASCII. On the other hand, the function calls to create and save the file do handle Unicode for the requested path (except I'm using my own 'U' version rather than the 'W' from Microsoft, as I consider that Microsoft completely dropped the ball by not providing UTF-8 APIs, and I am remedying this in my own [`msapi_utf8.h`](https://github.com/pbatard/rufus/blob/master/src/msapi_utf8.h)). [UTF-8 FTW](http://utf8everywhere.org/)!

pbatard, 2017-08-23:
> Do the ...A editions of the WinINET functions understand UTF_8 based URLs?

You mean [punycode](https://en.wikipedia.org/wiki/Punycode)? Of course they do, since it's ASCII.
Besides, in case you haven't noticed, none of the URLs being used in Rufus come from anything but hardcoded data, so of course I know that non-ASCII will never be needed. If were to write generic function calls for all the subroutines I code, I'd be writing an OS. Rufus is not an OS, so I only need subroutines that perform the task I need them for, nothing more. Being able to access Unicode URLs outside of Punycode, or even using Punycode is not one of these tasks, so I logically choose not to care one bit about this.

jpluimers, 2017-08-24:
@pbatard I meant calls like `HttpOpenRequestA` so it's clear to me know that in order to use them with Unicode URLs, you need to use [punycode](https://en.wikipedia.org/wiki/Punycode) encoding.

This was all out of curiosity, as I did understand the URLs were hard coded so can be guaranteed to not contain Unicode.

pbatard, 2017-08-29:
I don't think there's anything that needs to be done on this issue, so I will close it.

closed 2017-08-29 by pbatard