pbatard/rufus #1006: Error while partitioning

TherbianGranbat, 2017-08-22:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Error while partitioning
-----------------

I was trying to make a bootable USB pen for Ubuntu for my new PC (making this on my old PC). However, in the process of making the USB-pen, it failed and I got the following log. Now, my whole interface keeps "flashing" while the USB-drive is connected. This is the 2nd Verbatim USB drive making the exact same error (I bought them brand new from a regular PC-retailer in Denmark and went back to the shop to tell them it was bugged, they gave me a new one - but now that it happened twice, I'm not sure anymore). 
As you can see, it keeps repeating the same error. I can see the USB pen in Rufus still though. However, I am unable to check for bad blocks as it simple gives an "Error while partitioning". I'm quite inexperienced with Github but I tried my best to search for the error so please bear with me if someone already had this issue. Currently, I am unable to use the USB drive for anything and receive an error when I attempt to format it. 

My system is Danish, so if anything needs to be translated, please just ask and I'll do that.

Log
---
```
Rufus version: 2.16.1170
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0406
Will use default UI locale 0x0406
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'da-DK'
NOTE: The Danish (Dansk) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
0 devices found
0 devices found
Checking for Rufus updates...
Checking release channel...
0 devices found
No new release version found.
Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0243)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1912, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x5F3FC761
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15728615424 bytes)
  Start Sector: 48, Boot: Yes
Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0243)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1912, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x5F3FC761
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.6 GB (15728615424 bytes)
  Start Sector: 48, Boot: Yes
Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0243)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1912, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x5F3FC761
[... 662 lines cut ...]
Drive does not have an x86 Master Boot Record
Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0243)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1912, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
read_sectors: Read error [0x00000015] Enheden er ikke klar.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have an x86 Master Boot Record
Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0243)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1912, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
read_sectors: Read error [0x00000015] Enheden er ikke klar.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have an x86 Master Boot Record

```

pbatard, 2017-08-22:
Thanks for the report. I'm afraid that a write error during the copying of files is usually indicative of a hardware error, most likely defective flash memory. I understand that you tested 2 drives of the same model, and got the same error, which could be explained by a bad batch of drives or other factors (such as the retailer having gotten "fake" drives from its supplier without their knowledge).

What I would suggest is this: 

* If the drive is still being recognized in Windows, please try to run a bad blocks check. This will most likely indicate an issue, and you can take drive back to the manufacturer and tell them that it was detected as defective. This may also tell you if the drive you have is a "fake" drive (i.e. a drive that was modified to report more memory than it actually has, e.g. 16 GB instead of 1 GB).
* If the drive is not being recognized in Windows, you should try to get it exchanged again, and maybe (since sings seem to point to a bad batch, or fake drives), try to get a different model/brand. Then, before you do anything with the new drive, you should run a bad blocks check to ensure that the drive has no defects.

Note that you don't have to use Rufus to perform a bad blocks check. You can use any other tool that provides the same functionality (though most of the ones I know, such as `badblocks` from `e2fsprogs` would be on Linux rather than Windows, and I don't think they have "fake" drive detection). Or you can also try copying a bunch of large files until you fill the drives, and then verify that the MD5 or SHA of the files that were copied to the USB matches the one from the file you have on your HDD. Applications like 7-zip can provide MD5 and SHA computation through a drop down menu in Windows explorer. Still, since I am 100% positive that Rufus is not the source of your defects (see [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive) for more details), I would advise you to use the bad blocks check from Rufus to validate any new drive you plan to use, to make sure that it doesn't come with a factory defect, as is likely to be your case here.

TherbianGranbat, 2017-08-23:
Thanks for the reply. I'll try to find a bad blocks checker. My own hypothesis is that I was wrong to use the frontpanel USB ports as they're only USB 2.0 and the drive was 3.0 and furthermore farther away from the motherboard. The USB pens seemed normal before I started writing to them.

pbatard, 2017-08-23:
> I'll try to find a bad blocks checker.

Just to be clear, Rufus __can__ perform bad blocks check (you just have to click the "Check device for bad blocks" box and uncheck the "Create bootable disk using..." one). I'm just mentioning that you can use another bad blocks checking application if you somehow think that Rufus is the issue.

> My own hypothesis is that I was wrong to use the frontpanel USB ports as they're only USB 2.0 and the drive was 3.0 and furthermore farther away from the motherboard. The USB pens seemed normal before I started writing to them.

I don't think your device is USB 3.0. Rufus is able to detect USB 3.0 devices even when they are plugged on an USB 2.0 port, and will always indicate if that is the case. Also, I looked at the specs for Verbatim STORE N GO USB drives, and it seems these devices are USB 2.0. As per the Rufus log, your device is seen as:
```
Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0243)
```
If it was an USB 3.0 device, you would see something and additional line like:
```
Found USB 2.0 device 'SanDisk Extreme USB Device' (0781:5580)
NOTE: This device is an USB 3.0 device operating at lower speed...
```

TherbianGranbat, 2017-08-23:
Then it is almost certainly an issue with the USB pen. The wrapping specifically said USB 3.0.

TherbianGranbat, 2017-08-25:
I will not have time during the next 2-3 days to test the USB pen but I'll try to get around it.

TherbianGranbat, 2017-08-29:
I still haven't had time to test it, sorry. If you want to close this, you may. I will not be able to test it within the next 7 days.

pbatard, 2017-08-29:
Okay, I will close it for now. But you can still comment on it once you've had a chance to run your test.

closed 2017-08-29 by pbatard