pbatard/rufus #1007: Rufus silently fails to write an ISO containing large files in file mode w/ FAT32

sqweek, 2017-08-23:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

I used Rufus to write a 7.1 GB .iso to USB (an archlinux derivative). The UI is nice and simple so thank you and props for no installation required :). Anyway, I used mostly default/recommended settings which meant FAT32 and the file-based mode rather than direct-disk mode.

Unfortunately it turns out that the ISO was predominantly comprised of a single 6.9 GB .sfs file, which is of course too large to store on a FAT32 partition. Rufus however happily ignored this problem and just copied ~3.5 GB of the .sfs across. There were no errors or warnings (or at least none prominent) -- it looked like the process had succeeded and it wasn't until attempting to boot from the USB that I noticed something was wrong.

I'm rewriting the USB now in direct-disk mode which I expect will solve my problem so no big deal, it would just save some time to catch this scenario earlier :)

Log
---

I lied -- I didn't save the log of the run on account of I thought it succeeded!

pbatard, 2017-08-23:
I'm very surprised by your report. Rufus __does__ detect > 4 GB files residing on an ISO, and will take measures to work with them. As a result, I expect that Rufus switched to using NTFS instead of FAT32 for the file system, but you haven't noticed (of course, having a log would tell this very explicitly, and nothing prevented you from trying the same thing twice to get a log).

Now the issue is that while some Linux distros (Debian and IIRC Arch) have no issues with NTFS as the file system used on a bootable USB Flash Drive, there's a minimum level of support that needs to be added to the kernel and the init script for this to work, which your Arch derivative might not follow).

Now, if you could give me a link to the ISO you downloaded I could definitely __test__ and see what's up. So can you please provide that information, especially if you don't have a log to provide. Makes it very difficult to treat your request as a relevant one otherwise, as I __must__ confirm an issue before I can address it. Thank you.

sqweek, 2017-08-24:
OK, interesting. The result was definitely a FAT32 usb stick, as my first attempt to fix the problem was to manually copy the full 6.9 GB .sfs file across, which failed because it was too large for the filesystem. I didn't have time to retry the burn last night but I'll give it a shot now to get a log and see if the problem is reproducible...

Yep, it is. The log suggests Rufus has copied the first 4.0 GB of the sfs across, then overwritten it with the last 2.9 GB? xD In case it's something particular to the iso, it's from https://blackarch.org/downloads.html#iso-download  (BlackArch Linux 64 bit Live ISO, 2017.06.14).

```
Rufus version: 2.16.1170
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0809
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 942, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x28A85F9F
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 64 MB (67108864 bytes)
  Start Sector: 164, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'BLACKARCH_201706'
  Size: 7.1 GB (Projected)
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
Using image: blackarchlinux-live-2017.06.14-x86_64.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/' for Syslinux installation

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{8af8524e-144d-11e7-83de-2816ad2a662d}\
Installing Syslinux 6.03...
Opened \\?\Volume{8af8524e-144d-11e7-83de-2816ad2a662d} for shared write access
Using existing './rufus_files/syslinux-6.03/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.03/ldlinux.bss' ✓
[... 129 lines cut ...]
Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
Using autorun.inf label for drive D: 'BLACKARCH_201706'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 942, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x035ED7C1
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.2 GB (7747174400 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
Using autorun.inf label for drive D: 'BLACKARCH_201706'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 942, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x035ED7C1
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.2 GB (7747174400 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2017-08-24:
Thanks for the log. This looks weird, and I'm thinking the issue doesn't come from Rufus, but from the way the ISO was mastered, as the BlackArch people seemed to have split the SFS in 2 parts that use the same name. As such I'm think that ISO9660 file system shenanigans are to blame rather than Rufus. of course I'm currently downloading the ISO to have a closer look at its content, but, from having worked and tested ISOs that do contain > 4GB file, the behaviour being displayed here seems just too weird not to come from ISO-9660 filesystem abuse.

As I was mentioning, Rufus does detect > 4GB files when processing an ISO, and should have displayed an `Has a >4GB file` notice like:
```
ISO label: 'SSS_X64FRE_EN-US_DV9'
  Size: 5.3 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: en_windows_server_2016_x64_dvd_9327751.iso
```
The fact that it didn't, as well as the following two lines from the log tells me that the people who created the ISO used some kind of trickery to split the file in 2 while using the same name for the 2 parts:
```
Extracting: D:\blackarch\x86_64\airootfs.sfs (4.0 GB)
Extracting: D:\blackarch\x86_64\airootfs.sfs (2.9 GB)
```

Anyway, I'll know more after I have a closer look at your ISO.

sqweek, 2017-08-24:
Gotcha. FWIW opening the ISO in 7zip shows a single file of 7,445,232,000 bytes:

[image]

The properties pane of airootfs.sfs suggests that something like you describe may be happening at the ISO9660 level though (I note "Files: 2"):

[image]

pbatard, 2017-08-24:
I got the ISO now, and can confirm that, as far as 7-zip is concerned, `airootfs.sfs` is a single file of ~7 GB, but libcdio (the library that Rufus uses for ISO processing) sees it as two separate files:
```
/d/libcdio/example # ./isolist /d/ISOs/blackarchlinux-live-2017.06.14-x86_64.iso
Application: ARCH LINUX LIVE/RESCUE CD
Preparer   : PREPARED BY MKARCHISO
Publisher  : ARCH LINUX <HTTP://WWW.ARCHLINUX.ORG>
Volume     : BLACKARCH_201706
d [LSN     30]     2048 /.
d [LSN     21]     2048 /..
- [LSN  55927]       47 /airootfs.md5
- [LSN  55928] 4294965248 /airootfs.sfs
- [LSN 2153079] 3160266752 /airootfs.sfs
```

Looks like this could be a libcdio bug. I'll get in touch with the libcdio mailing list to find out what they think the issue is, and hopefully get a fix, that I can then integrate in Rufus. It may be a little while before I am able to do so though...

pbatard, 2017-09-11:
While I'm still waiting for libcdio to fix this issue (though, by the looks of it, I may have to submit a fix myself), I have now added a warning about multi-extent ISO-9660 files in Rufus 2.17.

commit 44c9cb0b (2017-09-14): [iso] add support for ISO9660 multiextent files

* Closes #1007
closed 2017-09-14 by pbatard