pbatard/rufus #1013: Problem Conflict with Eset boot.

Alexander1414x, 2017-09-01:
Hello Rufus! I have problem boot for Eset SysRescue https://www.eset.com/int/support/sysrescue/ -this download ISO 478 mb , and to write USB 32 gb . Whis write is seccusiful , and to restart -this problem boot , this black in the ALT+TAB , this root  and error ubuntu engine. 
Please test this.

pbatard, 2017-09-04:
Short answer:
Edit `\isolinux\txt.cfg` on the USB drive and change all instances of `eSysRescueLiveCD` to `ESYSRESCUEL`.

Long answer: Will come later...

pbatard, 2017-09-05:
Long answer:
The ESET SysRescue Live init scripts were not designed to work with a FAT32 file system.
The problem is that, as per `\isolinux\txt.cfg`, the init system is being told to look for a disk labelled `eSysRescueLiveCD` when looking for the live media. This works well for an ISO9660 file system, that doesn't have much limitation on how the disc can be labelled, but when converted to FAT32 for USB boot, the disk label must follow the FAT restrictions, which force it to be 11 UPPERCASE character maximum.

As there is no way to work around this file system limitation, the USB will internally be labelled `ESYSRESCUEL` (which is 11 UPPERCASE characters), and this will be the name that will appear on the initial Linux system (the one that looks for the medium before continuing the boot process) under `/dev/disk/by-label/`.

Hence, the issue is that, if the `\isolinux\txt.cfg` is left unmodified, the ESET initial boot environment will try to mount the disk `/dev/disk/by-label/eSysRescueLiveCD` whereas, due to the FAT32 file system conversion, the disk will be `/dev/disk/by-label/ESYSRESCUEL`.

Ideally, as __many other Linux distribution maintainers already do__ (e.g. Arch Linux, who use `ARCH_YYYYMM` for their disc labels) the ESET people should realize that some people will convert the ISO media to FAT32, and therefore that they need to use a label that is FAT32 friendly. So the best solution would be to log an issue with them to let them know.

Otherwise, Rufus does attempt to modify the label in the SysLinux/IsoLinux config files it detects, but, since a config file can include a config file which can include another config file (as is the case here), and there's only so much logic I want to add, there's a limit to what can be achieved at a __reasonable__ development cost (i.e. I cannot spend my days adding logic that will parse all possible Syslinux/Isolinux included files to replace labels, especially when I expect that any competent distro maintainer should be well aware of the ISO &rarr; FAT label conversion for it and will take action to ensure that their system can boot from FAT32).

I may still update the logic so that it parses all `.cfg` files that reside under a `syslinux` or `isolinux` directory, but I can't promise anything at this stage.

Another workaround, provided that the distro maintainers are using a recent version of Syslinux (v6.x or later) and provide NTFS support in the kernel, would be to create the media as NTFS instead of FAT32 in Rufus. Unfortunately however, the ESET people are still using a rather old version of Syslinux (v4.05), so that can't be applied here.

commit 63eb747c (2017-11-01): [iso] make the lookup for updatable .cfg file more generic

* For ESET and other ISOs, such as Arch derivatives
* Closes #1013
closed 2017-11-01 by pbatard