pbatard/rufus #1014: Error: Undetermined error while formatting using rufus 2.16.1170

justHafid, 2017-09-03:
Checklist
---------
- [ x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ x ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ x ] Plugging the USB into a different port
  - [ x ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi, I just bought new 4GB USB drive for Windows USB Installer, but have a trouble couple of times while I try to create the installer using rufus.

I need some help to informing me what's going on with my attempt.

Log
---
```
Rufus version: 2.16.1170
Windows version: Windows 8 64-bit (Build 9200)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Flash USB Disk USB Device' (1221:3234)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 509, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x003B2DB5
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.9 GB (4193255424 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Checking txtsetup.sif:
  OsLoadOptions = /fastdetect /noguiboot /nodebug
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'VRMPOEM_EN'
  Size: 427.5 MB (Projected)
  Uses: WinPE
Using image: XPOEM.iso

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'H:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Flash USB Disk USB Device' (1221:3234)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 509, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x003F1F92
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 3.9 GB (4193255424 bytes)
  Start Sector: 2048, Boot: Yes

```

pbatard, 2017-09-04:
Okay, the first thing you nee to know is that, if you are trying to install XP (as is the case is, since you are using an XP ISO), you're pretty much on your own, as I have ceased to support XP a long time ago.

Now, given that this is a formatting issue, chances are that it may happen regardless of the ISO. However, `Undetermined error while formatting.` doesn't come from Rufus but from Windows so my question are:

1. What happens if you try to repartition and reformat your drive using Windows. Obviously, if you get the same error, this is not a Rufus issue
2. Do you see the same problem when using a different USB Flash Drive?

justHafid, 2017-09-04:
First of all, thank you for your respond.

Well, this is my answer.
1. I'm using `diskpart` and it done with no error.
2. Yes, I try to another 4GB USB drive, and end up with same problem.

I'm trying to install XP because some of my client's laptop still use it (out of date). And, for myself I am trying to create a bootable Windows 8.1 Pro (lite). For now my pc running on Windows 8 Pro.

I'll try another attempt with another USB drive. Will informing you if I still in trouble.

pbatard, 2017-09-04:
> I'm using diskpart and it done with no error.

After you use `diskpart`, can you format the drive by right clicking on it in the Windows file explorer and choosing `Format...`. Considering that Rufus was able to partition the drive, if you only used `diskpart` to repartition the drive, it makes sense that it would succeed too. And if you have been using the `diskpart` commands to reformat as well, I would still encourage you to try the Windows UI formatting, as it should behave the same as Rufus, and I would like to know if you actually see a difference there.

justHafid, 2017-09-06:
Yes, I still can format it by right click on file explorer. By the way, I get some conclusion, the usb drive cannot format to ntfs, it stuck with fat32, maybe thats the problem comes. So, I pick another usb, and its cool now. Thank you for your assistance.

pbatard, 2017-09-06:
> Yes, I still can format it by right click on file explorer

Interesting. The exact same method of formatting a drive should be used by Rufus and file explorer, so that is very strange.

> By the way, I get some conclusion, the usb drive cannot format to ntfs, it stuck with fat32, maybe thats the problem comes.

Does that mean that only FAT32 worked with file explorer and not NTFS? If that's the case then it makes sense that same behaviour is being observed in Rufus.

> So, I pick another usb, and its cool now. 

Okay. So that means there's definitely an environmental component to the issue as well that as to do with the flash drive being used.

justHafid, 2017-09-08:
> Does that mean that only FAT32 worked with file explorer and not NTFS? If that's the case then it makes sense that same behaviour is being observed in Rufus.

I've been tried formatting another 4GB USB, they won't formatted on NTFS as well. by right click or cmd `fs=NTFS` to me it's lil bit strange, because before I upgrade to Windows 8.1, everytime I create a USB installer, it's done well.

pbatard, 2017-09-08:
> they won't formatted on NTFS as well. by right click or cmd `fs=NTFS`

Okay. That confirms it's not a Rufus issue then.
Unfortunately, I don't think there's any advice I can provide, besides trying to use yet another flash drive, or trying a different port, which you already did.

Since this is not a Rufus problem, I'm gonna close this issue.

closed 2017-09-08 by pbatard