pbatard/rufus #1015: Rufus corrupts pendrive.

willbr10, 2017-09-04:
Whenever I use Rufus, the pendrive is damaged and can not be formatted again. :( I lost 4 pendrives with Rufus.

pbatard, 2017-09-04:
Please provide logs. Also, did you run a bad blocks check __before__ using any of these drives (so that you can have some confidence that you aren't simply gotten a batch of fake drives).

I would expect that, by the 4th time you tried, you would try to keep the log to see what happens, or, at the very least, run a bad blocks check.

Going by your logless report, it's hard to figure out if the drives are actually damaged, or if it's just a different issue (such as that it is possible to use ISOs, such as the official Chromium ones, that will make a drive look as if it cannot be reformatted on Windows, even though you'll have no trouble recovering it using Linux). And I also get no information whatsoever about the manufacturer and model of your drives.

You may also want to read [this](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive). As much as you may want to believe otherwise, because your drives appear to have failed while you were using Rufus, it's just not possible for an application like Rufus to damage flash drives.

Joseph22a, 2017-09-08:
.
 willbr10
“the pendrive is damaged and can not be formatted again.”

Consider trying DBAN 2.3.0 
DBAN = Darik’s Boot and Nuke
dban-2.3.0_i586.iso
15.9 MB (16,719,872 bytes) 

1.	Use DBAN to test if DBAN can see your pendrive. 

If DBAN cannot see your pendrive in the 
DBAN drive list then that fact gives more evidence your pendrive is damaged.  
(While in DBAN first screen Press ENTER, wait 4 minutes while DBAN loads.  
Do not use AUTONUKE command because all drives get erased.)  

2.	If DBAN can see your drive. 
Select one drive, your pendrive.  
Consider using DBAN to wipe your pendrive.  
All data is lost, wiped, shredded, nuked.  
I found my USB drive = pendrive, was faster after DBAN quick wipe 1 pass. 

3.	I have had success-using DBAN quick wipe 1 pass versus 
using DBAN default DOD wipe of 3 passes.  
This means after DBAN 2.3.0 quick wipe 1 pass on a drive 
like a HDD or 
pendrive (usb drive) then 
Zero (0) files found using 
Recuva 1.53.1087 Quick scan or 
Recuva 1.53.1087 Deep scan. 

4.	But, one time DBAN quick wipe 1 pass on a HDD (Hard Disk Drive) did not work.  
HDD did not format.
So, DBAN DoD 3 passes did work on that drive 
the HDD was then 
initialized in Windows 7 and 
MBR chosen and 
then formatted in Windows 7.  
And then HDD drive worked fine as a backup drive after DBAN DoD 3 passes.  

In short:
DBAN might help determine if “the pendrive is damaged” and 
DBAN might help with problem of “can not be formatted again”
Be careful using DBAN because all data is lost, wiped.  

--

pbatard, 2017-09-11:
Since there was no update from OP, I will close this issue.

closed 2017-09-11 by pbatard

MrAppAndCrap, 2017-09-11:
Never had this issue myself.

willbr10, 2017-09-12:
Thank you all for your help.

The problem must be myself.

Maybe I'm too dumb to use this software.

Once again thank you very much for the help.

I am sorry my english skills are bad

pbatard, 2017-09-12:
No worries. There are plenty of ways one can think USB Flash Drives have become defective, that have nothing to do with Rufus. For instance, if you use some ISOs of Chromium, you will not be able to repartition or reformat the drive in Windows (have to use Linux or another OS), and that is a Windows bug. And there are also Flash Drives that are known to have buggy firmware (such as the PNY Retract USBs). And that's not counting the various "fake" drive that unscrupulous people will sell to unaware buyers on the Internet, for a cheap price, that will become corrupted as soon as you go over the real capacity.

Without additional precision with regards to the model and manufacturer of the drive you use, what the bad blocks check reported, and especially the log from Rufus at the time of failure, and especially considering that, despite what people tend to think, [it's just impossible for Rufus on its own to corrupt pen drives](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive), I have to assume that whatever "corruption" you think you experienced has nothing to do with Rufus.

PS: I deleted your duplicate message, as you posted the same thing twice.

willbr10, 2017-09-12:
I understand.

I had not thought of this possibility.

Grateful.

Rufus is the best anyway.