pbatard/rufus #1016: Cannot mount GUID volume

sdled, 2017-09-05:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ x] I also tried one or more of the following:
  - [x ] Using a different USB drive
  - [x ] Plugging the USB into a different port
  - [x ] Running Rufus on a different computer
- [x ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
 as per description in Cannot mount GUID volume #476 the problem in my cannot mount condition was due to power iso virtual drive was using drive letter assigned to usb device by rufus...eg... usb drive A changed to drive Z by rufus at which time drive Z was assigned by power iso virtual drive...thus by changing poweriso virtual drive to 2 drives, instead of 4 drives, opened up the z drive for rufus and all is well, but why would rufus change the mount location of usb on drive A to drive Z

Log
---
```
Rufus version: 2.16.1170
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'SD/MMC Card  Reader USB Device' (058F:6335)
Found USB 2.0 device 'USB 3.0 Plugable SATA U3 USB Device' (1759:5000)
Device eliminated because it was detected as a Hard Drive (score 21 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 2.0 device 'USB 3.0 Plugable SATA U3 USB Device' (1759:5000)
Device eliminated because it was detected as a Hard Drive (score 21 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1936, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x008081DE
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.8 GB (15930490880 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'Win7'
  Size: 5.5 GB (Projected)
  Note: File on disk is larger than reported ISO size by 534 KB...
  Has a >4GB file
  Uses: EFI (win7_x64)
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: windows 7.iso

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE4 for exclusive write access
Requesting lock...
No drive letter was assigned...
Will use 'Z:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{0ce5ae4e-88be-11e7-a294-f4f26d000963}\
Could not remount \\?\Volume{0ce5ae4e-88be-11e7-a294-f4f26d000963}\ on Z:\: [0x000000B7] Cannot create a file when that file already exists.

Found USB 2.0 device 'SD/MMC Card  Reader USB Device' (058F:6335)
Found USB 2.0 device 'USB 3.0 Plugable SATA U3 USB Device' (1759:5000)
Device eliminated because it was detected as a Hard Drive (score 21 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 2.0 device 'USB 3.0 Plugable SATA U3 USB Device' (1759:5000)
Device eliminated because it was detected as a Hard Drive (score 21 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1936, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00855DBB
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.8 GB (15930490880 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE4 for exclusive write access
Requesting lock...
No drive letter was assigned...
Will use 'Z:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{0ce5ae4e-88be-11e7-a294-f4f26d000963}\
Waiting for access on \\?\Volume{0ce5ae4e-88be-11e7-a294-f4f26d000963} [\Device\HarddiskVolume20]...
Warning: Could not obtain exclusive rights. Retrying with write sharing enabled...
NOTE: Could not identify the process(es) or service(s) accessing \Device\HarddiskVolume20
Opened \\?\Volume{0ce5ae4e-88be-11e7-a294-f4f26d000963} for shared write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted \\?\Volume{0ce5ae4e-88be-11e7-a294-f4f26d000963}\ on Z:
Copying ISO files...
Extracting files...
Image is an UDF image
```

pbatard, 2017-09-05:
Thanks for the report.

You're actually the first person I can remember who reports that they are using `A:` to map an USB Flash Drive. My understanding was that `A:` and `B:` were reserved on Windows systems (as a legacy from the time computer had floppy disks), and therefore I designed Rufus around this restriction, but I've tried assigning a flash drive to `A:` on Window 10, and found that it works.

Obviously then, I need to review the assumption that `A:` and `B:` can not be used, so that's one fix to apply.

So now, we can explain the behaviour:

```
No drive letter was assigned...
```
Means Rufus was not able to detect the drive letter assigned to your drive (because it only checks drive letters starting at `C:`)
```
Will use 'Z:' as volume mountpoint
```
Means that. because was not able to detect a drive letter assigned (because it it didn't pick `A:`), it thinks it will need to assign a new one. And looking more closely at the code that does drive letter assignation, I found that there is a bug in that we may try to use the `Z:` drive to re-mount the USB Flash Drive even if that drive is in use (note however that this doesn't mean that Rufus could erroneously format or partition that drive by mistake, as we don't rely on the drive letter for these operations, and there are additional safeties). So what happened is that, because you appear to be using all 26 drive letters, which is another situation you're the first one to report, Rufus wasn't able to find a mount letter before `Z:`, and because it didn't properly check if `Z:` was actually in use, it erroneously tried to use that letter.

I think I have a good idea of how to fix these 2 issues. 
Can you please download rufus-2.17_TEST5.exe from [here](https://rufus.akeo.ie/testing/rufus-2.17_TEST5.exe) and let me know if that works better?

pbatard, 2017-09-06:
I'm pretty sure the problem is now fixed, so this issue will be automatically closed as I push the next commit.
If that is not the case, feel free to update it.

closed 2017-09-06 by pbatard

sdled, 2017-09-10:
Lol for real….i always thought that window would assign double letter after the singles w was used up, but that’s not the case due to I could not mount 4 virtual drives with poweriso, no more letters I guess, everything worked fine after I drop 2 of the virtual drives and allowed rufus to mount on Z, windows assigned drive A to the usb flash drive.
Yea I hate win 10 it sure is pretty but it has wreaked havoc on my network, I spent weeks researching it and fixing win 10  for MS and they didn’t pay me crap….lol I sent them a bill for 23567$ my time spent fixing it. So I was goning back to win7 and dual booting linux Ubuntu, trying to get away from MS completely. As for Rufus it worked fine loading my ntlite slipstream win7, booted perfect thru usb boot and loaded win7

Dl test rufus will check it out soon or later…lol…”TIME IF ONLY I HAD MORE”

Thank you
sdled

pbatard, 2018-12-23:
@Claus1 

Are you using `A:` or `B:` as your drive mount point and also getting this error?
```
Could not remount \\?\Volume{0ce5ae4e-88be-11e7-a294-f4f26d000963}\ on Z:\: [0x000000B7] Cannot create a file when that file already exists.
```

If not, then this is __not__ the same issue and you shouldn't piggy back on a closed issue, that was opened for a completely different problem, as it is impossible for me to help you since the problem described here was indeed fixed, and of course I can't "guess" what your actual issue is.

At the very least you need to provide a full log from Rufus.

Claus1, 2018-12-23:
@pbatard  I tried the tool on win 7. after several attempts with errors iso on flash was created.  May be my win 7 issue.

pbatard, 2018-12-23:
Again, can't do __anything__ unless you provide a full log with the error.

Claus1, 2018-12-24:
I did  not copy the error log.

commit 4a213c3b (2017-09-06): [core] allow the use of A: and B: as drive letters and fix in-use detection for Z:

* Closes #1016
* Also expand support for Arch Linux derivatives Syslinux config files