pbatard/rufus #1017: USB drive is not booting to install Windows 7.  (Rufus 2.16.1170 Portable)

Joseph22a, 2017-09-08:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

Friday-08-September-2017
Problem:  
USB drive is not booting to install Windows 7.  (Rufus 2.16.1170 Portable)  
The boot process hangs (stops).  
The message were the boot hangs is "Windows is loading files…"
While showing above message, the USB light continues to Blink (On, Off, On)  

Details: 
1.	Rufus 2.16.1170 Portable was used to put ISO data on USB drive.  

2.	The ISO is:
Win7_Pro_SP1_English_COEM_x64.iso  
3.09 GB (3,320,903,680 bytes)  
the above ISO is from web site  
https://www.microsoft.com/en-us/software-download/windows7

3.	The computer works Ok, because the computer boots fine from 
CDROM drive with Windows 7 CD, and this CD installs Windows 7 correctly.  

4.	Log file done.  

Some variables are: 
- USB drive (tried 2 drives: 16 GB and then 128 GB)  
- ISO, see above 
- Rufus 2.16.1170 Portable 

Questions:  
What is the corrective action?  
Since computer is working fine with CDROM boot, what steps to take to isolate problem, so computer boots from USB?  
What steps to take to Divide and Conquer the problem?  

How to get Windows 7 to install from a USB drive?  

--  

Log
Rufus version: 2.16.1170 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A838)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 15566, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00341C55
Drive has a Rufus Master Boot Record
Partition 1:
[... 949 lines cut ...]

 125033471 KB total disk space.
   3282412 KB in 882 files.
       376 KB in 210 indexes.
     71047 KB in use by the system.
     65536 KB occupied by the log file.
 121679636 KB available on disk.
      4096 bytes in each allocation unit.
  31258367 total allocation units on disk.
  30419909 allocation units available on disk.
NTFS Fixup completed.

Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A838)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 15566, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0001585C
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 119.2 GB (128034275328 bytes)
  Start Sector: 2048, Boot: Yes

---

pbatard, 2017-09-08:
Most likely, the issue is with missing USB 3.0 or other modern disk drivers. While the DVD-ROM version of Windows may not need those, when you are installing Windows 7 from USB on a modern computer, you should bear in mind that, for instance, Windows 7 has no native support for USB 3.0, so if you are booting from an USB 3.0 port, the installation process will fail.

To make it work, you may have to go to your computer's manufacturer website and see if they provide the drivers you need for Windows 7 installation, as well as a method of injecting them into the bootable USB. I would also encourage you to contact your manufacturer, as I doubt this is a Rufus issue and there's very little advice I can provide to help when this issue is hardware specific.

But my first recommendation, to validate that this is indeed an issue with obsolete or non-existing Windows drivers would be to try with Windows 10. If Windows 10 doesn't produce the same issue, then you'll have your answer.

And yes, contrary to what many people like to believe, there does exist valid reasons for Microsoft to release new versions of Windows, and these have to do with being able to support installation on recent hardware. If you plan on sticking with Windows 7, you should be mindful that, just as you have found, the installation may not work at all on recent hardware.

Joseph22a, 2017-09-08:
Friday-08-September-2017, follow up: 

Windows 10 Single Language English was downloaded from
https://www.microsoft.com/en-us/software-download/windows10ISO 
Win10_1703_SingleLang_English_x64.iso
3.94 GB (4,241,291,264 bytes)

Use Rufus 2.16.1170 Portable to transfer ISO to a USB drive.

Boot to Windows 10 and the boot process still hangs (stops),  
Boot stops on window blue Logo screen with 4 blue panes in the window.  
See Log below.

Do you have any suggestions?  
What to try next?

“Most likely, the issue is with missing USB 3.0 or other modern disk drivers.“	
And does your above message change much if my Hewlett Packard computer 
BIOS is dated 7/18/2007?  (likely all USB 2 ports) 

=========================================
=========================================
Rufus version: 2.16.1170 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A838)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 15566, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x000465C4
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 119.2 GB (128034275328 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CSLA_X64FREO_EN-US_DV5'
  Size: 3.9 GB (Projected)
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1703_SingleLang_English_x64.iso

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
[... 1751 lines cut ...]

 125033471 KB total disk space.
   4157520 KB in 1716 files.
       892 KB in 109 indexes.
     71815 KB in use by the system.
     65536 KB occupied by the log file.
 120803244 KB available on disk.
      4096 bytes in each allocation unit.
  31258367 total allocation units on disk.
  30200811 allocation units available on disk.
NTFS Fixup completed.

Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A838)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 15566, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00043DDA
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 119.2 GB (128034275328 bytes)
  Start Sector: 2048, Boot: Yes

--

pbatard, 2017-09-08:
> Do you have any suggestions?

Try Windows 10 again, but using the Windows Media Creation Tool (from [here](https://www.microsoft.com/en-gb/software-download/windows10)).

I'm pretty sure your issue has nothing to do with Rufus. If you have the same problem with the MCT, you'll have to take this with either Microsoft or your manufacturer.

Joseph22a, 2017-09-08:
Please add some clarity.  Which ISO to download?  

While at this site:
Download Windows 10 Disc Image (ISO File)
https://www.microsoft.com/en-gb/software-download/windows10ISO 

There are 6 choices: 
1.	Windows 10 Creators Update, Windows 10
2.	Windows 10 Creators Update, Windows 10 N
3.	Windows 10 Creators Update, Windows 10 Single Language

4.	Windows 10 Anniversary Update, Windows 10
5.	Windows 10 Anniversary Update, Windows 10 N 
6.	Windows 10 Anniversary Update, Windows 10 Single Language

fyi 
Windows 10 N and 
Windows 10 KN editions include the same functionality as 
Windows 10, except that these versions of Windows do not include 
Windows Media Player, and related technologies.
https://www.microsoft.com/en-us/download/details.aspx?id=48231

--

pbatard, 2017-09-08:
You need to download the creation tool (hopefully that link will work - the problem is that the Microsoft website is designed to display differently depending on the OS you are running http://go.microsoft.com/fwlink/?LinkId=691209) and then select the image from within the tool. Since you're not actually going to install Windows 10 (that is unless you want to, but you'll need a valid license for it), any version should do. If you don't have a preference, you should go with Windows 10 Creators Update.

pbatard, 2017-09-14:
No update from OP &rarr; Closing.

closed 2017-09-14 by pbatard