pbatard/rufus #1020: Grub: Grub 1.99 needed or grub_divmod64_full missing

seanspotatobusiness, 2017-09-09:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I use Rufus to put an ISO on a USB. Rufus complains that "this image uses Grub 1.99-21ubuntu3.1 but the application only includes the installation files for Grub 2.02".

I click 'yes' and the image is written to the USB flash drive. I use this to boot my machine and the error given by the machine is is: "error: symbol not found: 'grub_divmod64_full'."

Is this because the wrong Grub version is used? I went here (https://launchpad.net/ubuntu/+source/grub2/1.99-21ubuntu3.1) and downloaded all three of the available files to the same directory which Rufus runs from but it does not detect them.

The Rufus version is 2.16.1170 but the log does not include this. I did not remove it from the log; it is not there to begin with.

Log
---
```
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 1.99-21ubuntu3.1
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'EasyRE'
  Size: 118.9 MB (Projected)
  Uses: GRUB2
Using image: EasyRE_WinXP.iso
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3893, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0042E82C
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32022462464 bytes)
  Start Sector: 2048, Boot: Yes
Downloading 'core.img' from https://rufus.akeo.ie/files/grub-1.99-21ubuntu3.1/core.img
Unable to access file: 404
Extended version was not found, trying main version...
Downloading 'core.img' from https://rufus.akeo.ie/files/grub-1.99/core.img
Unable to access file: 404
https://rufus.akeo.ie/files/grub-1.99-21ubuntu3.1/core.img was not found - will use embedded version
Warning: Conflicting process search failed to complete due to timeout

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE7 for exclusive write access
Requesting lock...
Will use 'J:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Grub 2.0 Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Bad Blocks: Checking from block 0 to 62545919
[... 257 lines cut ...]
     44315 KB in use by the system.
     42688 KB occupied by the log file.
  31084044 KB available on disk.
      4096 bytes in each allocation unit.
   7817983 total allocation units on disk.
   7771011 allocation units available on disk.
NTFS Fixup completed.

Found USB device 'Lexar JumpDrive USB Device' (05DC:A205) [ID]
Found non-USB removable device 'WDC WD20 02FAEX-00MJRA0 SCSI Disk Device' => Eliminated
Found USB 2.0 device 'Generic- SD/MMC USB Device' (0BDA:0119)
Found non-USB removable device 'WDC WD10 02FAEX-00Y9A0 SCSI Disk Device' => Eliminated
Found non-USB removable device 'WDC WD20 03FZEX-00Z4SA0 SCSI Disk Device' => Eliminated
Found non-USB removable device 'OCZ-VERT EX2 SCSI Disk Device' => Eliminated
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3893, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00B1E7A3
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32022462464 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2017-09-09:
I'm afraid the version of GRUB used by EasyRE_WinXP.iso is too old. Also, I no longer support anything that has to do with Windows XP, mostly for the same reason and because Microsoft dropped support for Windows XP for years now. Therefore if you still choose to use anything that's XP based, I have to tell you that you are on your own. Sorry.

closed 2017-09-09 by pbatard

billyswong, 2019-03-12:
Same bug occured for EasyRE_Win7.iso