pbatard/rufus #1022: "Please click on  The disc button to select  bootable image"

neonkodak, 2017-09-13:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

neonkodak, 2017-09-13:
Hi I have a problem attaching an ISO file for windows 10 64-bit to Rufus. 

 It just gives me the following error in the picture above.  I am currently at the last step of downloading windows 10 on an Acer chrome book and have followed the YouTube video below "https://youtu.be/SyrNH_qcvJg" can someone let me know what I'm doing wrong. Thanks.

neonkodak, 2017-09-13:
[image]
[image]
[image]

neonkodak, 2017-09-13:
[image]

neonkodak, 2017-09-13:
[image]

pbatard, 2017-09-13:
As Rufus indicates:
_Please click on the disc button_:
[image]
_to select a bootable image_.

closed 2017-09-13 by pbatard

neonkodak, 2017-09-13:
Hi, I did attach threw that disc image and properly selected my iso file and it still gives me that option. I could upload a small clip of my process if need be. Thanks.

pbatard, 2017-09-13:
No. You should follow the checklist and include the log. As specified when opening an issue, if you __choose__ not to follow the checklist, you shouldn't expect to receive support.

neonkodak, 2017-09-13:
[image]

neonkodak, 2017-09-13:
Will this be sufficient or would you rather me resubmit an issue statement I am more than willing thanks for your time.

pbatard, 2017-09-13:
> Will this be sufficient

No. I want a __FULL__ log, and I want you to attach it as a text file.

At any rate, it looks like the ISO you are trying to use is incompatible with Rufus. The problem is purely with the fact that either you're not using an official ISO or that it is broken, and this is precisely what this part of the checklist is about:

>  - [ ] If using an ISO image, I clicked on the # button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

I'm sorry but I have no time to support people who can't follow simple instructions, so please don't expect further help here. Google is your friend.