pbatard/rufus #1023: After using rufus to make usb bootable it is not working anymore

AverageGuyLT, 2017-09-14:
I was using Rufus 2.16.1170 and i tried to make the usb drive bootable with windows 10 Pro 64Bit.
Waited till the end, it was a success, but after plugging the drive out and putting in another system it was not discoverable. Then i taught its dead, so i bought a new one with 16gb 3.0, but did the same thing, and the drive is not detecting again. (same with the previous one)
Cant detect them in device manager in disk manager... Tried showing USB hard drive option, doesn't work either. Please help

pbatard, 2017-09-14:
What's the manufacturer and model of your USB flash drives?
Are you using PNY drives by any chance (such as the ones [listed at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Hardware) &mdash; those drives have a buggy firmware and shouldn't be used)?
Also, did you check the second drive to see if that wasn't a fake drive? Unfortunately, fake drives are relatively common, and will typically fail in the manner you describe (everything looks good when creating the drive, but once you replug the drive, it's dead), which is why you need to make sure you always buy a flash drive from a reputable vendor.

At any rate, you should RMA your drives. If you bought them from a reputable vendor, you should have no trouble getting them replaced. And if you do try again with a 3rd drive, please make sure you run a bad blocks check in Rufus first &mdash; it will tell you if your drive is fake or defective. You should also try to plug them on a Linux system to see if they can be detected there.

For the record, it is simply impossible for Rufus to damage drives on its own, so your issue has to come from the drives you used. For more on this, see [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive).

AverageGuyLT, 2017-09-14:
1st one was kingston DataTraveler 100 G3
(16 GB 3.0)
And the new one is Sandisk cruser glide 16GB 2.0/3.0

pbatard, 2017-09-14:
Thanks for the info.
From what I recall, the Sandisk cruser glide will be seen as a FIXED driver by Windows, as opposed to mist drives which are seen as REMOVABLE, though that shouldn't be an issue.

Just in case you missed something, can you please run Rufus with one of the drives plugged in, press [kbd]Alt[/kbd]-[kbd].[/kbd] and copy/paste your full log? This will enable USB debug and may give some information as to whether the drive is really missing, or if it's something else.

AverageGuyLT, 2017-09-14:
Rufus version: 2.16.1170
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
0 devices found
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_8087&pid_0024#5&1c9b8e1e&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[036]: USB\VID_0000&PID_0002\6&13F4C3E&0&2
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub20#4&3113f1a8&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[191]: USB\VID_8087&PID_0024\5&1C9B8E1E&0&1
Processing Hub 3:
  Hub[3] = '\\?\usb#root_hub20#4&36206eee&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[026]: USB\VID_8087&PID_0024\5&355C47BA&0&1
Processing Hub 4:
  Hub[4] = '\\?\usb#vid_8087&pid_0024#5&355c47ba&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[255]: USB\VID_04D9&PID_1603\6&DF2EE03&0&1
  Found ID[099]: USB\VID_046D&PID_C05A\6&DF2EE03&0&2
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskWDC_____WD10EARS-00MVWB051.0'
Found non-USB non-removable device 'WDC WD10EARS-00MVWB0' => Eliminated
0 devices found
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_8087&pid_0024#5&1c9b8e1e&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[036]: USB\VID_0000&PID_0002\6&13F4C3E&0&2
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub20#4&3113f1a8&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[191]: USB\VID_8087&PID_0024\5&1C9B8E1E&0&1
Processing Hub 3:
  Hub[3] = '\\?\usb#root_hub20#4&36206eee&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[026]: USB\VID_8087&PID_0024\5&355C47BA&0&1
Processing Hub 4:
  Hub[4] = '\\?\usb#vid_8087&pid_0024#5&355c47ba&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[255]: USB\VID_04D9&PID_1603\6&DF2EE03&0&1
  Found ID[099]: USB\VID_046D&PID_C05A\6&DF2EE03&0&2
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskWDC_____WD10EARS-00MVWB051.0'
Found non-USB non-removable device 'WDC WD10EARS-00MVWB0' => Eliminated
0 devices found

AverageGuyLT, 2017-09-14:
There is now 3usb's dead, 3rd one is also sandisk (same as before)
Both 3 usb's are connected to the pc while im sending you this log

AverageGuyLT, 2017-09-14:
3rd usb did not finish the installation, and failed in the middle of the process

pbatard, 2017-09-14:
Did you run the bad blocks check on the 3rd device as I asked? And do you have the log from that 3rd device?

AverageGuyLT, 2017-09-14:
Yea, the 3rd one is my fault, because it passed only 1/2 tests on bad blocks, but i need windows ASAP so i had no other option..  But both of these usb's was bought today, so i just RMA them, but i cant find kingstons warrianty check, so i just have to fix it, and maybe both sandisk

AverageGuyLT, 2017-09-14:
All i remember from the 3rd death's log that there is something failed to write, and then there is a message, that no usb found

pbatard, 2017-09-14:
> Yea, the 3rd one is my fault, because it passed only 1/2 tests on bad blocks

There you go. Do you have the failure log from Rufus? It will tell you if the bad blocks check failed because you got a fake device or if it's because of bad flash memory.

If it's because it's a fake drive, then your supplier definitely need to get their drives from a more reputable source and not the cheapest one they can find (which is likely to contain a batch of fake drives, as they don't have to pay as much for flash memory as they would will a proper drive).

At any rate, you should not use a drive that Rufus says failed the bad blocks check, but RMA it immediately.

pbatard, 2017-09-14:
> All i remember from the 3rd death's log that there is something failed to write, and then there is a message, that no usb found

How I wish you had saved that log, so that you could copy/paste it. Even if the drive failed, you should have been able to save the log...

AverageGuyLT, 2017-09-14:
On the 3rd drive i used another windows iso and another version of rufus 15.xx.xxx
so the problem is not with the iso and not with the rufus.. there was something more. But on that 3rd drive i fully take my responsibility, because it failed bad blocks test, but those other 2 is something strange. Because i did it a lot of times and it worked

pbatard, 2017-09-14:
> and another version of rufus 15.xx.xxx

There's no such thing as Rufus 15.x. The latest version is 2.17.

> but those other 2 is something strange. Because i did it a lot of times and it worked

Fake drives will tend to work as long as you don't go over the actual internal capacity (e.g. as long as you copy less than 2GB of data on a "fake" drive that pretends to be 16GB, you won't see anything). However, when using Rufus to create Windows bootable ISOs, you will tend to copy a large amount of data (at least more than 2GB), which will typically make a drive fail.
Of course, that only applies for the case your drives were fake. It's still possible that you got drives with bad flash memory, or drives that were stored for a long time in a wet environment, or something else altogether.

For the record, Rufus gets downloaded close to 3 million times each month, and the number of reports I get about drives getting defective when using Rufus is about what's expected from using devices based on flash-memory (which is not as everlasting as people think), as well as unscrupulous retailers selling devices that advertise a "fake" capacity. If Rufus did destroy drives, the internet would be filled with reports about it, especially from people running repair shops (which I know tend to use Rufus with a wide variety of flash drives), as well as reports from flash drive manufacturers who would obviously be eager to find a cause for an increase in RMA'd drives.

AverageGuyLT, 2017-09-14:
Well, thanks a lot for your help. Will RMA those drives tomorrow.

pbatard, 2017-09-14:
Okay. If you think you need any further help with this, don't hesitate to let me know.

AverageGuyLT, 2017-09-15:
Checked for bad blocks, same issue.

Rufus version: 2.17.1198
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_8087&pid_0024#5&1c9b8e1e&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[251]: USB\VID_0781&PID_5575\20044317500515136278
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub20#4&3113f1a8&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[191]: USB\VID_8087&PID_0024\5&1C9B8E1E&0&1
Processing Hub 3:
  Hub[3] = '\\?\usb#root_hub20#4&36206eee&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[026]: USB\VID_8087&PID_0024\5&355C47BA&0&1
Processing Hub 4:
  Hub[4] = '\\?\usb#vid_8087&pid_0024#5&355c47ba&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[255]: USB\VID_04D9&PID_1603\6&DF2EE03&0&1
  Found ID[099]: USB\VID_046D&PID_C05A\6&DF2EE03&0&2
Processing IDs belonging to 'USBSTOR':
  USB\VID_0781&PID_5575\20044317500515136278
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskSanDisk_Cruzer_Glide____1.27'
  Matched with ID[251]: USB\VID_0781&PID_5575\20044317500515136278
  Matched with Hub[1]: '\\?\usb#vid_8087&pid_0024#5&1c9b8e1e&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskWDC_____WD10EARS-00MVWB051.0'
Found non-USB non-removable device 'WDC WD10EARS-00MVWB0' => Eliminated
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1887, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0007FC01
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.5 GB (15521021952 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'W10X64.PRO.ENU.AUG2017'
  Size: 5.1 GB (Projected)
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: W10X64.PRO.ENU.AUG2017.iso (5.1 GB)
SearchProcess: Could not open process 4: Access Denied
SearchProcess: Could not open process 384: Access Denied
SearchProcess: Could not open process 516: Access Denied
SearchProcess: Could not open process 636: Access Denied
SearchProcess: Could not open process 764: Access Denied
SearchProcess: Could not open process 3260: Access Denied
SearchProcess: Could not open process 3352: Access Denied
SearchProcess: Could not open process 4984: Access Denied
SearchProcess: Could not open process 2648: Access Denied
SearchProcess: Could not open process 4664: Access Denied
SearchProcess: Could not open process 4: Access Denied
SearchProcess: Could not open process 384: Access Denied
SearchProcess: Could not open process 516: Access Denied
SearchProcess: Could not open process 636: Access Denied
SearchProcess: Could not open process 764: Access Denied
SearchProcess: Could not open process 3260: Access Denied
[... 548 lines cut ...]
Extracting: D:\sources\imagingprovider.dll (208.3 KB)
Extracting: D:\sources\inf\setup.cfg (5.2 KB)
Extracting: D:\sources\input.dll (314.8 KB)
Extracting: D:\sources\install.swm (4.0 GB)
  Write error [0x00000015] - retrying...
  Write error [0x00000015] - retrying...
  Write error [0x00000015]
  Error writing file: [0x00000015] The device is not ready.
Could not get Disk Extents: [0x00000002] The system cannot find the file specified.

Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub20#4&3113f1a8&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[191]: USB\VID_8087&PID_0024\5&1C9B8E1E&0&1
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub20#4&36206eee&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[026]: USB\VID_8087&PID_0024\5&355C47BA&0&1
Processing Hub 4:
  Hub[3] = '\\?\usb#vid_8087&pid_0024#5&355c47ba&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[255]: USB\VID_04D9&PID_1603\6&DF2EE03&0&1
  Found ID[099]: USB\VID_046D&PID_C05A\6&DF2EE03&0&2
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskWDC_____WD10EARS-00MVWB051.0'
Found non-USB non-removable device 'WDC WD10EARS-00MVWB0' => Eliminated
0 devices found

pbatard, 2017-09-15:
I don't see the result of your bad blocks check in the log. Also, the fact that it failed when writing a large file would tend to point to your device being a fake drive (which, without the bad blocks report, is hard to confirm or disprove). Also your device is an USB 2.0, not an USB 3.0 one.

The other possibility is that the PC you are using has an issue with its USB ports (e.g. voltage spikes or something), in which case I would strongly recommend trying from a different PC. You may also want to avoid using SanDisk Cruzer Glide devices, especially if you got them from the same supplier.

pbatard, 2017-09-19:
Any update on this? I'm afraid there's not much I can do, and I'd like to hear what happens if you try with a different type of USB Flash Drive or from a different PC, and especially, see the bad blocks report from Rufus when using that new device.

AverageGuyLT, 2017-09-19:
Well, i found an old kingston DataTraveler usb 2.0 8GB, ran bad blocks test, passed it and successfully installed windows. Thanks again for your help. (Btw used the same pc)

closed 2017-09-19 by AverageGuyLT

pbatard, 2017-09-19:
> Well, i found an old kingston DataTraveler usb 2.0 8GB, ran bad blocks test, passed it and successfully installed windows.

Thanks for the update. So at least one device managed to work! I would definitely try to tell your supplier that it looks like they got a bad batch of devices on their hand, as this is the most likely explanation for your troubles, and it may avoid other people running into similar issues...