pbatard/rufus #1024: Idea: enhancement USD device reset features for v2.18 b1201

VictorVG, 2017-09-15:
In to MSDN [**IOCTL_USB_HUB_CYCLE_PORT control code**](https://msdn.microsoft.com/en-us/library/windows/hardware/ff537340.aspx) writes then Windows Vista/7/2008 can't turn off USB device power, but some software (Zentimo xStorage Manager , [**Far Manager**]{http://svn.code.sf.net/p/farmanager/code/trunk/) is can turn off on Windows Vista/7 and Server 2008 (see ./unicode_far/hotplug.cpp and ./unicode_far/hotplug.hpp for example). Do you can use some idea like Far Manager for enchantment new features also for Windows Vista/7 and Server 2008?

pbatard, 2017-09-15:
First of all, I'd prefer if you commented on the commit itself (there's a space for comments at the end of 8286a0f63afa5c4422df38de82d2a70c35d041fa) if you want to discuss a point, rather than create a new issue.

Second, I know about the `SetupDi` call and their ability to issue a reset (as a matter of fact, I have code dating back from 2011 that can produce a reset &mdash Attached below), but the problem is: for `SetupDi` reset to work, the current executable must have the same bitsize as the native Windows. However, Rufus is a 32-bit application, meaning that, if I wanted to have a "universal" USB reset to work, I'd either have to start to publish both a 32 and 64-bit version of Rufus (overkill, if the only difference is the ability to trigger a reset, plus I'd have to ensure that 64-bit users can't use the 32-bit version, which would add to the amount of time I need to spend with basic user support), or embed a small 64-bit executable that can issue a reset in the 32-bit executable, which is doable, but an overkill.

Considering that Vista will no longer be supported by Rufus after 2.18 is out, that I don't care that much  about Server 2008 users (Rufus is aimed at the general public, __not__ sysadmins), and that this is a __hidden cheat mode__ in the first place, that aims at providing a convenient shortcut for something that's __very easy__ to replicate (unplug and replugging the device), I don't think I'm going to lose too much sleep over Window 7 users having to be asked to replug their device if I ever need reset as part of a non cheat-mode workflow (which remains to be seen).

Besides, this is just one of various things that Windows 7 users can't do in Rufus, such as not being able to create a Windows To Go drive for instance (requires Windows 8 or later), and there's only so far I am willing to go to support __old__ platforms. As far as I'm concerned, and despite what users of the platform may believe, there exists tons of differences under the hood that should prompt Windows 7 to upgrade (the first one being the ability to install Windows from an USB 3.0 port, which 7 can't do natively, and which is causing much grievance to all the people who want to stick with 7 on modern hardware).

In other words, the fact that the current reset __cheat mode__ only works on Windows 8 or later (as well as Windows XP, due to Microsoft's weird API decisions, though I haven't actually tested XP) is a non issue, and I have no plans to bring any improvements to the current cheat mode in order to help Windows 7 users.

For those interested, here's the code you can use to reset an USB device using `SetupDi`, which you will also see works only if you're running a 64-bit application on Windows x64, and fails when running it as a 32-bit app:

```C
/*
 * Reset a the USB device. However, this call only works if the arch is the same
 * meaning that if Rufus is compiled as 32 bit, this reset will not work on x64
 * and an external x64 compiler app must be used to do the same
 */
BOOL ResetUSBDevice(char* hwid)
{
	HDEVINFO dev_info = NULL;
	SP_DEVINFO_DATA dev_info_data;
	SP_PROPCHANGE_PARAMS prop_params;
	DWORD size, i, datatype;
	char buffer[MAX_PATH];
	BOOL r = FALSE;

	if (hwid == NULL)
		return FALSE;

	dev_info = SetupDiGetClassDevsA(&_GUID_DEVINTERFACE_DISK, NULL, NULL, DIGCF_PRESENT|DIGCF_DEVICEINTERFACE);
	if (dev_info == INVALID_HANDLE_VALUE) {
		uprintf("SetupDiGetClassDevs (Interface) failed: %d\n", WindowsErrorString());
		return FALSE;
	}

	dev_info_data.cbSize = sizeof(dev_info_data);
	for (i=0; SetupDiEnumDeviceInfo(dev_info, i, &dev_info_data); i++) {
		memset(buffer, 0, sizeof(buffer));
		if (!SetupDiGetDeviceRegistryPropertyA(dev_info, &dev_info_data, SPDRP_ENUMERATOR_NAME,
				&datatype, (LPBYTE)buffer, sizeof(buffer), &size)) {
			uprintf("SetupDiGetDeviceRegistryProperty (Enumerator Name) failed: %d\n", WindowsErrorString());
			continue;
		}

		if (safe_strcmp(buffer, usbstor_name) != 0)
			continue;
		memset(buffer, 0, sizeof(buffer));
		if (!SetupDiGetDeviceRegistryPropertyA(dev_info, &dev_info_data, SPDRP_HARDWAREID,
				&datatype, (LPBYTE)buffer, sizeof(buffer), &size)) {
			uprintf("SetupDiGetDeviceRegistryProperty (Hardware ID) failed: %d\n", WindowsErrorString());
			continue;
		}
		uprintf("Got HW ID '%s'\n", buffer);
		if (safe_strcmp(buffer, hwid) == 0) {
			uprintf("MATCH!!!!\n");
			prop_params.ClassInstallHeader.cbSize = sizeof(SP_CLASSINSTALL_HEADER);
			prop_params.ClassInstallHeader.InstallFunction = DIF_PROPERTYCHANGE;
			prop_params.StateChange = DICS_PROPCHANGE;
			prop_params.Scope = DICS_FLAG_CONFIGSPECIFIC;	//DICS_FLAG_GLOBAL; //
			prop_params.HwProfile = 0;
			if ( (!SetupDiSetClassInstallParamsA(dev_info, &dev_info_data, &prop_params.ClassInstallHeader, sizeof(prop_params)))
			  || (!SetupDiCallClassInstaller(DIF_PROPERTYCHANGE, dev_info, &dev_info_data)) ) {
				// will return ERROR_IN_WOW64 if ran from 32 bit app on 64
				uprintf("BOY DID IT FAIL: %s\n", WindowsErrorString());
			} else {
				uprintf("WHOA... IT ACTUALLY WORKED!\n");
				r = TRUE;
			}
			break;
		}
	}
	SetupDiDestroyDeviceInfoList(dev_info);
	return r;
}
```

closed 2017-09-15 by pbatard

VictorVG, 2017-09-15:
Ok, no question.