pbatard/rufus #1027: Rufus 2.17p no devices

AParticipant, 2017-09-19:
i was going to use rufus to upgrade my system to windows 10 but i got a problem, tried pressing Alt + f nothing happend

Here is the log.

Rufus version: 2.17.1198 (Portable)
Windows version: Windows 7 SP1 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.
0 devices found

pbatard, 2017-09-19:
Is your device really an USB one? Please bear in mind that Rufus will only list USB disks, not SATA or SSD.

What is the type, model and capacity of the drive you are trying to access? Unless you can provide that information, which you should know, I will have to close this issue...

If you are confident that you are indeed using an USB device, you can try to enable USB Debug (<kbd>Alt</kbd>-<kbd>.</kbd>) and post a new log. It should give more information about the devices that are detected on your system.

AParticipant, 2017-09-19:
Oh found my problem! thank you very much its working now.

closed 2017-09-19 by pbatard