pbatard/rufus #1035: About 2.17

BatNedas, 2017-10-03:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Hello i want to submit a problem i ve experience with rufus 2.17 i ve wanted to make bootable  ubuntu-16.04.3-desktop-amd64 but the program didnt  wanted to download syslinux . i ve got errors , after that i ve downloaded rufus 2.14 ,and i had no problems making my bootable flash !>

Log
---
```
<FULL LOG>
```

JonnyTech, 2017-10-03:
You forgot to include the log generated by Rufus when the problem occurred - please do so.

pbatard, 2017-10-04:
Can't support you if you don't follow the checklist &rarr; Closed.

closed 2017-10-04 by pbatard