pbatard/rufus #1037: syslinux/GRUB all-in-one (offline) extra package?

Tony-il-Capo, 2017-10-09:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Hi there

Could you consider either making a full Rufus package, including all syslinux/GRUB files from the server and Rufus executable, or maybe make an extra package with all syslinux/GRUB files only? It's very annoying when we try to run Rufus on PCs that are offline (LAN only) and Rufus fails to do its job because it can't connect or find syslinux files locally. 

Thanks for your time.

Log
---
```
NO LOG NEEDED
```

pbatard, 2017-10-10:
If you use a recent Syslinux or Grub based distro, you should find that, if you run Rufus without being connected to the internet, it should actually manage to create the bootable USB as expected, even as it wasn't able to download the files (if Rufus can't download the files, it will use the embedded version, which, since SysLinux and GRUB haven't produced a release for a while, usually works just fine).

Have you actually tested what happens when running offline. It may not be obvious, but there does exist a fallback mechanism that allows Rufus to actually do its job in most cases.

The online download is mostly there:
1. If you are using older versions of Syslinux/GRUB
2. As a __safety mechanism__ in case the embedded versions are incompatible with whatever version the distro chose to use.
3. If you want to create blank drivers

I would string encourage you to test the offline behaviour of Rufus on multiple recent distros, and confirm for yourselves whether Rufus actually fails to do its job.

As to providing an offline download, no, that's not gonna happen as this would only be helpful for a handful of power users, and that's not the core target of what I can spend my limited time on.

Therefore, I will close this issue. If you have a custom scenario where you need a bunch of files to be available offline, I would encourage you to create your own version of Rufus that suits this need.

closed 2017-10-10 by pbatard

Tony-il-Capo, 2017-10-10:
> Have you actually tested what happens when running offline. It may not be obvious, but there does exist a fallback mechanism that allows Rufus to actually do its job in most cases.

You think I would waste my time to come here without testing this properly?? jfc!

Again, it FAILS for every distro I've tried. Most recent  Live Fedora, Debian, Parabola... It simply shows error complaining it can't connect to server and stops. Yesterday I've tried with Fedora 27 Live Beta, failed again. 
This because
> If you are using older versions of Syslinux/GRUB
As a safety mechanism in case the embedded versions are incompatible with whatever version the distro chose to use.

> As to providing an offline download, no, that's not gonna happen as this would only be helpful for a handful of power users

Well, I found that most power users don't use this software, now I know why.

pbatard, 2017-10-10:
> Again, it FAILS for every distro I've tried. Most recent Live Fedora, Debian, Parabola... It simply shows error complaining it can't connect to server and stops.

Well, that's not what I am observing, which is precisely why I asked you about it.

FYI, I just ran this exact test against Debian's `debian-9.1.0-amd64-DVD-1.iso`:
1. Made sure there weren't any leftover from previous Rufus downloads, by ensuring that the `rufus_files` directory was deleted.
2. Disconnected the network cable from the Windows 10 machine I tested with
3. Run Rufus 2.17 with the ISO above
4. After Rufus completed (since it did complete), tested the USB on a physical BIOS-based x64 machine
5. Confirmed that Debian booted properly to the Graphical installation interface even though it wasn't able to download the files.

Rufus will of course complain that it can't access the files, but, as this test validates, it is able to complete and, provided that you are using a modern distro that didn't do anything too fancy with its bootloader, the resulting USB will actually be bootable.

Thus, I'm afraid that your statement that Rufus FAILS for recent distros, such as latest Debian, when used on a non-connected machine, can be disproved as factually incorrect, which, again, goes towards the point that I was trying to make earlier, that what you are asking for should be irrelevant in a lot of cases.

Here's the log from my test for your reference:

```
Rufus version: 2.17.1198
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0809
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x000EA573
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.2 GB (31405899776 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20170128 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Debian 9.1.0 amd64 1'
  Size: 3.6 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: debian-9.1.0-amd64-DVD-1.iso (3.6 GB)
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive G: 'こんにちは'
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x000EA573
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.2 GB (31405899776 bytes)
  Start Sector: 2048, Boot: No
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/20170128/ldlinux.sys
Network is unavailable: Internet has not be initialized.
Could not download the file - will try to use embedded 6.03 version instead
Downloading 'ldlinux.bss' from http://rufus.akeo.ie/files/syslinux-6.03/20170128/ldlinux.bss
Network is unavailable: Internet has not be initialized.
Could not download the file - will try to use embedded 6.03 version instead
[... 53 lines cut ...]
Extracting: G:\tools\loadlin.txt (31.9 KB)
Extracting: G:\win32-loader.ini (233 bytes)
Extracting: G:\.disk\base_components (5 bytes)
Extracting: G:\.disk\base_installable (0 bytes)
Extracting: G:\.disk\cd_type (4 bytes)
Extracting: G:\.disk\info (77 bytes)
Extracting: G:\.disk\mkisofs (1 KB)
Extracting: G:\.disk\udeb_include (34 bytes)
Created: G:\syslinux.cfg
Finalizing, please wait...
G:autorun.inf already exists - keeping it

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive G: 'Install Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x024A5ADF
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.2 GB (31405899776 bytes)
  Start Sector: 2048, Boot: Yes
```

Tony-il-Capo, 2017-10-10:
> Most recent Live

Try LIVE images

pbatard, 2017-10-10:
Sure. Just tested `debian-live-9.1.0-amd64-xfce.iso` in the same circumstances as the one I described before, and found __no issue there either__:

```
Rufus version: 2.17.1198
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0809
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive G: 'Install Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x024A5ADF
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.2 GB (31405899776 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20170128 (from '/isolinux/isolinux.bin')
  Detected EFI bootloader(s) (from '/boot/grub/efi.img'):
  ● 'bootx64.efi'
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'd-live 9.1.0 xfce amd64'
  Size: 1.8 GB (Projected)
  Note: File on disk is larger than reported ISO size by 336 KB...
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI (through '/boot/grub/efi.img')
Using image: debian-live-9.1.0-amd64-xfce.iso (1.8 GB)
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/20170128/ldlinux.sys
Network is unavailable: Internet has not be initialized.
Could not download the file - will try to use embedded 6.03 version instead
Downloading 'ldlinux.bss' from http://rufus.akeo.ie/files/syslinux-6.03/20170128/ldlinux.bss
Network is unavailable: Internet has not be initialized.
Could not download the file - will try to use embedded 6.03 version instead

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE5 for exclusive write access
Requesting lock...
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{a684bdd1-93b6-11e7-945b-fcaa14e7bd8e}\
Installing Syslinux 6.03...
Opened \\?\Volume{a684bdd1-93b6-11e7-945b-fcaa14e7bd8e} for shared write access
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted \\?\Volume{a684bdd1-93b6-11e7-945b-fcaa14e7bd8e}\ on G:
Copying ISO files...
Extracting files...
Image is an ISO9660 image
This image will be extracted using Rock Ridge extensions (if present)
Extracting: G:\boot\grub\efi.img (416 KB)
Extracting: G:\boot\grub\font.pf2 (4.9 KB)
Extracting: G:\boot\grub\grub.cfg (13.0 KB)
Extracting: G:\boot\grub\loopback.cfg (21 bytes)
Extracting: G:\boot\grub\x86_64-efi\acpi.mod (15.0 KB)
Extracting: G:\boot\grub\x86_64-efi\adler32.mod (1.9 KB)
Extracting: G:\boot\grub\x86_64-efi\ahci.mod (22.2 KB)
Extracting: G:\boot\grub\x86_64-efi\all_video.mod (704 bytes)
Extracting: G:\boot\grub\x86_64-efi\aout.mod (1.5 KB)
Extracting: G:\boot\grub\x86_64-efi\appleldr.mod (5.1 KB)
Extracting: G:\boot\grub\x86_64-efi\archelp.mod (4.4 KB)
Extracting: G:\boot\grub\x86_64-efi\ata.mod (8.7 KB)
Extracting: G:\boot\grub\x86_64-efi\at_keyboard.mod (6.8 KB)
Extracting: G:\boot\grub\x86_64-efi\backtrace.mod (2.4 KB)
Extracting: G:\boot\grub\x86_64-efi\bfs.mod (9.1 KB)
Extracting: G:\boot\grub\x86_64-efi\bitmap.mod (3.1 KB)
(...)
Extracting: G:\pool\main\x\xserver-xorg-input-evdev\xserver-xorg-input-evdev-udeb_2.10.5-1_amd64.udeb (23.4 KB)
Extracting: G:\pool\main\x\xserver-xorg-input-libinput\xserver-xorg-input-libinput-udeb_0.23.0-2_amd64.udeb (22.2 KB)
Extracting: G:\pool\main\x\xserver-xorg-video-fbdev\xserver-xorg-video-fbdev-udeb_0.4.4-1+b5_amd64.udeb (7.8 KB)
Extracting: G:\pool\main\z\zlib\zlib1g-udeb_1.2.8.dfsg-5_amd64.udeb (47.1 KB)
Extracting: G:\.disk\base_components (4 bytes)
Extracting: G:\.disk\base_installable (0 bytes)
Extracting: G:\.disk\cd_type (4 bytes)
Extracting: G:\.disk\info (58 bytes)
Extracting: G:\.disk\udeb_include (49 bytes)
Extracting: G:\efi\boot\bootx64.efi (from '/boot/grub/efi.img', 381.5 KB)
Created: G:\syslinux.cfg
Finalizing, please wait...
Created: G:autorun.inf
Created: G:autorun.ico

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive G: 'd-live 9.1.0 xfce amd64'
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x026672D4
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.2 GB (31405899776 bytes)
  Start Sector: 2048, Boot: Yes
```

May I suggest that you run some more tests, or provide more details on how you ran them, because what I'm observing through verifiable methodology is a bit different from the picture you are trying to paint...

Tony-il-Capo, 2017-10-10:
Don't have any Debian image at hand
Fedora Live Beta **LOG**

Rufus version: 2.17.1198 (Portable)
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0723)
Using autorun.inf label for drive D: 'Fedora-LXDE-Live-27_B-1-5'
1 device found
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 966, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001A9B37
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.4 GB (7947157504 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Fedora-LXDE-Live-27_B-1-5'
  Size: 1.1 GB (Projected)
  Note: File on disk is larger than reported ISO size by 750 KB...
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Fedora-LXDE-Live-x86_64-27_Beta-1.5.iso (1 GB)
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Could not download the file - cancelling

pbatard, 2017-10-10:
Well, the error you get on download failure is obviously different from the one I'd expect from a truly disconnected machine (which is why you should __ALWAYS__ provide a log, even if you don't think it's relevant). There's a possibility that Windows handles non-internet connected LAN vs completely offline in a manner that changes the deal in terms of error handling, so I'll check that out (obviously, since I expect Windows to behave in pretty much the same way for these 2 scenarios, and my time is limited, I only tested the easiest for me to check, which was "just unplug the cable"). But that's about as far as I'll go, as, at least for recent Debian, the fallback should work provided Windows networking does not freeze the app on error...

Tony-il-Capo, 2017-10-10:
That's probably it. Remember, I told you that computers are in a LAN environment, but **yes** I should had provided a log.

I just tested the same as above in an online pc but with Rufus blocked by firewall, and obviously got the same problem,
You should reopen this and try to fix it.

pbatard, 2017-10-10:
Well, the thing is, if I try nonexistent URLs, I still get a proper non-blocking error such as `Unable to send request: The server name could not be resolved.` and everything completes as expected.

So, it looks to me like your firewall is doing something dodgy to connection requests, and, sorry, but no, if you're using software that prevents an application to actual perform the tasks it was designed to perform, then you're on your own.

Tony-il-Capo, 2017-10-10:
> So, it looks to me like your firewall is doing something dodgy to connection requests, and, sorry, but no, if you're using software that prevents an application to actual perform the tasks it was designed to perform, then you're on your own.

As I told you, the firewall scenario was an extra test that actually give me the same problem.
What it looks to me that you're trying to ditch this asap.

The log I posted above was from a test run in a LAN setup (basically there's no internet available at all, computers are connected to each other)

I can't 'help' you more than this. I think it was very clear that the fallback mechanism doesn't always work. 
Unplugging the Ethernet cable shouldn't be the solution for this problem.

You are free to do nothing about it.

Bye.

Tony-il-Capo, 2017-10-15:
You probably will not believe it, but a friend asked me for help and guess what... here's the partial log

`...
Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Fedora-Everything-netinst-x86_64-27_Beta-1.5.iso (511 MB)
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys
Network is unavailable: Internet has not be initialized.
Could not download the file - cancelling`