pbatard/rufus #1042: Not able to format my pendrive. Getting time out period error.

sanketgupta07, 2017-10-17:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
Rufus version: 2.7.855
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'hp c350o USB Device' (03F0:F707)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 983, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0975CB23
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 6.3 MB (6658048 bytes)
  Start Sector: 104772, Boot: No, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
No drive letter was assigned...
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Could not delete drive layout: [0x00000079] The semaphore timeout period has expired.
Could not reset partitions
```

pbatard, 2017-10-17:
Rufus 2.7 is very old. The latest version of Rufus is 2.17. Please try that version.

pbatard, 2017-10-19:
Any update? If not, I will close this issue.

sanketgupta07, 2017-10-19:
Its not recognising my pendrive now with latest version of rufus. Don't
know what happened to it.

pbatard, 2017-10-19:
By the looks of it, it seems like a hardware issue. Please try with another flash drive.

sanketgupta07, 2017-10-20:
Ok Pete, thanks for your help. Happy diwali.

pbatard, 2017-10-20:
Okay. I'll close this issue. If you still encounter issues, feel free to get back to me.

closed 2017-10-20 by pbatard