pbatard/rufus #1043: BlInitializeLibrary failed error

KarlReimer, 2017-10-19:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Laptop hardware: Lenovo Thinkpad T450 20BV
Windows image: Windows 10 Enterprise Version 1709 Pro/Ent/Edu/N 64-bit ISO 4.3 GB

When creating a Win10 bootable USB using Rufus (GPT, NTFS, log below), I get the following output/error:

```
*** UEFI:NTFS (x64) ***
<some [INFO] messages that look relatively benign about NTFS drivers>
[INFO] Launching NTFS EFI loader 'efi\boot\bootx64.efi'
BlInitializeLibrary failed 0xc00000bb
```

I could provide a picture with the full log if that would be helpful.

Looking up this error brings me to [this Windows page](https://support.microsoft.com/en-za/help/4020050/blinitializelibrary-failed-xxx-error-when-you-install-or-start-an-oper
)

When remaking the USB using GPT and FAT32 I get only the following error with no other messages:
BlInitializeLibrary failed 0xc00000bb

When using "MBR partition scheme for UEFI" and the same image file, the boot succeeds and I'm able to install Windows.

Log
---
```
Rufus version: 2.17.1198
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {0B6DD735-F711-404A-BFA5-0FCE83D4B830}
Max parts: 128, Start Offset: 17408, Usable = 15472012800 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {2AE2E323-466A-469E-B07D-BA3962003507}
  Size: 14.4 GB (15469913088 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {52627D89-7FDF-49A6-A68F-628F24C520C6}
  Size: 256 KB (262144 bytes)
  Start Sector: 30216722, Attributes: 0x0000000000000000
Scanning image...
[... 1041 lines cut ...]
   3776834 total allocation units on disk.
   2642429 allocation units available on disk.
NTFS Fixup completed.

Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {86918EF3-75C3-4EC3-B47A-3FEC6FC1652B}
Max parts: 128, Start Offset: 17408, Usable = 15472012800 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {2F2E7FFF-D546-4048-AE23-AEE53B67063B}
  Size: 14.4 GB (15469913088 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {62C9B867-C15E-4400-B3E0-940370C9E362}
  Size: 256 KB (262144 bytes)
  Start Sector: 30216722, Attributes: 0x0000000000000000
Saved C:\Users\mccoitadmin\Desktop\rufus2
```

pbatard, 2017-10-20:
Thanks for the report.

I'm afraid your `BlInitializeLibrary failed 0xc00000bb` seems like a straight UEFI limitation from your Lenovo Thinkpad firmware. This is the first report I'm seeing of this error when using UEFI:NTFS, and, from googling similar errors, it seems that this is not something that can be fixed through a software change (a change which I'd have no clue where to apply, since I have literally no idea what the underlying cause is, as you're the only person reporting this).

All I can advise you to do is contact Lenovo because:
1. There is no reason UEFI:NTFS should fail with this error when it has been confirmed working with hundreds of other UEFI firmwares out there, leading to the strong deduction that this specific Lenovo firmware __is__ the issue, not UEFI:NTFS
2. Even if this was an UEFI:NTFS issue, which is doubtful, Lenovo are the only people who can investigate it, since I don't have access to any hardware that can replicate this error and the error is way too generic/obtuse to provide any clue as to what is the cause of the problem.

I wish I could help, but I really don't see any way to do so apart from getting insights on this error from the people who designed the UEFI firmware...

pbatard, 2017-10-20:
I'll just add this: As far as I can see, UEFI:NTFS seems to have done its job properly, and the issue seems to be with the Windows bootloader itself.
I understand that you may have to use UEFI:NTFS because the `install.wim` is larger than 4 GB, but have you tried creating a FAT32 drive manually, without `install.wim`, to see if you see the same error.

Clearly, if you do, then it means that neither UEFI:NTFS or Rufus are responsible for your issue.

KarlReimer, 2017-10-20:
Thanks for your response, this makes sense and I don't expect you to troubleshoot Lenovo firmware.

One clarification: it seems that GPT using NTFS or FAT32 both fail in a similar manner, but using "MBR partition scheme for UEFI" works fine. This makes it seem to me like it is not exclusively a NTFS issue.

Also, because of the other issue 1044 about the new microsoft ISOs not working, I am going to try the 1703 version of Win 10 Enterprise rather than the 1709 using the same rufus settings to see if that works. I will report back.

Also more generally I wanted to mention thanks for your work and your detailed responses. I've plumbed the depths of the other issue threads here and I've learned a lot about UEFI/Windows images from them.

pbatard, 2017-10-20:
> it seems that GPT using NTFS or FAT32 both fail in a similar manner

OK. So this eliminates UEFI:NTFS as being the cause of the problem. 

> but using "MBR partition scheme for UEFI" works fine

That's interesting. I sure wish we had some insight from Lenovo on this, to explain why it fails for GPT but not for MBR. It shouldn't really matter for an UEFI firmware what the partition scheme is, as both should be bootable in the same way, and all a (64-bit) UEFI firmware needs to do is mount the partition and execute the `/efi/boot/bootx64.efi` file it finds there.

I suspect that the issue could be with how the Windows EFI bootloader interacts with the UEFI libraries provided by the firmware, and possibly how it may try to access GPT partitions, in a manner that the firmware doesn't like...

KarlReimer, 2017-10-20:
As a test I used an older Windows image using 1703, again with GPT/NTFS. It failed in the same manner.

I don't plan on doing any further troubleshooting, since I have a working configuration, and it seems that rufus is not responsible here.

Thanks for your help on this issue.

pbatard, 2017-10-22:
I will close this issue, since I don't think there's more that I can do for the time being.

closed 2017-10-22 by pbatard