pbatard/rufus #1044: [W2G] FCU Update

Hammerfest, 2017-10-20:
Latest ISO files provided by MSFT for the FCU don't properly work (not really your fault, MSFT changed how they did things).

Use of the SW_DVD5 ISO (AiO from MSFT), fails to mount, use of the clientbusiness which includes education and enterprise, instead of prompting like I am used to, it auto sets it to Education edition.

I will update this post when I get my main testing system is back online (volswap.sys GSOD/BOSD on insider 170xx... zero chance of recovery :( ), will have ISO names, and SHA1's for reference

pbatard, 2017-10-20:
> Latest ISO files provided by MSFT for the FCU don't properly work

I tested `en-gb_windows_10_multi-edition_version_1709_updated_sept_2017_x64_dvd_100090819.iso` and `en_windows_10_multi-edition_version_1709_updated_sept_2017_x64_dvd_100090817.iso` and found no issue with the creation process. Can you please provide the logs from Rufus?

> Use of the SW_DVD5 ISO (AiO from MSFT), fails to mount

I'm sorry but have no idea what the SW_DVD5 is. I'm currently downloading an education version from the Windows 10 download site (which I'm not sure actually includes business, as it may be a different ISO from MSDN) and will test that.

> will have ISO names, and SHA1's for reference

Please do! And by all means post your USB creation log from Rufus.
Without these, I've gotta waste a lot of time trying to figure out how to replicate the issues you've seen...

pbatard, 2017-10-20:
Just tested `en_windows_10_multi-edition_vl_version_1709_updated_sept_2017_x64_dvd_100090741.iso`, which apparently is the same as `SW_DVD5_Win_Pro_Ent_Edu_N_10_1709_64BIT_English_MLF_X21-50143.ISO` (SHA-1: `1bbf886697a485c18d70ad294a09c08cb3c064ac`) and I was able to choose the WinToGo version I wanted during USB creation, with no issue mounting the ISO for image extraction...

pbatard, 2017-10-20:
And I have now tested a WinToGo drive created from `en_windows_10_multi-edition_vl_version_1709_updated_sept_2017_x64_dvd_100090741.iso` after selecting the _Enterprise_ edition when prompted by Rufus, and saw no issue. The resulting USB drive was able to boot and run Windows 10 without issues...

KarlReimer, 2017-10-20:
My issue [1043](https://github.com/pbatard/rufus/issues/1043) is using an ISO: "SW_DVD5_Win_Pro_Ent_Edu_N_10_1709_64BIT_English_MLF_X21-50143.ISO". I received this ISO from [Microsoft's Volume Licensing Service Center](https://www.microsoft.com/Licensing/servicecenter/default.aspx)

pbatard, 2017-10-20:
As I pointed out in #1043, the issue could very well be that the latest 1709 ISOs are not compatible with your Lenovo UEFI firmware. At any rate, you will need to contact the Lenovo people to see what they have to say, and you may also want to try the FAT32 test, without Rufus, to confirm that this doesn't have anything to do with how Rufus created your drive.

Also, you're not OP. Please avoid hijacking an issue, when I'm still waiting on OP to report the actual issue they have, which didn't have anything to do with a Lenovo UEFI firmware error...)

pbatard, 2017-10-22:
Since I am not seeing any issues when creating Windows To Go drives from 1709 ISOs and OP hasn't followed up, I will close this issue.

closed 2017-10-22 by pbatard