pbatard/rufus #1045: Hello. Rufus, this is latest SysLinux 6.0 or not?

Alexander1414x, 2017-10-20:
Hello Rufus 2.17 have latest Syslinux 6.0.4 http://www.syslinux.org/wiki/index.php?title=Syslinux_6_Changelog or not ? Or next version 2.18 add?

Alexander1414x, 2017-10-20:
About this 4.0.5 version is bad =(

pbatard, 2017-10-20:
> Rufus 2.17 have latest Syslinux 6.0.4

SysLinux 6.04 has __NOT__ been release (see [here](https://www.kernel.org/pub/linux/utils/boot/syslinux/) for all the __OFFICIAL__ releases), therefore it is not embedded in Rufus.

However, if you do come across an ISO that uses Syslinux 6.04-pre1 (which is the __ONLY__ pre-release of Syslinux 6.04 that has been released yet), Rufus will happily be able to support it, since it will download any required files [from our servers](https://rufus.akeo.ie/files/syslinux-6.04/).

So, even though Rufus does not embed 6.04, it does fully support any ISO based on the official 6.04 releases.

> About this 4.0.5 version is bad 

Please avoid criticizing an application __unless you actually tested it__!

First of all, as per the log, Rufus does embed __TWO__ versions of Syslinux: 4.07 __AND__ 6.03, 6.03 being the very latest __OFFICIAL RELEASE__ of Syslinux (again 6.04 has __NOT__ been officially release, and therefore it is not something that can be included in Rufus).
```
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
```

Second, if you actually bothered to test, you would have found that Syslinux 6.04-pre is already supported by Rufus anyway.

Now, if you are unhappy about the fact that Syslinux 6.04 has not yet been released, and therefore is not included in Rufus, you can try to see with the Syslinux project when they plan to release 6.04.

pbatard, 2017-10-20:
So, just to reiterate, __Rufus does use the latest Syslinux 6.0, which is Syslinux 6.03__.
Syslinux 6.04 has not been released yet, therefore it is not embedded in the Rufus application (but the -pre1 version can still be used nonetheless).

closed 2017-10-20 by pbatard

Alexander1414x, 2017-10-20:
Thank you, I’m think feature support 6.0.4 in the next time . Thank you.

kevinjwalters, 2018-07-24:
FYI, Fedora 28 iso won't write due to a similar issue with being unable to find syslinux 6.04 - the dialogue box pops up with error **Unknown internet error 0x00002f19** 

`
Rufus version: 3.1.1320
Windows version: Windows 10 64-bit (Build 17134)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found card reader device 'Generic SD16G SD Card'
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1911, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x3A663A44
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 9.1 MB (9551872 bytes)
  Start Sector: 119940, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Fedora-WS-Live-28-1-1'
  Size: 1.7 GB (Projected)
  Note: File on disk is larger than reported ISO size by 896 KB...
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Fedora-Workstation-Live-x86_64-28-1.1.iso (1.7 GB)
Downloading 'ldlinux.sys' from https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: Unknown internet error 0x00002F19
Could not download the file - cancelling
`

pbatard, 2018-07-24:
The https://rufus.ie/files/syslinux-6.04/ldlinux.sys link is valid and I am seeing no issue downloading that file.

Internet error `0x00002F19` is `ERROR_INTERNET_SEC_CERT_REV_FAILED` which means your computer was not able to complete the certificate revocation check (which __is__ completely different from meaning that the certificate using by the server has been revoked).
In other words, the problem is 100% on your end. One possibility is that your clock is not set to the current date & time (revocation validation is very time sensitive), or you have disabled revocation checks altogether.

At any rate, you will have to solve that error yourself, as it is purely environmental and does not come from  https://rufus.ie.