pbatard/rufus #1048: button layout issue

424778940z, 2017-10-24:
when I'm applying a dd image to usb_hdd, the [log] and [start] button at bottom of the window overlaps each other.

pbatard, 2017-10-24:
Please provide your __FULL__ log (__YES__, I do need to see your log, even for a cosmetic issue!!) and a screenshot, thank you. I can't do anything about this issue unless I get both these things.

pbatard, 2017-10-27:
No update &rarr; Closed.

closed 2017-10-27 by pbatard

424778940z, 2017-10-27:
[image]

here is the picture, but the usb hdd are already in use for data, I can't wipe it again, I tested on a vhd, but everything seems normal

pbatard, 2017-11-01:
I tried with multiple drives, including HDDs, but I just can't replicate that issue.