pbatard/rufus #1051: [BUG] ReFS on Windows 10 1709

o-l-a-v, 2017-10-28:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
ReFS is now only available in Windows 10 Pro for Workstasions, starting with the 1709 release.
Therefore, the ReFS choice in the file format combobox show "UDF", undefined, on my Windows 10 Pro 1709. 

Rufus should detect wheather ReFS is available, before adding it to the dropdown list.

[image]

Log
---
```
<FULL LOG>
```

pbatard, 2017-10-28:
1. Just to be pedantic about terminology: A _bug_ is when a programmed feature that does not behave as expected for the set of circumstances it was designed for. Therefore this does not qualify as a bug, as we programmed the use of ReFS with the expectation that it would always be available for all versions of Windows starting with Windows 8.1, and the only reason we have an issue is that Microsoft changed the set of circumstances we designed Rufus for. In other words, be mindful about not implying that the car manufacturer should be to blame if a car falls off a bridge, when the only reason that happened is that the bridge collapsed.
2. With this out of the way, unfortunately Windows does not provide the means of telling which file systems are available and which aren't (which is precisely why we have this issue, as the current code checks for the Windows version to decide whether Windows is available). I already had to extend the version check to include the build number for other features, and I'm not sure I want to go as far as checking the type of Windows being installed just to figure out whether ReFS should be proposed on account that:
    * There's no way of telling from the Windows version name whether ReFS will be included or not. e.g. if Microsoft releases a new "Windows Pro for Arm64" version, that does include ReFS but I do not propose it because I only known about "Windows Pro for Workstation", people are going to complain that I should just always allow ReFS, and let the formatting fail.
    * There might be a registry trick to enable the use of ReFS on non Workstation, just like there was a registry trick to enable the use of ReFS in versions of Windows 8.1 that weren't supposed to support it. yet another reason why you don't want to remove ReFS from the list
    * When selecting ReFS on a platform where Windows doesn't support it, the error message is fairly explicit about it, so I don't think I need to hold a user's hand further: 
[image]
    * ReFS is an advanced feature in the first place that is only available if you enable fixed drive listing, which, as I clearly state in the FAQ on multiple occasions, is not officially supported.

For these reasons, I don't think I'm gonna bother trying to add a half assed check (that will fail to perform its purpose as Microsoft introduces new editions with ReFS support) just so that the few advanced users who use Rufus to format a fixed drive to ReFS don't get to see a fairly explicit error message about ReFS not being supported on their platform.

closed 2017-10-28 by pbatard

o-l-a-v, 2017-10-28:
Hmm. Can't you check if "UDF" gets fed into that combobox instead of the name of the actual file system, and if it is, don't add it?

pbatard, 2017-10-28:
I haven't seen any issue with UDF. I know that what happens after ReFS formatting fails is that UDF will be selected as default, because the FS selection process is based on the current state of the USB after formatting, not the last selection, but I don't consider this a problem.

If this is not your issue, can you please elaborate. I'm not sure I quite understand what UDF has to do with all of this, and your screenshot doesn't seem to show much of anything wrong. Maybe a step by step description of what to do to see your UDF issue would help. Or better, a __log__, so i can see exactly what you did.