pbatard/rufus #1052: Unable to boot from USB

smounche, 2017-10-30:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi there,
Using win 10 64bit and latest version of Rufus. I have an win 10 ISO i downloaded from MSDN, i am trying to use rufus to create a bootable usb.
i have a new Lenovo 2017 laptop, my system support UEFI bios.
the computer won't boot into the USB -  i tried 4 different USB's.
i only Choose "GPT partition scheme fo UEFI"  - AFTER adding the ISO,
secure boot is of course DISABLED in bios.
i tried both NTFS/FAT32... nothing!
I am adding the log file;

Thanks.

  >

Log
---
```
[rufus.log](https://github.com/pbatard/rufus/files/1426477/rufus.log)

```

pbatard, 2017-10-30:
Two things:

1. Please check whether your UEFI firmware is 32-bit or 64-bit. Even if your computer has a 64-bit CPU, the UEFI firmware may be 32-bit, in which case, it will never boot a 64-bit Windows ISO in UEFI mode.

2. Try to format your USB manually to FAT32 and copy all the files from the ISO using something like 7-zip. If you see the same issue, which I'm pretty sure you will, then clearly this has nothing to do with Rufus and you will need to contact Lenovo support to find out what you need to do to make your computer boot from USB, as they are the best placed to answer that question. Unfortunately, each UEFI firmware and computer has its specific quirks when it comes to boot, and I can't help you with that.

smounche, 2017-10-31:
Ok,
SO i can confirm there is a problem with the ISO i downloaded form MSDN.
There is also a related post i found:
[https://github.com/pbatard/rufus/issues/1044](url)

So until MS fix it - don't use this ISO from MSDN:
`SW_DVD5_Win_Pro_Ent_Edu_N_10_1709_64BIT_English_MLF_X21-50143.ISO`

I just used a different one i downloaded from ms.

pbatard, 2017-10-31:
Thanks for the update. I will close this issue then.

closed 2017-10-31 by pbatard