pbatard/rufus #1057: Error extracting ISO image

Lars1145, 2017-11-17:
Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x ] I also tried one or more of the following:
  - [x ] Using a different USB drive
  - [x ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<I'm getting an error "Unable to extract ISO-imag" when trying to make a bootable USB. I hope you can help me with this issue >

Log
---
```
<Rufus version: 2.18.1213
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0406
Will use default UI locale 0x0406
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'da-DK'
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A81D)
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16018022400 bytes)
  Start Sector: 48, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'en_windows_7_8.1_10_AIO'
  Size: 11.3 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr
Using image: en_windows_7_8.1_10_AIO.iso (11.3 GB)
Warning: Conflicting process search failed to complete due to timeout

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'J:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Windows 2000/XP/2003 Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 48 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
[... 1124 lines cut ...]
Extracting: J:\sources\inf\setup.cfg (5.3 KB)
Extracting: J:\sources\input.dll (267.3 KB)
Extracting: J:\sources\install.esd (10.9 GB)
libcdio: fread (): Permission denied
  Error reading UDF file /sources/install.esd

Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A81D)
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1947, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001FE3B2
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.9 GB (16016998400 bytes)
  Start Sector: 2048, Boot: Yes

Computing checksum for 'C:\Users\Lars\Desktop\Windows 7-8.1-10 AIO (6in1) x86x64 En-us Feb2016 incl Activator-=TEAM OS=-\en_windows_7_8.1_10_AIO.iso'...
Failed to wait for event for checksum thread #1: [0x00000057] Forkert parameter.
Failed to wait for event for checksum thread #1: [0x00000057] Forkert parameter.
Failed to wait for event for checksum thread #1: [0x00000057] Forkert parameter.
Read error: [0x00000017] Datafejl (cyklisk redundanstjek)
>
```

Lars1145, 2017-11-17:
It is the same iso as #151 . I can provide you with a .torrent if that makes it easier for you

pbatard, 2017-11-17:
From the error messages you are getting, it looks to me like your ISO is corrupted.
Also, for the record, I cannot support __unofficial__ Windows ISOs. Only ISOs provided by Microsoft are supported, so if you encounter an issue while using an __unofficial__ Windows ISO, you're pretty much on your own.

What I would advise you to do is compute the SHA-1 (using a different tool than Rufus if Rufus doesn't appear to work), and then compare it with the one provided by the people who created that ISO.

pbatard, 2017-11-21:
Considering that the issue is most likely with a corrupted ISO, I will close this issue.

closed 2017-11-21 by pbatard