pbatard/rufus #1058: Error while partitioning drive

RuslanPrimak, 2017-11-21:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [X] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive
  - [X] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [X] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Error while partitioning drive

Log
---
```
Rufus version: 2.18.1213 (Portable)
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Seagate Desktop USB Device' (0BC2:3300)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 954, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 7.3 GB (7851737088 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20151222 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 16.04.3 LTS amd64'
  Size: 1.5 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-16.04.3-desktop-amd64.iso (1.5 GB)
Checking for Rufus updates...
Checking release channel...
No new release version found.

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE2 for exclusive write access
Requesting lock...
Will use 'G:' as volume mountpoint
Requesting lock...
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as 'G:' after error

Found USB 2.0 device 'Seagate Desktop USB Device' (0BC2:3300)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 954, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 7.3 GB (7851737088 bytes)
  Start Sector: 0, Boot: No
Closed Windows format prompt

Computing checksum for 'E:\Install\OS\ubuntu-16.04.3-desktop-amd64.iso'...
  MD5:	 0d9fe8e1ea408a5895cbbe3431989295
  SHA1:	 da21a922bd19b3b87931ea44095a02efba2decfe
  SHA256: 1384ac8f2c2a6479ba2a9cbe90a585618834560c477a699a4a7ebe7b5345ddc1
```

pbatard, 2017-11-21:
`Access is denied.` is pretty explicit: either your USB flash drive has an issue (but if you tried with a different one, then that's probably not the case) or something on your computer is preventing the level of access that Rufus requires to be able to repartition the drive.

You can try something like <kbd>Alt</kbd>-<kbd>Z</kbd> in Rufus to try to zero the drive before giving it another try (you can cancel the zeroing after it's been running for a little while), but I suspect you're probably going to run into the same issue. If that is the case, then I'm afraid it will be up to you to identify what other application or system restriction is preventing Rufus from accessing your drive.

RuslanPrimak, 2017-11-21:
Hello!
Thank you for your fast reaction.
I have found the cause and also have found a solution. So this thread can be marked as resolved.
By the way I supposed that the solution can be applied to the plenty of the Windows issues are linked to the Rufus. So it may be useful for adding in FAQ for the Access Denied question.
The cause: some months ago I installed MBR filter against of ransomware like a Petya. It prevents changing MBR on all drives.
The solution: use Rufus in Windows Safe Mode.

pbatard, 2017-11-21:
> some months ago I installed MBR filter against of ransomware like a Petya. It prevents changing MBR on all drives.

Can you provide the name of that software so that I can add it to the [list of software that is not compatible with Rufus](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software)?

RuslanPrimak, 2017-11-21:
> Can you provide the name of that software so that I can add it to the list of software that is not compatible with Rufus?

The name is same "MBR Filter" from here: https://www.talosintelligence.com/mbrfilter

pbatard, 2017-11-21:
Thanks. I have added that software to the list, and make sure there's also a link to the list of incompatible software in the FAQ entry for Error [0x00000005].