pbatard/rufus #1060: Read only issue and failed to created autorun.inf

Huwalahubla, 2017-11-21:
hey i have an issue
everytime i format my usb i always get a same issue, i already use 2 different usb but always get a same issue, can you please help me?
my computer always restarted everytime i turn it on after i install the usb that i think it already installed windows 7
(i can send the log if you want it)
thank you.

JonnyTech, 2017-11-21:
> (i can send the log if you want it)

**ALWAYS** send a full log of a run when the error occurs - it is part to the checklist to complete when you open a new issue - and also make sure to run a bad blocks check, by clicking the "bad blocks" check box in Rufus.

FYI, here is a copy of the checklist that you dismissed:

PLEASE READ THIS CAREFULLY: You *MUST* read and complete the checklist below, by placing an x into each [ ] (so that it shows [x], without any space inside [ and ]), BEFORE clicking on 'Submit new issue'. Failure to perform these steps, WHICH ARE ONLY THERE TO HELP *YOU*, will result in the issue being dismissed without warning.

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2017-11-21:
> my computer always restarted everytime i turn it on after i install the usb that i think it already installed windows 7

That doesn't seem to be a Rufus issue.

Also, the log and checklist are not optional.

closed 2017-11-21 by pbatard