pbatard/rufus #1065: Unsupported filesystem

Konstantin2, 2017-12-06:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hello

It does not work well for a popular multiboot disc 2k10 https://ipv6.nnm-club.name/forum/viewtopic.php?t=806125

The error appears on versions above 2.9

Error:
Unsupported filesystem

[image]

Log
---
```
Rufus version: 2.18.1213 (Portable)
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0419
NOTE: No Windows Language Pack is installed for English (English) on this system.
This means that some controls may still be displayed using the system locale.
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'JetFlash Transcend 128GB USB Device' (8564:1000)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Disk size: 128GB, Sector size: 512 bytes
Cylinders: 15367, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xBFEFCBCD
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 117.7 GB (126399053824 bytes)
  Start Sector: 960, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/2k10/grub4dos/syslinux.cfg' for Syslinux
  Warning: Could not detect Isolinux version - Forcing to 4.07 (embedded)
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: '2K10 LIVE 7.8'
  Size: 3.7 GB (Projected)
  Uses: Syslinux/Isolinux v4.07
  Uses: EFI
  Uses: Bootmgr
Using image: 2k10_7.8.iso (3.7 GB)

```

pbatard, 2017-12-06:
What happens if you change the file system (most likely to NTFS) as Rufus advises you?

Also, please be mindful that the file systems you can select depends on the option you have selected for "Partition scheme and target system type". For instance, only NTFS can be selected with the first option, because there's literally no reason to ever use FAT32 for a Windows BIOS bootable drive (which is what UEFI-CSM also means), especially as you may run into issues if the ISO contains a file larger than 4GB file.

Now, if you are booting an UEFI system, then you should select "GPT for UEFI", and you should find that you can then select FAT32 if you want.

For more background information about file system selection and Windows boot options, you may also want to read [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_dual_BIOSUEFI_mode) or [this one](https://github.com/pbatard/rufus/wiki/FAQ#Blah_UEFI_Blah_FAT32_therefore_Rufus_should_Blah).

pbatard, 2017-12-08:
No update &rarr; Closed.

closed 2017-12-08 by pbatard

Konstantin2, 2017-12-09:
I can change to NTFS but I need FAT32.
I think this check is not correct.
Prior to version 2.10, your program was written well and still use this version, the newer versions refuse to write.

Below is a link to the iso image.

magnet:?xt=urn:btih:c5a3f4f2f9e6f07cec4a8bc459e4052ac44df8e5&tr=http%3A%2F%2Fbt02.nnm-club.cc%3A2710%2F00330ff27dd41ed635946b73bc5034ba%2Fannounce&tr=http%3A%2F%2F%5B2001%3A470%3A25%3A482%3A%3A2%5D%3A2710%2F00330ff27dd41ed635946b73bc5034ba%2Fannounce

pbatard, 2017-12-09:
> but I need FAT32.

No you don't.

> Prior to version 2.10 (...)

Yes, this was a bug, which got fixed. I have yet to see a report from someone who could demonstrate that NTFS was incompatible with booting BIOS.

And, just in case you "think" you need FAT32 because you heard that UEFI can only boot FAT32 (which is factually wrong), please be mindful that, as I tried to explain above, you can select FAT32 in Rufus for UEFI if you want.

Again, if you want to assert that you need FAT32 for BIOS boot, then you will have to demonstrate why that is.