pbatard/rufus #1068: Can't build Rufus with MS VS 2017 v15.5.0

skipik, 2017-12-06:
I'm getting this:

error MSB8036: The Windows SDK version 10.0.16299.0 was not found. Install the required version of Windows SDK or change the SDK version in the project property pages or by right-clicking the solution and selecting "Retarget solution".

Everything was fine on v15.4.5 though.

pbatard, 2017-12-06:
Yup, you need to reinstall the 10.0.16299.0 SDK, as it seems Microsoft broke something in their 15.5 installer - just unselecting and selecting it might do:
[image].

I had the same issue earlier, and sorted it once I got the VS installer to reinstall the SDK.

skipik, 2017-12-06:
Yeah, it worked! Thanks a lot!

closed 2017-12-06 by pbatard