pbatard/rufus #1069: Port to .net core ?

averyfreeman, 2017-12-07:
Hi, 

I'm sorry to ask what is essentially another "can you port Rufus to Linux" question, but I was wondering how hard it would be to port the code to .NET core so it could be executed on different platforms using the CORE runtime? :)

Is Rufus even written in .NET?  Sorry, I guess that belies my level of familiarity with coding in general...

Thanks!  Love Rufus, it's absolutely the best...

pbatard, 2017-12-07:
> how hard it would be to port the code to .NET core so it could be executed on different platforms using the CORE runtime? :)

About 2x or 3x as __worse__ as simply porting the existing C code (a lot of which is actually cross-platform and comes from Linux, such as ms-sys, libcdio or the bad blocks check). Probably even worse, as .NET core would be sheltering us further from accessing the low level functions we need from the OS (though, we would probably gain with regards to simplifying file/folder selection and other things).

Having programmed in it, I do like the .NET core model, but the problem is that Microsoft decided __not__ to offer any means of providing a cross platform UI on top of it, and all the "solutions" that exist to work around that (e.g. electron) are utterly dreadful.

> Is Rufus even written in .NET?

Nope. __pure__ C.

closed 2017-12-07 by pbatard

averyfreeman, 2017-12-07:
Cool, thank you for your thoughtful reply, I sincerely appreciate it! :)