pbatard/rufus #1070: Copying ISO files excruciatingly slow

rpolve, 2017-12-07:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi,
It's taking me over 1 hour to create a bootable USB live medium of Ubuntu 16.04

I tried copying the whole .iso on the very same USB drive and it took around 6 mins with a USB 3.0 port.

I don't know if I'm doing something wrong, but it doesn't seem right.

At the time of me writing this the progress bar is roughly at 40%.

Thank you in advance,
Cheers.

Log
---
```
Rufus version: 2.18.1213
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0410
Will use default UI locale 0x0410
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'it-IT'
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Using autorun.inf label for drive E: 'alpine-standard 3.7.0 x86_64'
1 device found
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 942, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x008A8ED9
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.2 GB (7750025216 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20151222 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 16.04.3 LTS amd64'
  Size: 1.5 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-16.04.3-desktop-amd64.iso (1.5 GB)
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/20151222/ldlinux.sys
Unable to access file: 404
[... 276 lines cut ...]
Extracting: E:\boot\grub\x86_64-efi\usbserial_usbdebug.mod (2.4 KB)
Extracting: E:\boot\grub\x86_64-efi\usbtest.mod (5.5 KB)
Extracting: E:\boot\grub\x86_64-efi\usb_keyboard.mod (5.8 KB)
Extracting: E:\boot\grub\x86_64-efi\verify.mod (19.2 KB)
Extracting: E:\boot\grub\x86_64-efi\video.lst (41 bytes)
Extracting: E:\boot\grub\x86_64-efi\video.mod (9.0 KB)
Extracting: E:\boot\grub\x86_64-efi\videoinfo.mod (5.7 KB)
Extracting: E:\boot\grub\x86_64-efi\videotest.mod (5.4 KB)
Extracting: E:\boot\grub\x86_64-efi\videotest_checksum.mod (3.8 KB)
Extracting: E:\boot\grub\x86_64-efi\video_bochs.mod (9.1 KB)
Extracting: E:\boot\grub\x86_64-efi\video_cirrus.mod (9.5 KB)
Extracting: E:\boot\grub\x86_64-efi\video_colors.mod (9.9 KB)
Extracting: E:\boot\grub\x86_64-efi\video_fb.mod (28.6 KB)
Extracting: E:\boot\grub\x86_64-efi\xfs.mod (10.4 KB)
Extracting: E:\boot\grub\x86_64-efi\xnu.mod (41.1 KB)
Extracting: E:\boot\grub\x86_64-efi\xnu_uuid.mod (3.3 KB)
Extracting: E:\boot\grub\x86_64-efi\xnu_uuid_test.mod (3.3 KB)
Extracting: E:\boot\grub\x86_64-efi\xzio.mod (20 KB)
Extracting: E:\boot\grub\x86_64-efi\zfscrypt.mod (8.4 KB)
Extracting: E:\casper\filesystem.manifest (66.5 KB)
Extracting: E:\casper\filesystem.manifest-remove (3.7 KB)
Extracting: E:\casper\filesystem.size (11 bytes)
Extracting: E:\casper\filesystem.squashfs (1.4 GB)

```

pbatard, 2017-12-07:
```
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
```

Your device is operating at USB 2.0 speeds.

Please make sure that you use an USB 3.0 device and that Rufus reports `Found USB 3.0 device...` in the log.

As long as you don't see that, it's going to be slow, especially when there are many small files to copy (which is usually what happens when you __extract__ content from an ISO, as Rufus does, instead of copy the ISO as a whole... which tou can't do if you want to boot).

For the record, on my machine and with a decent USB 3.0 drive, I've never seen a Linux distro that took more than 5 minutes to be converted to bootable USB, so the problem is on your side.

rpolve, 2017-12-07:
Could there be a problem with the USB controller drivers?

I am completely sure both the device and the port are USB 3.0 capable.

pbatard, 2017-12-07:
> I am completely sure both the device and the port are USB 3.0 capable.

You may want to double check that your flash drive is really plugged on an USB 3.0 port. It can be an easy mix up with some computers.

Also, normally, if your device is really USB 3.0 but plugged on an USB 2.0 port, Rufus would report something like:
```
Found USB 2.0 device 'SanDisk Extreme USB Device' (0781:5580)
NOTE: This device is an USB 3.0 device operating at lower speed...
```

Considering that Rufus reports exactly what Windows sees by default, the fact that you don't get the note about the device operating at lower speed tells me that, as far as Windows is concerned, your `Kingston DataTraveler 3.0 USB Device` __is__ an USB 2.0 device. It definitely wouldn't be the first time a manufacturer inserts a version number into their product (`3.0`), that doesn't have anything to do with the effective USB bus speed of the device, just so that unaware customers think the device is faster than it really is.

Unless your device needs a special driver to be accessed (which of course applications like Rufus are not going to support, as we rely on the __standard__ Windows mass storage drivers), I have to say that I ma pretty certain that your device __IS__ USB 2.0, and that you will never achieve the speeds that you expect with it. I would strongly advise to get your hands on a real USB 3.0 drive (one that will get detected as USB 3.0 by both Rufus and Windows), with decent flash memory speeds (as if you have an USB 3.0 device, but with slow write speed for the memory, it's not going to change much) as you should find that your slow speed issues disappear then. An USB 3.0 flash drive with advertised __write__ speed of more than 40 MB/s should do the trick.

rpolve, 2017-12-08:
I tested the writing speed with a benchmarking software and checked the real specifications on the manifacturer's website, and it turns out you were right, it has shitty write speed.
I'm not sure how they get away with reporting "USB 3.0 performance" everywhere on the product material, seems to me like false advertising.

Thank you.

closed 2017-12-08 by rpolve