pbatard/rufus #1071: Trying to find where Windows 10/Rufus/ASUS is going wrong with trying to make bootable USB for elementary OS 0.41

fyora95, 2017-12-10:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [X] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [X] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I am a beginner at trying out Linux, and I thought Elementary OS would be a good place to start out. I have been following what to do on their site, and I downloaded Rufus. I have had several unsuccessful installs. Now it did tell me at first the program it was running into was C:\Windows\explorer.exe (r) which didn't help matters because when typing that in, it just gave me the file directory home page. I then also proceeded at trying to create bootable USB through older versions of Rufus, PowerISO and Yumi. But they also failed with errors halfway through the process. So now there are only two conclusions I can come up with.

1. The new version of elementary ISO somehow doesn't like any of the programs I have used so far (which I doubt is the case)
2. Ore Windows 10/ASUS ROG computers have some strange software that is preventing me to move forward in this department.

Here is my log:

Rufus version: 2.18.1213
Windows version: Windows 10 64-bit (Build 16299)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Found USB 2.0 device 'HGST HTS 541010A9E680 USB Device' (4971:8014)
NOTE: This device is an USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3781, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xB8D2714A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.0 GB (31103909888 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Found USB 2.0 device 'HGST HTS 541010A9E680 USB Device' (4971:8014)
NOTE: This device is an USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3781, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xB8D2714A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.0 GB (31103909888 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.05/20140113 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'elementary OS 0.4.1 Loki'
  Size: 1.3 GB (Projected)
  Note: File on disk is larger than reported ISO size by 172 KB...
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v4.05
  Uses: EFI
Using image: elementaryos-0.4.1-stable.20170814.iso (1.3 GB)

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE2 for exclusive write access
Requesting lock...
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{103832da-6566-11e5-827f-086266570685}\ was already mounted as G:\
Re-mounted volume as 'G:' after error

Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Found USB 2.0 device 'HGST HTS 541010A9E680 USB Device' (4971:8014)
NOTE: This device is an USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Disk size: 32GB, Sector size: 512 bytes
Cylinders: 3781, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xB8D2714A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.0 GB (31103909888 bytes)
  Start Sector: 2048, Boot: Yes

Any help at understanding what the issue is will be great. Thank you for taking the time to read this post. If you have any questions let me know.

fyora95, 2017-12-10:
Oh, fixed it. Looks like you can't put the ISO file in the same location as the USB, because it won't extract all of the files before deleting the ISO

closed 2017-12-10 by fyora95

pbatard, 2017-12-11:
Yup, that's precisely why you got the following line in the log:
```
ABORTED: Cannot use an image that is located on the target drive!
```