pbatard/rufus #1072: Thumb drives unreadable/crashing Windows after Rufus ISO failure

blackletum, 2017-12-19:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [x] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Tried to write an ISO to thumb drive via Rufus, failed 95% of the way through or so, and since then the thumb drive is having all sorts of strange issues on various machines. 

I tried the fixes from before -- going into Rufus and doing the List USB Hard Drives option and otherwise, but the second I plug in the usb from before (used on a different machine) it freezes immediately. Also the thumb drive makes Windows flash continuously and crashes explorer, and I get a popup like you would normally for a new attached USB device that comes and goes every second or so.

I've tried to use diskpart but it doesn't even recognize the thumb drive at all (since it seems to be coming online/going offline continuously would be my guess).

Not sure where to go from here. after using Rufus on 2 thumb drives, they cannot be recognized in windows no matter what I do.

Thanks,

blackletum

---------------

Log
---
```
<Rufus version: 2.14.1086
Windows version: Windows 10 64-bit (Build 16299)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~rc2
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0138)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:DB8A)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1886, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have an x86 Master Boot Record
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0138)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:DB8A)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1886, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have an x86 Master Boot Record
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0138)
Device eliminated because it appears to contain no media
[... 1479 lines cut ...]
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have an x86 Master Boot Record
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0138)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:DB8A)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1886, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have an x86 Master Boot Record
Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0138)
Device eliminated because it appears to contain no media
0 devices found
>
```

blackletum, 2017-12-19:
Also finally got Rufus to stop freezing and tried to format, was still flashing:

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE2 [\Device\Harddisk2\DR9] for write access
Will use 'D:' as volume mountpoint
Analyzing existing boot records...
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have an x86 Master Boot Record
read_sectors: Read error [0x00000057] The parameter is incorrect.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Could not delete drive layout: [0x00000015] The device is not ready.
Could not reset partitions
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0138)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:DB8A)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1886, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have an x86 Master Boot Record

blackletum, 2017-12-19:
A follow up: I tried with a different brand thumb drive (Corsair) using the same ISO and Rufus (newest version on that machine) and it wrote fine. 

Might be a weird thing with ADATA thumb drives and working with Rufus? 

But if you still have any answers I'm all ears.

blackletum, 2017-12-19:
Last update: 

My ADAATA is a UV128 / 16 GB USB 3.0 drive. 

I was able to get the drive back to normal by going to the ADATA website and downloading their recovery tool (which in my case was "SM3259_67 High Level Recover tool Q0509_ADATA_Q1016.exe")

The tool actually failed 3/4 the way through on both thumb drives, but aftewards I was able to reformat them through Windows as per normal. 

Please feel free to reopen this thread and drop any comments you may have, but please do not delete it in case someone in the future goes through the problems I did with these ADATA thumb drives. 

The Corsair drive worked just fine.

closed 2017-12-19 by blackletum

pbatard, 2017-12-19:
> Tried to write an ISO to thumb drive via Rufus, failed 95% of the way through or so

Usually the sign of a defective drive or defective connection.

> I tried the fixes from before -- going into Rufus and doing the List USB Hard Drives option and otherwise,

These aren't "fixes". Enabling the listing of USB HDDs does absolutely nothing with regards to accessing a drive that was detected before.

> but the second I plug in the usb from before (used on a different machine) it freezes immediately.

> and I get a popup like you would normally for a new attached USB device that comes and goes every second or so.

Definitively signs of a defective drive or defective USB connection. Also, if your device is plugged on an USB 3.0 port and you are getting the message `NOTE: This device is an USB 3.0 device operating at lower speed...`, there clearly is something wrong with your drive or USB connection (e.g. bad power supply from the USB port, bad extension cable, etc) which could explain why you saw the issue with 2 drives. Or it could be a bad batch, if you used 2 drives of the same type.

From the symptoms, I strongly suspect that your drive(s) have bad flash memory. The recovery tool may be able to somewhat hide that issue (flag the defective memory cells as non-usable or something), but ultimately, this looks like a pure hardware problem. Obviously then, there isn't much that can be done in Rufus (or Windows, since Windows had an issue detecting your drive) with regards to hardware problems, and I would encourage you to RMA the drive.