pbatard/rufus #1073: Dr.Web LiveCD .

Alexander1414x, 2017-12-19:
Hello! Im Download Dr.Web LiveCD and go write .iso  and Rufus download Syslinux . This problem boot.
Please test .iso   http://repo.drweb.com/testing/livedisk/ -this download.
LOG:
[rufus.log](https://github.com/pbatard/rufus/files/1573000/rufus.log)

Alexander1414x, 2017-12-19:
Rufus version: 2.18.1213
Windows version: Windows 10 64-bit (Build 16299)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0419
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
NOTE: The Russian (Русский) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'UFD 2.0 Silicon-Power4G USB Device' (13FE:3E00)
Using autorun.inf label for drive E: 'DrWeb LiveDisk'
1 device found
Disk type: Removable, Disk size: 3.9 Гб, Sector size: 512 bytes
Cylinders: 471, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x002927D3
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.6 GB (3879403520 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20151222 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'DrWeb LiveDisk'
  Size: 797.0  Мб (Projected)
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
Using image: live.iso (798  Мб)
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/20151222/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/ldlinux.sys
File length: 68599 bytes
Successfully downloaded 'ldlinux.sys'
Downloading 'ldlinux.bss' from http://rufus.akeo.ie/files/syslinux-6.03/20151222/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version...
Downloading 'ldlinux.bss' from http://rufus.akeo.ie/files/syslinux-6.03/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE2 for exclusive write access
Requesting lock...
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
[... 110 lines cut ...]
Extracting: E:\syslinux\ru.hlp (11.8 KB)
Extracting: E:\syslinux\ru.tr (4.0 KB)
Extracting: E:\syslinux\splash.pcx (13.5 KB)
Extracting: E:\syslinux\splash.png (2 KB)
Extracting: E:\syslinux\stdmenu.cfg (508 bytes)
Extracting: E:\syslinux\syslinux.cfg (178 bytes)
Extracting: E:\syslinux\txt.cfg (290 bytes)
Extracting: E:\syslinux\vesamenu.c32 (26.1 KB)
Created: E:\syslinux.cfg
Finalizing, please wait...
Created: E:autorun.inf
Created: E:autorun.ico

Found USB 2.0 device 'UFD 2.0 Silicon-Power4G USB Device' (13FE:3E00)
Using autorun.inf label for drive E: 'DrWeb LiveDisk'
1 device found
Disk type: Removable, Disk size: 3.9 Гб, Sector size: 512 bytes
Cylinders: 471, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x002CB2DF
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.6 GB (3879403520 bytes)
  Start Sector: 2048, Boot: Yes

pbatard, 2017-12-20:
Have you tried `DD-mode` as suggested by Rufus?

[image]

Please don't ignore messages from Rufus that are there to help you.

If you see the same issue in `DD-mode`, you will need to contact the makers of the ISO.

Alexander1414x, 2017-12-24:
Okay, I want to go test this.  Please Wait.

pbatard, 2018-01-01:
I'm gonna close this issue now. However, you can still update it when you've had a chance to test.

closed 2018-01-01 by pbatard