pbatard/rufus #1074: Unable to use an "Extended Volume Tag"

stdedos, 2017-12-20:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [x] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.
```
http://releases.ubuntu.com/16.04.3/SHA256SUMS
e28a99e1754a5b04741cd09ff557df2173a0177df79782aa0eb952d3d9803a2b
```

Issue description
-----------------
For the following image `ubuntu-16.04.3-desktop-i386.iso` I could not create an "extended" Volume Tag.
`Ubuntu 16.04.3 LTS i386` is not accepted (passing as `UBUNTU 16_0`), plain `Ubuntu` works (as `UBUNTU`).

"I don't care to fix this so as it is working in the end", but maybe there is a way that rufus can "understand" when that option is not "available" (for not-tech-savvy users)

I would like to run
```
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
```

but, I have now created my image, which I don't want to redo, unless 100% necessary (since creating "working" bootable USBs has always been a "real" challenge for me

Log
---
[rufus-ubuntu_error.log](https://github.com/pbatard/rufus/files/1574591/rufus-ubuntu_error.log)
```
<FULL LOG>
```

pbatard, 2017-12-20:
> I could not create an "extended" Volume Tag

That's because the USB will use a FAT32 file system (as opposed to an ISO9660 file system), which severely limits how a drive can be labelled. As opposed to ISO9660, a FAT label can only be 11 characters long and uppercase. That's a limitation of the file system, and there is nothing Rufus can do about that.

> but maybe there is a way that rufus can "understand" when that option is not "available" (for not-tech-savvy users)

From my experience, non tech-savvy users are very confused by limitations that only occur in one file system (as far as they are concerned, all file systems "can" do the same thing, and I already have all the trouble in the world making them understand that conversion from bootable ISO to bootable USB is __way__ more tricky than they imagine), so, unless these limitations have a direct effect on their ability to boot the device (which is why Rufus has to present users with terms like BIOS, UEFI and CSM, which already cause a lot of gried to non tech-savy users), Rufus will do its best __not__ to involve the user with what is going on behind the scenes.

Furthermore, even as the file system itself has label limitations, in Windows, there is a mechanism (extended labels in `autorun.inf`) to make it __look__ like longer, mixed case labels are possible. So Rufus tries to use that by default, to keep users mostly oblivious to something they should not concern themselves with. But of course, the "extended" label thing only works when using Windows.

Oh, and before you mention that some distros do use the volume label to locate the installation drive, bear in mind that, as a result of the FAT label limitation, Rufus does look for config files that reference a volume label, and edit them behind the scenes so that they reference the FAT label. Again, this is a process that is designed to be transparent to the user, as we really want to hide the complexity that goes into the ISO &rarr; USB conversion process so as not to confuse regular users. From what I can see from the log above (and from what I remember), this volume label conversion is not needed for Ubuntu.

So, in summary:
1. You will never be able to use an extended volume label on FAT32. If you want to, then you will need to complain to Microsoft themselves so that they create a new version of the FAT file system and wait until everyone adopts that new version, as this is a pure file system limitation.
2. If you are accessing your drive from Windows, the `autorun.inf` trick used by Rufus should make it look __as if__ you can actually use extended labels. Won't work in Linux though.
3. If you really, really want your extended label, you can __try__ using NTFS instead of FAT32 for your file system. NTFS does have some label limitations, but they are not as drastic as the FAT32 ones, and, from what I could see (when testing Debian and Arch), modern Linux distros __may__ also support booting from NTFS. Note however that the use of NTFS is completely unsupported as far as I'm concerned, so if that doesn't work, you're 100% on your own.

As I believe I answered your question, I will close this issue.

PS: When you see multiple instances of write errors in your log, as you do, you __REALLY__ want to run a bad blocks check against your drive (which can be done without using an ISO), as it really looks to me like your drive is becoming defective and is bound to cause you trouble.

closed 2017-12-20 by pbatard

stdedos, 2017-12-20:
@pbatard The reason I created this ticket, was to introduce a warning to the logging and a behind-the-scenes renaming of the flashdrive label (and not to talk FAT semantic/fuck-ups) :smile: 
Or wasn't this the reason that the process failed in the first place? :confused: 
<br>
Yeah, I get it (multiple times actually) that bootable ISO to bootable USB **is** way more tricky than I imagine (as I have exactly 0 ideas how it is done) as already I mentioned
> (since creating "working" bootable USBs has always been a "real" challenge for me)

<br>

PS: So, unchecking everything _but_ the bad blocks check, I could even do it now? :confused: Or you mean go manually with the Microsoft utility?

pbatard, 2017-12-20:
> was to introduce a warning to the logging

The Log is only meant to provide data that will help the developer (me) troubleshoot issues. It's not actually designed to provide data that users may want to see there, but that isn't useful for problem resolution. For the record, when a config file that contains a volume label is modified, there will be an explicit line in the log (which you don't see with Ubuntu as no such conversion is needed).

> Or wasn't this the reason that the process failed in the first place?

Where does your process fail? You specifically mentioned that the label thing was not creating an actual problem (_"I don't care to fix this so as it is working in the end"_). So I'm not sure I understand what you are saying now...

> So, unchecking everything but the bad blocks check, I could even do it now?

Yup. Of course, you will lose the data from your drive, but you don't need to provide an ISO or anything to just run a bad blocks check.

stdedos, 2017-12-20:
I think that
> Error writing file: [0x000003EE] Ο τόμος ενός αρχείου έχει τροποποιηθεί εξωτερικά με τέτοιο τρόπο, ώστε το ανοιχτό αρχείο να μην είναι πλέον έγκυρο.
this is where the process crashed. However, I didn't memorize exactly the message (since it crashed early).

While noticing the crashing, I saw that the label was changed and the message was saying something along the lines that "_Something_ cannot be found anymore". So then I made the connection that formatting & relabeling the drive (with a "different" volume tag) caused the issue.

My thought process went that:

1. Drive is formatted with the new label
1. In the process, it is "unmounted" for user applications
1. Drive finishes format, becomes available
1. Rufus tries a `strcmp` to verify that the drive it "commissioned" for format, is the one becomming available. After that Rufus, starts writing the files
1. `strcmp` fails, rufus crashed with the message

pbatard, 2017-12-20:
4 & 5 are wrong. Looking for labels to recognize a drive would be a very flimsy mechanism. Instead we use GUIDs, that are guaranteed by Windows not to change between re-mounts.
I would advise against "guessing" what Rufus might be doing &mdash; again, it's not as straightforward (and prone to failure) as people imagine...

And the write error is just that, a __write__ error, which means that somewhere between the moment Windows asked the data to be written and the flash memory bloc where bits should have been flipped, an error occurred, which, considering you seem to experiencing further write errors down the line, looks like a pure hardware issue.

stdedos, 2017-12-20:
> 4 & 5 are wrong. Looking for labels to recognize a drive would be a very flimsy mechanism.

I know, but my 0 knowledge and assuming-the-worst attitude lead me to wrong assumptions. Apologies for that :confused: 

Well, let's say that if I ever encounter something related, I will come back here :smile: 
Thank you for your patience and thorough explainations