pbatard/rufus #1077: Increase time showing "Press any key to boot from USB" to e.g. 30 seconds?

bsto, 2017-12-26:
I have a bootable Win7 resp. Win 10 ISO image file which can be put successfully onto an USB flash drive with Rufus.

When I start booting from this USB flash drive later then for only approx 3 seconds a message appears on the screen:

"Press any key to boot from USB.."

Is there a way (with Rufus) to increase this time to e.g. 30 seconds?

Ben

pbatard, 2017-12-26:
The duration time (5 seconds) is exactly the same as the one you get if you were to install Windows 7 from a DVD drive, and replicates Microsoft's programming choices. Rufus tries to creates bootable USBs that are as close as possible to the experience you'd get when installing from the original DVD, so the timeouts will be the same.

I'm afraid that there is no option to change that duration from within Rufus, and I am not planning to add one, as it would only benefit a very small number of people.

Can you explain the reason why you think 5 seconds is not long enough? Clearly, Microsoft decided that this duration should be more than enough when booting from DVD, especially as this "press a key to boot from..." targets semi-unattented installations and only matters for the first boot, where the operator is present to initiate the installation. I other words 5 seconds should be more than long enough...

bsto, 2017-12-26:
So if this is not a Rufus matter: Maybe I can change it myself by editing a file on the USB flash drive or inside the *.iso file?

Is this somehow possible?

pbatard, 2017-12-26:
> So if this is not a Rufus matter

Well, it is. But not directly as Rufus uses a custom MBR. So if you want to change this you need to:
1. Recompile the custom MBR
2. Recompile Rufus so that it embeds your new MBR

> Is this somehow possible?

If you know what you are doing yes. If you change [this line](https://github.com/pbatard/rufus/blob/master/res/mbr/mbr.S#L34) to `0x1D` (30 - 1 in hex) and recompile the MBR, you should get the 30 second delay you are after (you will get 30 dots onscreen though). Then you need to convert the MBR binary to hex and replace the content from [`mbr_rufus.h`](https://github.com/pbatard/rufus/blob/master/src/ms-sys/inc/mbr_rufus.h). From there, you just need to recompile Rufus, and voila! Or, if you don't want to have to recompile the MBR, you can try to figure out which of the `0x04` from `mbr_rufus.h` should be changed to `0x1D`, and recompile Rufus. However, this is as far as I'm going to help you, especially as you didn't answer my question as to why you are trying to increase the delay.

bsto, 2017-12-26:
Hmm, sounds pretty difficult.

The reason behind my question is a situation which occurs sometimes:

I plugged in the USB flash drive, boot up the computer, selected the USB flash drive as boot device and think thats all. Since the boot process from USB flash drive will normally take 2-3 minutes I switch my concentration to another computer/notebook nearby and continue my work over there (browser, email, writing something in editor).

So I oftentimes forgot that there is another prompt for pressing a key on the first computer. If I don't press it the boot will continue from hard disk (not from USB flash drive!!) and I have to wait until the OS is completely booted, then restart the computer, go into bootmanager.... This is rather time consuming and unnecessary.

You could say I am stupid and need a punishment. But I have to remember so many things (hotkeys, configuration parameters, passwords,...) for so many programs that my brains forgets less important things like a wait+keypress step when booting from USB flash drive.

If the duration time would be longer (30 or 60 seconds) there would be a good chance that I turn my head from the other screen back again and recognize the necessity to press a key.
Having a corresponding option in Rufus would ease it much.

Maybe you consider to add this option (as cmdline option would be sufficient).
I would appreciate it.

Thank you

pbatard, 2017-12-26:
> Since the boot process from USB flash drive will normally take 2-3 minutes

If you really have a PC that takes more than 60 seconds from power-up/reset to the "Press any key..." prompt, then you need to change that PC (or fiddle with your BIOS options to reduce some of the detection delays for HDD and so on).

I've used many many BIOS based computers, and I have yet to see one where it takes a full minute to get the MBR to execute (that is, unless you are playing with the boot device selection screen from your BIOS but then you are actively using that machine, and the delay between device selection and MBR execution  should be a few seconds at worst). The nice thing is that since the MBR is the very first thing being read from the boot device, and it's only 512 bytes, even if you are using a super slow USB 1.0 device, it cannot take more than a second for a PC to read and process that data, so USB speed is irrelevant, and the only delays that matters are the ones that are BIOS induced, which PC manufacturers try to keep to a minimum.

So, unless you are dealing with a poorly designed system, I have to dispute the idea that it's actually taking more than 60 seconds to get to the "Press any key..." prompt. On a standard BIOS based PC, it should take between 5 and 30 seconds.

This means that, should I decide to go with the feature you request, there would probably only ever be one person benefiting from that feature: you. Considering this, and the fact that my development time is limited, I hope you can understand that I'd rather use it on features that will benefit more than a handful of people, and as such, I have to reject your request.

You may also want to read this [FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want).

pbatard, 2017-12-26:
Just gonna add that you can disable the prompt altogether if you want: Just display the advanced options (white triangle near Format Options) and uncheck "Use Rufus MBR with BIOS ID..."

closed 2017-12-26 by pbatard