pbatard/rufus #1078: Rufus killed my brand new USB drive.

Industrial, 2017-12-29:
I have just bought a Kingston Data Traveler USB drive. I stick it into the PC. It mounts and windows sees it as a drive. I start rufus, select my ISO image and select ISO writing mode.

Rufus says its waiting for permissions. At this exact point my drive is bricked. Rufus stays in this state endlessly and I can't quit rufus. I log my user out. I log back in. Bricked.

What now?

blackcrack, 2017-12-29:
you should take a partitions manager and check if it is really brigged, because there is imho only no formatur on it, it name, the system can noch add a Drive, because there it is no partition on it, easy..

or, let it easy formatting via Rufus.. oh no, works not, because you maybe in the factory
because you work on industrial (like your name) maybe have your Systemadmin disable this and crash your stick.. Your Systemadmin is maybe the bad one ;)

You should know what you make with your hardware, if you use some tools,
on other hands, there it is no some error in the last years, Rufus is save imho..
and the software like i know, it is as it is..

how ever, pbatard have the last word :)

best regards
Blacky

pbatard, 2017-12-29:
> Rufus says its waiting for permissions.

This means that Rufus is unable to send any command to the drive. Clearly, if Rufus is being prevented from accessing the drive, Rufus cannot "brick" your drive because __it wasn't able to send any write commands to it in the first place__.

> At this exact point my drive is bricked.

Be careful about the terminology you use. Having an application telling you that it's not able to access the drive is very different from a "bricked" device. And of course, if Rufus is unable to access the drive, it won't be able to proceed. What does your log say? Even if Rufus says it can't access the drive, you should be able to access the log, which may have some information.

As @blackcrack indicates, if you can see the drive in Windows Disk Manager, then you should try to reformat it there.

From what you report, your issue is most likely that __another application__ is preventing Rufus from accessing the drive. You may have to find out which it is (you can find a list of some applications that can prevent drive access [at the end of the Rufus FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software)).

If you have a Linux or OS X platform available, I would also encourage you to try to repartition/reformat your drive there. Usually, it's only when you are unable to carry the same operations on different Operating Systems that you can declare that a drive is "bricked", and, by its design, Linux is a lot more likely to grant you access to a drive that you may believe is defective on Windows. For instance, if you use some images like some Chromium ISOs, Windows has a bug where it is possible to end up with a drive that Windows says cannot be repartitioned/reformatted (i.e. looks "bricked"), whereas Linux will have no trouble at all (and it doesn't matter whether the drive was created with Rufus, or Win32 Disk Imager of even `dd` in Linux for Windows to think that it is defective).

In other words, if you have only tried to access your drive using one application or one OS, you cannot declare that a drive is "bricked". What you must do instead is insist, with different applications and preferably a different OS.

pbatard, 2018-01-01:
@Industrial, Any update?

Industrial, 2018-01-02:
The culprit was ext2fs reader/writer. It held the permissions for the usb
stick.

The issue is resolved now ty

pbatard, 2018-01-02:
Okay, I will close the issue then.

closed 2018-01-02 by pbatard