pbatard/rufus #1080: Keymap issues after running Rufus 2.18

brundage, 2018-01-17:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [can't - see issue ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

I created a FreeDOS bootable stick using Rufus 2.18 in Windows 10 and put a third-party executable (SpinRite) on the drive.  Everything went fine, booted the drive, ran SpinRite and rebooted.  Now my keyboard map is all kinds of messed up.  Pressing some keys gives the control-key.  I can not log into either of my Linux or Windows environments.  I tried a different keyboard.  The motherboard is by ASUS, purchased two years ago.

**This issue persists across reboots without the Rufus drive inserted.**

Here is the AUTOEXEC.BAT file created by Rufus.  I'm not too familiar with DOS commands, but reading the man pages for [mode](http://help.fdos.org/en/hhstndrd/base/mode.htm) and [keyb](http://help.fdos.org/en/hhstndrd/base/keyb.htm) it seems like one or both of them may have made permanent changes to my BIOS.  Can I reverse them?

```
@echo off
set PATH=.;\;\LOCALE
display con=(ega,,1)
GOTO %CONFIG%
:1
mode con codepage prepare=((437) \locale\ega.cpx) > NUL
mode con codepage select=437 > NUL
keyb dv,,\locale\keyboard.sys
:2
```

Log
---
```
<FULL LOG>
```

brundage, 2018-01-17:
CONFIG.SYS

```
!MENUCOLOR=7,0
MENU
MENU   FreeDOS Language Selection Menu
MENU   �������������������������������
MENU
MENUDEFAULT=1,5
MENU 1) Use US-Dvorak keyboard with US-English codepage [437]
MENU 2) Use US-English keyboard with US-English codepage [437]
MENU
12?
```

pbatard, 2018-01-17:
> reading the man pages for mode and keyb it seems like one or both of them may have made permanent changes to my BIOS.

Nope, that is absolutely __not__ possible.

A standard BIOS environment does not have any readily available variable that sets the keyboard mapping, and even if a nonstandard BIOS did, it would be something so custom that it would be just impossible for an __external__ program, such as a DOS application, to figure out what it is and alter it.

The reason why `mode` and `keyb` are called is __precisely__ because the BIOS is completely agnostic to keyboard code pages (can't tell DOS which one should be preferred) and therefore, it is up to DOS itself to set one, with no possibility of relying on anything to figure which one it should use.

So, I'm afraid that, if you find that your keyboard maps are messed up, you will have to look at the configuration for each OS you boot to figure out what happens, as I can guarantee, with 100% certainty, that it wasn't the mere fact of running a Rufus created FreeDOS drive, that set your keyboard to use US mapping, that changed your settings (though, if you are not using a Dvorak keyboard, you should make sure that you select 2nd option when running FreeDOS &mdash; but again, that setting will only apply to FreeDOS and __not__ be persisted on reboot).

If I were you, I would concentrate on what you did when using SpinRite, as this is a much more believable cause for the source of your trouble than a tried and trusted keyboard mapping command from DOS. SpinRite is designed to alter your data. FreeDOS (on its own) isn't.

brundage, 2018-01-17:
Thanks for the reply, it makes sense.  The weird thing is that the keymap is wrong even before an OS is loaded.  At the BIOS screen the delete key no longer goes to the menu.  Once in the BIOS (using the alternate F2) the hotkeys for different screens no longer work either.

pbatard, 2018-01-17:
Yeah, that would sound like a possible hardware issue with your keyboard. Have you tried plugging a different keyboard if you have one?
I also suspect that you used SpinRite because you were having trouble with this machine already, right? In that case, maybe the issue you are having have to do with the hardware. For instance, I could see a PSU issue, with low voltage supply, causing all kind of weird problem such as keyboard issues or disk corruption.

Finally, just in case, you may want to check something basic like not having Caps Lock stuck or a permanently pressed key, as this could explain a lot.

brundage, 2018-01-17:
I tried a different keyboard with the same results.  When I get back home I'll plug the keyboard into this laptop to see if it produces incorrect keys.

I ran SpinRite for preventative maintenance and was not experiencing hardware issues.  Drives are all nominal.

pbatard, 2018-01-17:
Okay. The most logical guess, according to what you describe, would be that you may have one of the <kbd>Ctrl</kbd> key of your internal laptop keyboard stuck, which of course will cause trouble even if you try to plug an external keyboard. Of course, that means that, unless you actually disconnect your internal keyboard, you may not be able to properly use an external one...

brundage, 2018-01-17:
The machine in question is a desktop with an external keyboard. I'll make sure to unplug it and test a different keyboard tonight.  Thanks for the help.

brundage, 2018-01-17:
After being powered off all night and all day, problem has fixed itself. This keyboard is probably failing. Thanks again.

closed 2018-01-17 by brundage