pbatard/rufus #1083: USB3.0 Card reader being detected as a hard drive Windows 10

raidensnake, 2018-01-30:
This is the log file.

Rufus version: 2.18.1213
Windows version: Windows 10 64-bit (Build 16299)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0809
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
0 devices found
Found USB 3.0 device 'Generic- USB3.0 CRW   -SD USB Device' (056E:8008)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Generic- USB3.0 CRW   -SD USB Device' (056E:8008)
Device eliminated because it was detected as a Hard Drive (score 3 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found

pbatard, 2018-01-30:
Thanks for the report.

Can you please also post a full log after pressing <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus. This will enable device enumeration debugging, and provide the data I need to ensure that the card reader is listed by default.

raidensnake, 2018-02-02:
Here's the USB debugging.

Rufus version: 2.18.1213
Windows version: Windows 10 64-bit (Build 16299)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0809
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'Generic- USB3.0 CRW   -SD USB Device' (056E:8008)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Generic- USB3.0 CRW   -SD USB Device' (056E:8008)
Device eliminated because it was detected as a Hard Drive (score 3 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_045e&pid_02d9#5&15c311e1&0&20#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[003]: USB\VID_045E&PID_02D8\506445442542
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub20#4&5854a60&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[141]: USB\VID_8087&PID_8008\5&49593B2&1&1
Processing Hub 3:
Processing Hub 4:
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&78cd1a&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[043]: USB\VID_8087&PID_8000\5&18088FC2&1&1
Processing Hub 6:
  Hub[4] = '\\?\usb#root_hub30#4&1ed8c4ae&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[144]: USB\VID_17A0&PID_0001\5&15C311E1&0&5
  Found ID[080]: USB\VID_093A&PID_2510\5&15C311E1&0&3
  Found ID[143]: USB\VID_041E&PID_0402\5&15C311E1&0&6
  Found ID[202]: USB\VID_045E&PID_028E\01BA204
  Found ID[126]: USB\VID_0483&PID_2016\5&15C311E1&0&8
  Found ID[036]: USB\VID_045E&PID_02D9\5&15C311E1&0&9
  Found ID[035]: USB\VID_1C4F&PID_0016\5&15C311E1&0&4
  Found ID[168]: USB\VID_045E&PID_02D9\5&15C311E1&0&20
  Found ID[241]: USB\VID_2109&PID_0700\0000000000008146
  Found ID[059]: USB\VID_056E&PID_8008\201006010301
Processing Hub 7:
Processing IDs belonging to 'USBSTOR':
  USB\VID_056E&PID_8008\201006010301
  USB\VID_2109&PID_0700\0000000000008146
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskGeneric-USB3.0_CRW___-SD1.00'
  Matched with ID[059]: USB\VID_056E&PID_8008\201006010301
  Matched with Hub[4]: '\\?\usb#root_hub30#4&1ed8c4ae&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 3.0 device 'Generic- USB3.0 CRW   -SD USB Device' (056E:8008)
Device eliminated because it appears to contain no media
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskATA_____Samsung_SSD_850_2B6Q'
Found non-USB non-removable device 'Samsung SSD 850 EVO 250GB' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskATA_____Samsung_SSD_850_2B6Q'
Found non-USB non-removable device 'Samsung SSD 850 EVO 250GB' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskGeneric-USB3.0_CRW___-SD1.00'
NOTE: Matched instance from sibling for 'USB\VID_056E&PID_8008\201006010301'
  Matched with ID[059]: USB\VID_056E&PID_8008\201006010301
  Matched with Hub[4]: '\\?\usb#root_hub30#4&1ed8c4ae&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 3.0 device 'Generic- USB3.0 CRW   -SD USB Device' (056E:8008)
Device eliminated because it was detected as a Hard Drive (score 3 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives in 'Advanced Options' (after clicking the white triangle)
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskATA_____WDC_WD10EZEX-00B1A01'
Found non-USB non-removable device 'WDC WD10EZEX-00BN5A0' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskATA_____WDC_WD10EZEX-00B1A01'
Found non-USB non-removable device 'WDC WD10EZEX-00BN5A0' => Eliminated
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.

pbatard, 2018-02-06:
Thanks. I'll try to make sure that Rufus can properly list your drive in the next version (for now you can use [this](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected) to get it listed, or <kbd>Alt</kbd>-<kbd>F</kbd>).
It will probably be some time before the next version of Rufus is released though...

commit 95db209b (2018-02-14): [usb] fix default listing for Elecom card reader

* Closes #1083
closed 2018-02-14 by pbatard