pbatard/rufus #1084: Accompany binary releases with OpenPGP signature

gregar, 2018-02-01:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
It would be advantageous for binary releases to be accompanied by an OpenPGP signature (in addition to X.509 PKI Code Signing).

Many people depend on GnuPG-related GUIs as a root of trust for all subsequent software supply-chain verifications.  Without an EV (Extended Validation) code signing certificate, the risk can be too great for some to only rely on X.509 PKI code signatures, when contrasted with a well curated WoT (Web of Trust) model.

Thank you.

pbatard, 2018-02-01:
So, what you are saying is that you have __zero trust__ in the Windows ecosystem digital signature model?

I'm sorry, but, as much as I don't trust Microsoft, I have seen no reason to believe that the application validation model they use in Windows, through the standard Authenticode digital signature validation scheme, can actually be abused to the point where a second layer of validation would be needed.

I'm going to assume that your concern is that someone may be able to purchase their own credentials whilst pretending to be "Akeo Consulting". The problem with that is that, for instance, if they want to properly fool people, they'll also have to register their company in Ireland for that digital signature to be accepted by Rufus as a valid upgrade (since we do check the country of origin of the signature as well as other stuff before we accept it as valid), and even with a CA that's half asleep at their job, that's going to be difficult to pull. So I'm going to dispute that EV, or GnuPG, is actually going to bring much in terms of preventing MITM attacks. Sure, they may be able to pass themselves as a source for non-upgrade/first-time downloads, but then again, even with the best WoT, one only has to register as an entity with a somewhat similar name (e.g. "Ako Consulting"), then lay low for some time, and you can bet that the vast majority of people who download it as first instance are not going to bat an eyelid as to whether what they validated said "Ako" or something similar, whereas the website says it should be "Akeo".

Finally, and this is the most important point that people seem to forget, the whole digital signature validation model does include (near) real-time __revocation__, which, IMO, does actually make it an extremely well curated WoT.

That's because, the minute I receive evidence that someone produced a fake digitally signed version of Rufus, with malicious intent, you bet that I'm going to contact their CA so that they revoke that certificate... which Windows will automatically check (since you're downloading the app from the internet anyway, so you do have network access for revocation) and therefore prevent the malicious app from launching as a result. And, just in case that's your planned counterpoint to that, no, I'm afraid that I'm not subscribing to the idea that there could still exist rogue CAs, which won't revoke certs that they should be revoking, as having Microsoft not removing these CAs from their root CA repository the minute they are provided with evidence of what they are actually doing would undermine the whole trust mechanism of Authenticode, and antagonize an awful lot of major customers.

Being paranoid about Authenticode and the possibility of digital signature abuse is good. Heck, it keeps me up at night too. But unless you have concrete evidence that the issuing or revoking of standard Authenticode credentials is actually flawed (which I'm going to dispute: it took me more than a month and a lot of jumping through hoops to prove my identity, to get my last __regular__ Authenticode credentials issued), I'm going to continue to assume that, because there is so much at stake with letting even a handful of people abuse the system, that it is actually as secure as whatever "well curated" WoT you have in mind. Therefore, I don't think adding another layer of trust, just to satisfy a handful or super-paranoid people is not in worth the effort.

Besides, if you are that paranoid, you probably shouldn't be downloading Rufus from the web in the first place, digitally signed or not. Instead, you should be building your own version from the source. After all, what evidence do you have that the machine on which I produce Rufus hasn't been compromised in the first place, and that, even a GnuPG validated version is not going to end up doing something malicious...

Rejected.

closed 2018-02-01 by pbatard

gregar, 2018-02-01:
Well, thanks for the extended response, and I agree with many of your points.  I am a bit surprised with your tone, however.

I wasn't accusing anyone/anything of being insecure, I was merely stating that some people prefer to use OpenPGP for software validation.

Thanks again for your time.

pbatard, 2018-02-01:
Well, apologies if I misunderstood what you were saying. I was simply trying to unravel the logic that would make someone state something like _"Without an EV (Extended Validation) code signing certificate, the risk can be too great for some to only rely on X.509 PKI code signatures"_, which, at least to me, seems to be very dismissive of non EV Authenticode credentials, and which, I would assert, even if that wasn't your intention, other people may take that way, hence my attempt to counter the potential argument being made, even if that wasn't the one you were consciously trying to put forward.

gregar, 2018-02-01:
Ok, fair; I wouldn't want to create any unnecessary FUD around Authenticode either.

The keywords in my statement (for me) was "the risk *can* be too great for *some*", just to be clear.  I suppose I could have chosen softer/alternative phrasing.

Thanks.