pbatard/rufus #1086: Fail in writing ISO into micro SD Card

evrmji, 2018-02-03:
When I try to write the ISO ( 6.80GB ), I have been successful in writing once, and fail in other 11 times. 

The ISO file is complete, and had check by sha-256 ect. The ISO was support by Microsoft. The SD Card was made by SanDIsk, which was 16GB Class 3. It has been write the data less than 100GB. 

I Had try some ways in your FAQ, but it not works.
Here is the logs, but It was too big to put in here.

Complete logs.
[rufus.log](https://github.com/pbatard/rufus/files/1691788/rufus.log)
 
Apart of logs:
```
Rufus version: 2.18.1213 (Portable)
Windows version: Windows 10 64-bit (Build 16299)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0804
Will use default UI locale 0x0804
Localization set to 'zh-CN'
Found USB 2.0 device 'Mass Storage Device USB Device' (14CD:125D)
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1936, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x023A60B9
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.8 GB (15930490880 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'SSS_X64FREE_ZH-CN_DV9'
  Size: 6.8 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: 14393.0.161119-1705.RS1_REFRESH_SERVER_EVAL_X64FRE_ZH-CN.ISO (6.8 GB)

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'J:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Found volume GUID \\?\Volume{37681e30-08dd-11e8-891f-1c1b0d18f5d1}\
Volume is already mounted, but as D: instead of J: - Unmounting...
Successfully remounted \\?\Volume{37681e30-08dd-11e8-891f-1c1b0d18f5d1}\ on J:
[... 1069 lines cut ...]
Writing UEFI:NTFS partition...
  Write error [0x00000015] - retrying...
Closed Windows format prompt
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Found volume GUID \\?\Volume{3768205f-08dd-11e8-891f-1c1b0d18f5d1}\
Volume is already mounted, but as D: instead of H: - Unmounting...
Successfully remounted \\?\Volume{3768205f-08dd-11e8-891f-1c1b0d18f5d1}\ on H:
Copying ISO files...
Extracting files...
Image is an UDF image
Extracting: ..........

  Write error [0x00000015] - retrying...
  Write error [0x00000015] - retrying...
  Write error [0x00000015]
  Error writing file: [0x00000015] 设备未就绪。

0 devices found
```

pbatard, 2018-02-03:
Sorry but I have to close this issue because you chose __not__ to follow the checklist (which is there for a reason, and gives you some pointers such as trying to run a bad blocks check).

Unless you try to follow the checklist when logging an issue, I cannot support you.
Feel free to reopen an issue __after__ following the checklist and attempting what it suggests you to try.

PS: By the look of it, the issue is with your card (which is why, at the very least, you need to run a bad blocks check).

closed 2018-02-03 by pbatard

evrmji, 2018-02-04:
I am sorry to issue in here, I don't think issue in here can solve the problem, It may be too many reason to lead this issue. May the system, the poor data line, even the cosmic ray will lead this question.

I think the "issue", should be give some evident issue, or about the softwave. Didn't just give such an uncertain question.

My apology for this, thank you!