pbatard/rufus #1087: Error while partitioning drive

Spratnakus, 2018-02-06:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```
Rufus version: 2.18.1213
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Generic USB Flash Disk USB Device' (058F:1234)
Device eliminated because it appears to contain no media
0 devices found
0 devices found
Found USB 2.0 device 'Generic Flash Disk USB Device' (0000:7777)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 134GB, Sector size: 512 bytes
Cylinders: 16317, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 125 GB (134217728000 bytes)
  Start Sector: 0, Boot: No
Closed Windows format prompt
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'W10X64.PRO.ENU.JUNE.2017'
  Size: 3.4 GB (Projected)
  Uses: EFI
  Uses: Bootmgr
Using image: W10X64.PRO.ENU.JUNE.2017.iso (3.4 GB)
Warning: Conflicting process search failed to complete due to timeout

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'F:' as volume mountpoint
Requesting lock...
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000002] The system cannot find the file specified.
  StartSector: 0x0f9fff80, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as 'F:' after error

0 devices found

Spratnakus, 2018-02-06:
The USB is brand new

pbatard, 2018-02-06:
`Found USB 2.0 device 'Generic Flash Disk USB Device' (0000:7777)`

`0000` is _NOT_ a valid USB Vendor ID.
If someone is using `0000` as their vendor ID, it's a clear sign that they are __not__ producing devices with the agreement of the USB Forum. As such, I have to assume that the manufacturer of your flash drive is doing something dodgy (because using a VID that has not been assigned in consumer devices __is__ illegal), and that your drive might very well be a fake one.

Have you __really__ tried using a different flash drive or running a bad blocks check on this one?

pbatard, 2018-02-06:
I'm also wondering why you have two devices. Can you please unplug the one with VID `0000` and try only with the `058F:1234`.

I would also see your bad blocks check log for this device, so that I can confirm that you actually ran a bad blocks check against it.

Spratnakus, 2018-02-06:
I understand now... it 1:1 like Samsung MUF, but on mine there is nothing. And yes i've tried with the flash drive like first. Both of them are new. 

I'll unplug my mause and try again

Spratnakus, 2018-02-06:
Just made another try, same issue. I guess i need to buy normal USB flash drive.

Spratnakus, 2018-02-06:
FULL LOG

Partition 0:
  Type: Small FAT16 (0x00)
  Size: 125 GB (134217728000 bytes)
  Start Sector: 0, Boot: No
Found USB 2.0 device 'Generic Flash Disk USB Device' (0000:7777)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 134GB, Sector size: 512 bytes
Cylinders: 16317, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: Small FAT16 (0x00)
  Size: 125 GB (134217728000 bytes)
  Start Sector: 0, Boot: No

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'F:' as volume mountpoint
Requesting lock...
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
write_sectors: Write error [0x00000002] The system cannot find the file specified.
  StartSector: 0x0f9fff80, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as 'F:' after error

Found USB device 'Generic Flash Disk USB Device' (0000:7777)
Could not open '\\?\usbstor#disk&ven_generic&prod_flash_disk&rev_8.07#258ba81f&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000037] The specified network resource or device is no longer available.
A device was eliminated because it didn't report itself as a disk
0 devices found

pbatard, 2018-02-06:
Thanks for the update.

As much as I'd like to try to help you, I don't think a flash drive that uses a Vendor ID of `0000` is something that I ever want to support, as it's clearly not something a genuine flash drive manufacturer should ever sell. I'm also half convinced that your issue comes from using such a drive, so I would indeed request that you try with a more "genuine" flash drive, that has a proper Vendor ID assigned.

I will therefore close this issue.

But if you see a problem with a flash drive that has a non `0000` Vendor ID, feel free to reopen an issue.

closed 2018-02-06 by pbatard