pbatard/rufus #1088: Rufus Software improvements

Nima-Aram, 2018-02-11:
Look At YUMI Software to take ideas from it to make Rufus Multiboot Software please, For example I want Windows 10 and My Linux Distribution Like Manjaro Linux at one and same time on my USB Flash Disk, So, Check this Link and download and check it please:

https://www.pendrivelinux.com/yumi-multiboot-usb-creator/

I`m waiting for Rufus MultiBoot version.

Kind Regards

Nima Panahi Aram

JonnyTech, 2018-02-11:
I do not speak for @pbatard, but are you really trying to tell a developer to change their software into a different product? Just use YUMI instead. Rufus has a clearly defined usage criteria which works well.

pbatard, 2018-02-11:
This is a very frequent question that is answered [in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_add_multiboot_support). Next time, please follow the checklist, which asks you to check the FAQ to see if your questions hasn't already been answered.

closed 2018-02-11 by pbatard

Nima-Aram, 2018-02-12:
Thanks and good luck