pbatard/rufus #1089: Format error: Access to the device is denied.

lfehh54, 2018-02-13:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
Rufus version: 2.18.1213
Windows version: Windows 10 64-bit (Build 15063)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
Disk type: Removable, Disk size: 8GB, Sector size: 512 bytes
Cylinders: 979, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2877C5C5
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 7.5 GB (8052998144 bytes)
  Start Sector: 128, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'GSP1RMCPRXFREO_EN_DVD'
  Size: 3.1 GB (Projected)
  Uses: EFI (win7_x64)
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Win7_Pro_SP1_English_COEM_x64.iso (3.1 GB)
WIM extraction method(s) supported: wimgapi.dll
WIM apply method supported: wimgapi.dll

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Access denied
Format error: Access to the device is denied.
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 8GB, Sector size: 512 bytes
Cylinders: 979, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {B4B0522B-01A0-4977-99D8-0EDEC08ECB4D}
Max parts: 128, Start Offset: 17408, Usable = 8053029376 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {4683E390-2DCD-4556-B2CD-F3F1A26ABC21}
  Size: 7.5 GB (8051998208 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Closed Windows format prompt
WIM extraction method(s) supported: wimgapi.dll
WIM apply method supported: wimgapi.dll

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE1 for exclusive write access
Requesting lock...
Will use 'D:' as volume mountpoint
Requesting lock...
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Bad Blocks: Checking from block 0 to 15728639
Bad Blocks: Using offset 425 for fake device check
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Access denied
Format error: Access to the device is denied.
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 8GB, Sector size: 512 bytes
Cylinders: 979, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {D7AC5014-419C-4496-AACA-CD5BF50F76B4}
Max parts: 128, Start Offset: 17408, Usable = 8053029376 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {AB214D8B-AAD6-4C2C-80E9-80B338D3528E}
  Size: 7.5 GB (8051998208 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Closed Windows format prompt
```

pbatard, 2018-02-13:
1. Try to zero the drive with <kbd>Alt</kbd>-<kbd>Z</kbd>, then cancel it after a minute and try again
2. Try to create a FreeDOS drive and see if that works
3. I see that you are creating a GPT drive. Try to create your drive as _MBR partition scheme for UEFI_ (__not__ _MBR partition scheme for BIOS or UEFI-CSM_ which is completely different). This should work exactly the same as GPT for a Windows installation. 
4. Try to format you drive in Windows.
5. See if you are using any of the software listed as incompatible [at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software).

pbatard, 2018-02-17:
Any update on this? Did you manage to sort your issue?

pbatard, 2018-02-19:
No update &rarr; Closed.

closed 2018-02-19 by pbatard