pbatard/rufus #1096: كيف يمكنني أن اعيد كرت الذاكرة micro sd الى طبيعته بعد ان حولته الى قرص اقلاعي بواسطة rufus

Majdjojo, 2018-03-09:
Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [ ] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive
  - [ ] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2018-03-09:
I'm sorry, but support can only be provided in English. Please use [Google Translate](https://translate.google.com) if you need to translate your query to English.

Now, if you want to reformat your drive, just uncheck the _"Create bootable drive using"_ check box, select your file system and press Start. For the record, Rufus does not convert drives to "CD" or anything else. If you think that happened, then this is an issue with your computer settings and not a Rufus problem.

closed 2018-03-09 by pbatard

Majdjojo, 2018-03-09:
How can I reset the microSD memory card to its normal state after
turning it into a CD by rufus ؟

Because my mobile no longer recognizes it

pbatard, 2018-03-10:
I answered that above. In Rufus reformat your drive after unchecking the "Create bootable drive using" check box, selecting the file system you want and pressing Start.

If your SD card is still not recognized after that, then it is an issue that has nothing to do with Rufus.