pbatard/rufus #1097: USB stick hangs on boot

InteXX, 2018-03-18:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues, using keywords relevant to my problem.
- [x] I clicked the `Log` button in Rufus and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking the "bad blocks" check box in Rufus, and confirmed that my USB is not defective
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive
  - [x] Plugging the USB into a different port
  - [ ] Running Rufus on a different computer **(unable, only one computer)**
- [ ] If using an ISO image, I clicked on the `#` button (at the bottom of the Rufus interface), to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image. **(unable, Microsoft won't publish a hash for this ISO; repeated frustrating calls reveal that support staff doesn't know what a hash is)**

Issue description
-----------------
1. ISO is too big to fit on a DVD
2. Install.wim > 4GB
3. NTFS-formatted USB stick hangs on boot at `*** UEFI:NTFS (x64) ***`
4. Options for ISO -> PXE boot appear slim
5. It's starting to look like I won't be able to use this ISO (which I really, really need to do)

Log
---
```
Rufus version: 2.18.1213
Windows version: Windows 10 64-bit (Build 16299)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A815)
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1909, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {8DB6E547-A78C-490E-B3D9-2F3F20B8A572}
Max parts: 128, Start Offset: 17408, Usable = 15703439872 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Basic data partition'
  ID: {B49592DC-5197-4161-B385-579E09EBACBB}
  Size: 14.6 GB (15701377024 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CPBA_X64FRE_EN-US_DV5'
  Size: 4.7 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: Windows 10 v1709 Education, Professional, Enterprise.iso (4.7 GB)

Format operation started
Requesting disk access...
Opened \\.\PHYSICALDRIVE2 for exclusive write access
Requesting lock...
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Requesting lock...
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown Partition Boot Record
[... 1025 lines cut ...]
       380 KB in 103 indexes.
     25425 KB in use by the system.
     22752 KB occupied by the log file.
  10376540 KB available on disk.
      4096 bytes in each allocation unit.
   3833337 total allocation units on disk.
   2594135 allocation units available on disk.
NTFS Fixup completed.

Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A815)
1 device found
Disk type: Removable, Disk size: 16GB, Sector size: 512 bytes
Cylinders: 1909, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x3F9A9769
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.6 GB (15701349888 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 256 KB (262144 bytes)
  Start Sector: 30668747, Boot: No
```

pbatard, 2018-03-18:
Thanks for the report.
Three things:
1. Can you provide the SHA-1 for your ISO? I'm trying to find out if it's an official Microsoft ISO or a non-official one.
2. This looks like an incompatibility issue between the motherboard UEFI firmware you are using and UEFI:NTFS. So I'd like to know the manufacturer and model name of the motherboard you use. From what I gather some old firmware implementations may not be entirely UEFI compliant, which may cause trouble when trying to run the UEFI:NTFS bootloader. I am however very surprised that it seems to be hanging on very basic system calls. Can you confirm that `*** UEFI:NTFS (x64) ***` is really the only message you see, and that there aren't any other messages after that?
3. Have you tried with other official Windows 10 images? For the record you can download pretty much any official ISO of Windows 10 you want either from https://www.microsoft.com/en-us/software-download/windows10ISO/ (provided you access this site from a computer that doesn't run a version of Windows that can be upgraded to Windows 10, or modify your browser's User-Agent string to make the site believe that you aren't using Windows &mdash; Don't ask me why Microsoft think it's in their interest to help users get their hands on the ISOs they need) or from https://tb.rg-adguard.net/public.php (which also downloads straight from Microsoft and doesn't require to jump through hoops, even if it's not a Microsoft site).

pbatard, 2018-03-18:
Just downloaded the __official__ `Win10_Edu_1709_English_x64.iso` from Microsoft (SHA1 `1ad928cfef439f6aa4044ddc3a96b0b6830bdd0f`) and I can tell that you are definitely not using an official ISO if your `install.wim` is larger 4 GB.

The `install.wim` from Microsoft's official image is 3.82 GB.

As you can understand, since there's no telling in what incompatible manner non-official ISOs of Windows 10 may have been modified, I cannot support non-official ISOs. If you want to install Windows 10 Edu or Windows 10 Enterprise, please download the official ISO using the method highlighted above, as at the very least you should be able to create the bootable USB using FAT32.

I will no close this issue.

closed 2018-03-18 by pbatard

InteXX, 2018-03-19:
@pbatard 

`I can tell that you are definitely not using an official ISO`

I can assure you that I definitely am  ;-)

This ISO is download from Microsoft's website via my Action Pack subscription. My license keys don't work with any other images.

You won't be able to download anything without an active configured subscription, but here's the URL:

https://msppdd.microsoft.com/msppdd/Home/Downloads?selectedType=OnPremise

This `install.wim` file has six editions baked in:

> Windows 10 Education
Windows 10 Education N
Windows 10 Enterprise
Windows 10 Enterprise N
Windows 10 Pro
Windows 10 Pro N

I was able to get `install.wim` trimmed down by extracting only the edition I need, Enterprise:
```
dism /export-image /SourceImageFile:G:\sources\install.wim /SourceIndex:3 /DestinationImageFile:D:\install.wim /Compress:max /CheckIntegrity
```
I got the `SourceIndex` value with this:
```
dism /Get-WimInfo /WimFile:G:\sources\install.wim
```
So, armed with a smaller `install.wim`, I set about building a new USB stick. But I was only able to get it to work once. Once. Since that time, the system hangs at boot without anything printed to the screen whatsoever. (The time that it did work, I didn't use Rufus—I just copied the files to a FAT32 partition on a 32GB GPT stick. Now neither Rufus-created sticks nor manually-copied sticks work.)

As a side note, I find it odd that the USB stick appears twice in the available boot devices list.

I tried building a new ISO for Rufus, using the guidance found here:

http://forum.imgburn.com/index.php?showtopic=1779

**&mdash; WARNING &mdash;**

The last release of the oldish, but still excellent, ImgBurn bundles OpenCandy malware in its installer.

(More on OpenCandy: https://en.wikipedia.org/wiki/OpenCandy)

The safe recommendation is to use 7Zip after downloading to extract the .EXE from the installer. FYI this trick doesn't work for all installers, but we at least got lucky with this one. ImgBurn doesn't require any supporting files, libraries or configurations; the .EXE runs standalone without installation.

Rufusing up that ISO didn't work either, although with the smaller `install.wim` I was at least able to use FAT32.

So I undertook to burn it to a DVD-R. The burn/verify went off smoothly, but alas I find I'm unable to configure for DVD Boot in my UEFI settings. I have to switch to Legacy to get that.

It's starting to look like I'm running out of options for UEFI (which I'd like to use if at all possible). I did find this:

http://ipxe.org/

So I may yet be able to get a PXE boot. We'll see...

Since I've been able to show that I'm using an official Microsoft ISO—alas, no checksums available—you may want to consider reopening the issue. But that of course is up to you.

Thanks for going to the trouble of downloading and testing that other ISO.

InteXX, 2018-03-19:
@pbatard 

> I'd like to know the manufacturer and model name of the motherboard you use

It's a Dell Inspiron I purchased new about three years ago. See [attached](https://github.com/pbatard/rufus/files/1823648/Inspiron.pdf).

> Can you confirm that `UEFI:NTFS (x64)` is really the only message you see, and that there aren't any other messages after that?

Absolutely. When it hangs on `UEFI:NTFS (x64)` that's the only text on the screen, in bright white.

InteXX, 2018-03-19:
@pbatard 

It's working now. Apparently something's gone wonky with my USB hub. When I boot with it unplugged, I can load the USB stick. I can even load `UEFI:NTFS` without any problem.

So you'll probably want to keep the issue closed.

Thanks for your assistance/understanding. Great utility you've got here. Keep up the good work.

pbatard, 2018-03-19:
Glad you managed to sort it out, and sorry about thinking that you were using a non official ISO. I hadn't seen a combined Education + Professional + Enterprise from Microsoft, and usually, these tend to only be found in unofficial AIO (All-In-One) ISOs, which is why I assumed wrong.

But can you please tell me the MSDN name of that ISO (something like `en_windows_10_multi-edition_version_1709_updated_sept_2017_x64_dvd_100090817`) as well as its SHA-1? I'd like to take a look at it, but I can only do that if I can obtain this additional data from you...

As to your issue, I guess your wonky USB Hub is creating an issue with your UEFI firmware during device enumeration, hence the hangup. It's good to have this information, as I'll be able to tell users who run into similar symptoms to try unplugging all other USB devices they have, so thank you very much for your report and for sharing your findings.

InteXX, 2018-03-21:
YW, glad I can help  :-)

Here's the filename:

`SW_DVD9_Win_Pro_10_1709.1_64BIT_English_Pro_Ent_EDU_N_MLF_X21-67518.iso`

...and here's the SHA1 hash:

`1851A0007321FA084145EA24B8D30BF7A72BF1C6`

But I don't think you'll find it on MSDN. At least for my subscription, there's no Enterprise 1709 ISO.

InteXX, 2018-03-28:
@pbatard 

Just checking in...

Do you need anything more from me?

pbatard, 2018-03-28:
Okay, according to the SHA-1, `SW_DVD9_Win_Pro_10_1709.1_64BIT_English_Pro_Ent_EDU_N_MLF_X21-67518.iso` is also known as `en_windows_10_multi-edition_vl_version_1709_updated_dec_2017_x64_dvd_100406172.iso`, which I managed to get my hands on (and confirmed that the SHA-1 indeed matched with yours).

I created a UEFI:NTFS bootable USB from that image and tested it on an intel NUC, after plugging it to a USB 3.0 4-port hub, and saw no issue with the UEFI:NTFS boot. Windows was able to load as expected.

Therefore, I can only confirm that the issue has to do with your environment. But thanks for the details.

InteXX, 2018-03-28:
> I can only confirm that the issue has to do with your environment

Yes, I agree.

I'm now running on a clean Windows reinstall, originating from a Rufus-created USB stick using that same ISO—_after_ I unplugged my USB hub. (I decided to forego the WIM extraction; that operation has been relegated to an interesting exercise.)

This was a good follow-up, Pete...  I wonder where you get the time.